mendhak/waveshare-epaper-display #4: Error with credentials or calendar

feh123, 2020-12-02:
Hi @mendhak thanks for the project - it is a really nice idea. I wanted to use it just for my calendar. So I have installed it on a RPI +3B - a bit overkill but I have one spare and no PI zero at the moment. The Pi is running the Raspberry Pi 32bit plus desktop OS. installed via the Raspberry Pi Imager.  

The installation seemed error free and the google authorisation and credential download all went well. Because I only wanted the calendar I removed references to climacell and Pi-Hole from env.sh and run.sh - I have no Pi-Hole installed, I could use climacell as I have the api but I wanted to give all the space to the calendar.

The first problem I see is that I do not create a token.pickle file as far as I can see. So when I do ./run.sh I need to authorise again. This seems to work but I get a series of errors as the program attempts to create a .svg file. 
The putty.log shows the first error - something in the authorisation:
]0;pi@Calendar: ~[01;32mpi@Calendar[00m:[01;34m~ $[00m curl http://localhost:8080/?state=n3Qk8PU3vLxfkvmgkWAJMsQ2mpWMaM&code=4/0AY0e-g4hfIcYayzGz0GnIrylijnRJfZ-TSYs143vu71aW4RY7C-IMVLqcQMxRBiOhEgXrA&scope=https://www.googleapis.com/auth/calendar.readonly
[1] 1233
[2] 1234
]0;pi@Calendar: ~[01;32mpi@Calendar[00m:[01;34m~ $[00m The authentication flow has completed. You may close this window.
[1]-  Done                    curl http://localhost:8080/?state=n3Qk8PU3vLxfkvmgkWAJMsQ2mpWMaM
[2]+  Done                    code=4/0AY0e-g4hfIcYayzGz0GnIrylijnRJfZ-TSYs143vu71aW4RY7C-IMVLqcQMxRBiOhEgXrA
]0;pi@Calendar: ~[01;32mpi@Calendar[00m:[01;34m~ $[00m curl 
curl: try 'curl --help' or 'curl --manual' for more information
]0;pi@Calendar: ~[01;32mpi@Calendar[00m:[01;34m~ $[00m 
]0;pi@Calendar: ~[01;32mpi@Calendar[00m:[01;34m~ $[00m curl http://localhost:8080/?state=1GzkpZJEwMtgWGH7L0kt5IGbfvJwbg&code=4/0AY0e-g4znz7eyxfSZdgAAW8514Wr_RZFCus1f_dZJNSr4DSe10WTxUZBs9x2pe8j1ll68w&scope=https://www.googleapis.com/auth/calendar.readonly
[1] 1268
[2] 1269
]0;pi@Calendar: ~[01;32mpi@Calendar[00m:[01;34m~ $[00m The authentication flow has completed. You may close this window.
[1]-  Done                    curl http://localhost:8080/?state=1GzkpZJEwMtgWGH7L0kt5IGbfvJwbg
[2]+  Done                    code=4/0AY0e-g4znz7eyxfSZdgAAW8514Wr_RZFCus1f_dZJNSr4DSe10WTxUZBs9x2pe8j1ll68w
]0;pi@Calendar: ~[01;32mpi@Calendar[00m:[01;34m~ $[00m exit
logout
                                                                                                                                                                                                                                                                                                                                                                   Traceback (most recent call last):
  File "screen-calendar-get.py", line 32, in <module>
    creds = flow.run_local_server()
  File "/usr/local/lib/python3.7/dist-packages/google_auth_oauthlib/flow.py", line 474, in run_local_server
    self.fetch_token(authorization_response=authorization_response)
  File "/usr/local/lib/python3.7/dist-packages/google_auth_oauthlib/flow.py", line 288, in fetch_token
    return self.oauth2session.fetch_token(self.client_config["token_uri"], **kwargs)
  File "/usr/lib/python3/dist-packages/requests_oauthlib/oauth2_session.py", line 187, in fetch_token
    state=self._state)
  File "/usr/lib/python3/dist-packages/oauthlib/oauth2/rfc6749/clients/web_application.py", line 174, in parse_request_uri_response
    response = parse_authorization_code_response(uri, state=state)
  File "/usr/lib/python3/dist-packages/oauthlib/oauth2/rfc6749/parameters.py", line 227, in parse_authorization_code_response
    raise MissingCodeError("Missing code parameter in response.")
oauthlib.oauth2.rfc6749.errors.MissingCodeError: (missing_code) Missing code parameter in response.
]0;pi@Calendar: ~/waveshare-epaper-display[01;32mpi@Calendar[00m:[01;34m~/waveshare-epaper-display $[00m 
[K]0;pi@Calendar: ~/waveshare-epaper-display[01;32mpi@Calendar[00m:[01;34m~/waveshare-epaper-display $[00m exit
logout

I can run screen-calendar-get.py but I then get the error weather.svg file does not exist.

I have tried downloading the credentials several times with no success. Any thoughts on what is wrong much appreciated!

mendhak, 2020-12-02:
So I'm wondering if this is actually a file permission issue, considering that you mentioned a token file didn't appear.  
Could you try running that specific line manually and go through the process. 

    sudo -E python3 screen-calendar-get.py

You should see this first

> Please visit this URL to authorize this application: https://accounts.google.com/o/oauth2/auth?response_type=code&......

Then you copy paste that into a browser window.  Browser tells you that it's unsafe, but you choose to continue.  You get prompted, yes, allow reading the calendar, continue continue.  Eventually you get an error page because the browser tried to go to localhost 8080. 

Open a _new_ SSH session to the Raspberry Pi.  curl that URL from the browser, but **make sure you surround it with double quotes**.  

I say that because I just noticed this in your output

> [1] 1268
[2] 1269

That's showing two tasks that got sent to the background.  Because the `&` gets interpreted by bash isn't it.

mendhak, 2020-12-02:
Hey one more thing I edited your post to remove the codes, please be sure to remove those before posting online, the code can be used to make calls to the Calendar API and see your events.  I think to be on the safe side you should delete the credentials you created in the Google Console and create a new one and download that new credentials file.  Then do the above steps.

feh123, 2020-12-02:
Thanks very much for all the idaes and for the info about the codes - I did not notice the codes!

feh123, 2020-12-03:
Hi @mendhak - yes the "" made a big difference and I have created a token.pickle file. And I have deleted all my previous codes! However on running curl "http://localhost..." in a second ssh session from the home/pi folder I get the following - it still stalls at the .svg file:

Pickle is stale, calling the Calendar API                                                                      
Thu Dec 03 ..... Delivery sometime                                                                         
Thu Dec 03 .......Delivery sometime API                                                                          
Traceback (most recent call last):                                                                             
  File "screen-calendar-get.py", line 80, in <module>                                                          
    output = codecs.open(template , 'r', encoding='utf-8').read()                                              
  File "/usr/lib/python3.7/codecs.py", line 898, in open                                                       
    file = builtins.open(filename, mode, buffering)                                                            
FileNotFoundError: [Errno 2] No such file or directory: 'screen-output-weather.svg'                            

So its downloading my diary entities correctly but fails at some issue with creating the file. In waveshare-epaper-display the only svg file I have is screen-template.svg. Is there a log file I could check for the reason? Thanks.

feh123, 2020-12-03:
Hi @mendhak I have had some success. I can add my Climacell API and if I do that then I do create the bmp and png:

./run.sh
__        __         _   _                 _        __       
\ \      / /__  __ _| |_| |__   ___ _ __  (_)_ __  / _| ___  
 \ \ /\ / / _ \/ _` | __| '_ \ / _ \ '__| | | '_ \| |_ / _ \ 
  \ V  V /  __/ (_| | |_| | | |  __/ |    | | | | |  _| (_) |
   \_/\_/ \___|\__,_|\__|_| |_|\___|_|    |_|_| |_|_|  \___/ 
                                                             
snow_heavy 6 1 Thu Dec 03
  ____      _                _              _        __       
 / ___|__ _| | ___ _ __   __| | __ _ _ __  (_)_ __  / _| ___  
| |   / _` | |/ _ \ '_ \ / _` |/ _` | '__| | | '_ \| |_ / _ \ 
| |__| (_| | |  __/ | | | (_| | (_| | |    | | | | |  _| (_) |
 \____\__,_|_|\___|_| |_|\__,_|\__,_|_|    |_|_| |_|_|  \___/ 
                                                              
Found cached calendar response
Thu Dec 03 Joyce DiDonato Saffron Hall - cancelled
Fri Dec 04 One Stop to drain
 _____                       _   
| ____|_  ___ __   ___  _ __| |_ 
|  _| \ \/ / '_ \ / _ \| '__| __|
| |___ >  <| |_) | (_) | |  | |_ 
|_____/_/\_\ .__/ \___/|_|   \__|
           |_|                   
DPI: 150
Background RRGGBBAA: ffffff00
Area 0:0:800:480 exported to 640 x 384 pixels (76.8 dpi)
Bitmap saved as: screen-output.png
 ____  _           _             
|  _ \(_)___ _ __ | | __ _ _   _ 
| | | | / __| '_ \| |/ _` | | | |
| |_| | \__ \ |_) | | (_| | |_| |
|____/|_|___/ .__/|_|\__,_|\__, |
            |_|            |___/ 
0

bcm2835 init success !!! 

They are present in the folder and if I open the bmp it looks just like your picture. (Is there a way to get more calendar entries?)
Still one problem! Nothing appears on my epaper. I tried to run some of the demos but when I clone the waveshare demos then I do not see the file structure you describe. I did run one epd.py for 7x5 V2 and that gave no errors (but nothing appeared on the epaper). Seems close! Is there a good way to check the epaper is connected? Any advice much appreciated.

feh123, 2020-12-03:
Hi @mendhak apologies for so many updates but I wanted to say that I managed to get one of the epd files to work and the epaper is working and from the pictures the program ran through it looks correctly configured. So if I create the bmp and png what is stopping them appearing on the epaper?

mendhak, 2020-12-03:
Ah yes it's true that you need all the steps.  Because each step produces an output which the next steps reads from.  Yes I had missed that.  

OK so sounds like you now have a BMP but it's not appearing on screen.   But you tried the samples and that does work. Which means the display is connected and listening. 

You could try some troubleshooting.  In this same repo you will see a binary called `display` which takes a BMP and pushes to screen. Try it with your BMP. 

    sudo display/display screen-output.bmp 0

And then try it with that 7x5 BMP as well.  So the `display` binary but with their BMP.  

Remember that you need to push a .bmp to the screen, and one of the things I'm doing in the code is converting the SVG to PNG, then PNG to a 2-bit BMP, because the Waveshare display works with it the fastest. (Anything more complicated takes a lot of time).

feh123, 2020-12-03:
Hi @mendhak thanks for the ideas. I am still missing the Pi-Hole - but that does appear on the bmp - two lines re DNS and ADS blocked. It is as you say. I found a epd in the waveshare examples I installed from your website and that worked.
From the folder waveshare-epaper-display I tried sudo display/display screen-output.bmp 0 - the terminal showed one 0 and then bcm2835 init success but nothing appears on the epaper screen. I also tried the 7in5_V2.bmp from their sample folder - same terminal screen and nothing on the epaper.  Am I doing something wrong? I am not sure if I have pushed the bmp. 
I notice the folder names and structures for the git cloned version I downloaded from your gitpage are different from the ones you describe. Also my bcm2835 is 58 and the latest is 60 - should I update? Appreciate all the help!

mendhak, 2020-12-03:
OK I think there might be another clue.  When you ran the waveshare's sample code you must have had to do a `make` command first isn't it? 

But with my version you're running the `display` binary directly... and that was built for Raspberry Pi Zero two years ago.  I  wonder if that's causing a problem.  I wonder if it's related to architecture version.  You can find out the architecture by running this command

    uname -a

I got armv6l: 

> Linux epaper 5.4.51+ #1333 Mon Aug 10 16:38:02 BST 2020 armv6l GNU/Linux

Anyway what you can try is go into the [display folder](https://github.com/mendhak/waveshare-epaper-display/tree/master/display) and delete the `display` binary in there, then try to run that Makefile in there.  Just try `make` and that should find the Makefile and build the code inside `obj` and output a `display` binary. 

Then try again with that `display` binary.

feh123, 2020-12-04:
Hi @mendhak thanks for this. 

For uname I get: Linux Calendar 5.4.72-v7+ #1356 SMP Thu Oct 22 13:56:54 BST 2020 armv7l GNU/Linux 

To run the waveshare sample code I did not use the make command - should I do that? Instead I git cloned the waveshare demo link on your web site and went to /home/pi/waveshare-epaper-sample/RaspberryPi&JetsonNano/python/examples and from WinSCP I can run the command ./epd on file 7in5_V2_test.py - this then runs on the epaper and I see pictures. It does not complete and I do get an error right at the end of the test but at least I see pictures. The error message is: Command '"./epd_7in5_V2_test.py"'
failed with return code 0 and error message
INFO:root:epd7in5_V2 Demo
INFO:root:init and Clear
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy
INFO:root:1.Drawing on the Horizontal image...
DEBUG:root:Vertical
DEBUG:root:e-Paper busy
INFO:root:2.Drawing on the Vertical image...
DEBUG:root:Horizontal
DEBUG:root:e-Paper busy
INFO:root:3.read bmp file
DEBUG:root:Vertical
DEBUG:root:e-Paper busy
INFO:root:4.read bmp file on window
DEBUG:root:Horizontal
DEBUG:root:e-Paper busy
INFO:root:Clear...
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy
INFO:root:Goto Sleep...
DEBUG:root:e-Paper busy
DEBUG:root:spi end
DEBUG:root:close 5V, Module enters 0 power consumption ....

I deleted display and ran make again - it created a new display however ./run.sh still does not produce a picture. If I look at the bmp or png the run command made it is fine.

I will buy a Pi ZeroWH but maybe this +3B can work - I seem so close - calendar and weather are clearly working? Thanks for the help.

feh123, 2020-12-04:
Hi @mendhak I have been looking at the demos again. I have tried to make epd in /home/pi/waveshare-epaper-sample/RaspberryPi&JetsonNano/c. The make routine worked without error but sudo ./epd did nothing. The epd file does not seem to execute anything.  I cannot find a bcm2835 demo nor a main.py - only in c/python/examples can I find the epd files that work.

I tried ./run.sh from within my WinSCP - it has a bit of debugging - not sure how reliable it is! The error that gave was 'Host not communicating try turning off optimise connection buffer size. So what happened was that WinSCP disconnected from the Pi when executing run.sh.

Why does epd not work at all? Thanks.

mendhak, 2020-12-04:
It seems like the Python scripts work fine for you but the C binaries don't.  It's this script here right? 
https://github.com/waveshare/e-Paper/blob/master/RaspberryPi&JetsonNano/python/examples/epd_7in5_V2_test.py

But if make ran fine for you I would have expected that to just work for you right away. I really don't know, no answers on what to do.  

The next thing that comes to mind is that you could modify one of those _working_ Python files and make it look at screen output bmp, then run the script.  The only thing I vaguely remember (this is 2 years ago now) is that it was quite slow.  But if this works - if you can get the Python script to write to the display - and if you can live with slow refresh, this could be the way to do it for you.  You'd have to make a copy of the Python script and remove all the unnecessary bits... just limit it to do one draw and exit... and test it out.  Finally in the run.sh, make the last line call your Python script instead of display.

feh123, 2020-12-04:
Hi @mendhak Yes that's the one that works. I much appreciate the time you have given to my problem. I will have a go at the hack you suggest while I wait for the RPi Zero. Thanks again.

feh123, 2020-12-06:
Hi @mendhak your idea worked. I moved my folder containing credentials, env.sh, run.sh and the png/bmp etc to the waveshare-epaper-samples tree that I created from the waveshare demo link on your website and put them in python/examples. There I renamed the working epd file and pointed it to my bmp. I deleted figlet display and added a cp .bmp to copy it to the pic folder because that is where the epd expects to find the bmp. Finally I just added the sudo ./epd that you suggested and it works. Just need to add the cron job.

However (sorry) but one last request - I have tried hacking your screen-calendar-get.py and adding extra days following the way the two days were formatted. According the output from putty on ./run.sh five days of my diary are downloaded but the .svg file only contains the two day version. What am I missing? Also what file can I hack to alter the image size/fonts, it's in portrait - can you swop to landscape? Thanks!

mendhak, 2020-12-06:
Glad to hear that's working - it's going to be a bit messy for you now but hopefully you can manage haha. 

OK now I'll tell you how to mess around with the display output.  

Suppose you've added another few of these blocks, obviousl as event_three, and so on. 

```
event_two = events[1]
start = event_two['start'].get('dateTime', event_two['start'].get('date'))
start = start[:10]
day_two = time.strftime("%a %b %d",time.strptime(start,"%Y-%m-%d"))
desc_two = event_two['summary']
print(day_two, desc_two)
```

The important bit is the last block of code here

```
output = codecs.open(template , 'r', encoding='utf-8').read()
output = output.replace('CAL_ONE',day_one)
output = output.replace('CAL_DESC_ONE',desc_one)
output = output.replace('CAL_TWO',day_two)
output = output.replace('CAL_DESC_TWO',desc_two)
```

Which is opening up the SVG file and performing text replacements.  If you open up the [SVG Template](https://github.com/mendhak/waveshare-epaper-display/blob/master/screen-template.svg?short_path=40a34ff) in a text editor like nano, you'll see it's a giant XML file.  And you can see in there, CAL_ONE, CAL_TWO.  That's what the code is replacing.  

If you want more days, you'll need to add more blocks, and just position them differently

```
   <text
     id="textprcblock1"
     x="330"
     y="400"
     font-size="29px"
     style="font-size:29px;line-height:0%;font-family:'Nimbus Sans L';text-anchor:beginning">
     <tspan x="330" dy="25" style="font-size:23px;">CAL_TWO</tspan>
     <tspan style="font-weight:bold;" x="330" dy="25">CAL_DESC_TWO</tspan>
   </text>

```

Change the `x` and `y` in the `<text` element.  You can start by adding your additional ones, then position them slightly - first ensure your script can replace them properly, then through trial and error position them nicely.

mendhak, 2020-12-06:
Didn't understand your last question, it should already be in landscape orientation.  

But if you need to rotate it, you can use the rotate command, it will have been installed as part of imagemagick.

    convert your-image.bmp -rotate 90 your-image-output.bmp

feh123, 2020-12-06:
Hi @mendhak thanks very much for such a detailed and fast response. I will the try - I did not realise the svg file was so complex (looks like a file designed for typos!). My simple editor just opened a picture. It's odd about it being portrait - do I add convert your-image etc to run.sh? 

Thanks a lot!

mendhak, 2020-12-06:
You could add your convert line in the [run.sh](https://github.com/mendhak/waveshare-epaper-display/blob/master/run.sh#L17) right after the inkscape command.

feh123, 2020-12-08:
Hi @mendhak - success - the screen has been running for two days on a cron job displaying weather and seven diary entries. I don't have PiHole so the extra space allowed more entries, with some font size reductions. I could get more text on each entry if I could remove the calendar icon but so far that has defeated me. I much enjoyed messing with the XML!

The calendar looks really great.

I think I run it differently to you - the epd.py file I use has a time.sleep function that automatically restarts the epaper. Normally its set to 10 sec. I set this to 3550 sec and my cron to update every hour at 1 minute past. So my epaper actually powers down and restarts and in this time lag (about 1 minute) I use the png file rather than the bmp. I guess the +3B helps. It means the clock is just showing when it last updated - I have a lot of clocks anyway.

Thanks for creating such a nice project and giving me so much help.

closed 2020-12-12 by feh123

mendhak, 2020-12-12:
Good to hear you got it working, happy future hacking!

feh123, 2020-12-12:
Hi - thanks! Strange - I just sent you another comment re hacking! The last one I promise!

esprittordu, 2021-02-15:
This thread was very helpful, I actually got something on my display with this. Not the correct date, and the calendar entries aren't populated, but a good start.