mendhak/waveshare-epaper-display #6: Inkspace error in export

jobreu, 2020-12-12:
When I run ./run.sh, the Python scripts seem to work fine, but in the Export part I receive the following Inkspace error:

Failed to get connection
** (inkscape:954): CRITICAL **: 16:23:29.383: dbus_g_proxy_new_for_name: assertion 'connection != NULL' failed

** (inkscape:954): CRITICAL **: 16:23:29.385: dbus_g_proxy_call: assertion 'DBUS_IS_G_PROXY (proxy)' failed

** (inkscape:954): CRITICAL **: 16:23:29.387: dbus_g_connection_register_g_object: assertion 'connection != NULL' failed
DPI: 150
Background RRGGBBAA: ffffff00
Area 0:0:800:480 exported to 640 x 384 pixels (76.8 dpi)

** (inkscape:954): WARNING **: 16:23:35.684: Bitmap failed to save to: screen-output.png
convert-im6.q16: unable to open image `screen-output.png': No such file or directory @ error/blob.c/OpenBlob/2874.
convert-im6.q16: no images defined `screen-output.bmp' @ error/convert.c/ConvertImageCommand/3258.

A post on the ImageMagick forum (https://www.imagemagick.org/discourse-server/viewtopic.php?t=33851) says that "Failed to get connection" suggests the SVG contains a reference to a URL on the web, and this fails".

mendhak, 2020-12-12:
That warning you see, Failed to get connection, I get that too.  Here's my output.  

> Failed to get connection
> ** (inkscape:16401): CRITICAL **: 16:56:17.700: dbus_g_proxy_new_for_name: assertion 'connection != NULL' failed
> ** (inkscape:16401): CRITICAL **: 16:56:17.702: dbus_g_proxy_call: assertion 'DBUS_IS_G_PROXY (proxy)' failed
> ** (inkscape:16401): CRITICAL **: 16:56:17.704: dbus_g_connection_register_g_object: assertion 'connection != NULL' failed
> DPI: 150
Background RRGGBBAA: ffffff00
Area 0:0:800:480 exported to 640 x 384 pixels (76.8 dpi)
Bitmap saved as: screen-output.png

The main difference is I can see, "Bitmap saved as: screen-output.png" but you don't have that.  

Does the screen-output-weather.svg exist when this command runs? 

What happens when you try to run that command manually?

    inkscape  screen-output-weather.svg --without-gui -e screen-output.png -w640 -h384 --export-dpi=150

And what if you put sudo in front.  My  `inkscape --version` is "Inkscape 0.92.4 (5da689c313, 2019-01-14)"

feh123, 2020-12-12:
Hi @jobreu, @mendhak  I get this error too:

Failed to get connection
** (inkscape:22309): CRITICAL **: 17:01:05.558: dbus_g_proxy_new_for_name: assertion 'connection != NULL' failed

** (inkscape:22309): CRITICAL **: 17:01:05.565: dbus_g_proxy_call: assertion 'DBUS_IS_G_PROXY (proxy)' failed

** (inkscape:22309): CRITICAL **: 17:01:05.565: dbus_g_connection_register_g_object: assertion 'connection != NULL' failed

I did not copy it but like @mendhak I get bitmap saved. So it does not seem fatal but I am not sure what I lose because of the error.

I will try out @mendhak ideas too.

jobreu, 2020-12-12:
Thx a lot for the quick reply!

If I put sudo before the command "inkscape  screen-output-weather.svg --without-gui -e screen-output.png -w640 -h384 --export-dpi=150", the Bitmap is saved as screen-output.png

I also put sudo in front of the next command in run.sh: "convert -colors 2 +dither -type Bilevel -monochrome screen-output.png screen-output.bmp".

I still get the inkspace errors.

Under display it then prints 0 and "bcm2835 init success !!!"
However, nothing gets displayed on the screen.

mendhak, 2020-12-12:
The inkscape errors can be ignored.  I don't know why they're appearing but the script does still work. 
If those commands are working with sudo that sounds like a file permission issue.   

Do the waveshare samples work for you? See bottom of the README.

jobreu, 2020-12-12:
epd_7in5b_V2_test.py from the Waveshare examples works

mendhak, 2020-12-12:
What Raspberry Pi do you have?  I'm wondering if my compiled `display` only works on some Raspberry Pis.  The trouble is, it's really hard to test as I only have one 😁   You can try manually running `sudo display/display screen-output.bmp 1` (or 0 at the end).

jobreu, 2020-12-12:
It's a Raspberry Pi Zero WH.
If I run `sudo display/display screen-output.bmp 1`, I still get stuck at "bcm2835 init success !!!" and nothing shows up on the display (same if I replace 1 with 0 in the command).

feh123, 2020-12-12:
Hi @jobreu I sounds like my initial problem too. I saw the same bcm2835 init succsss too but no picture. With @mendhak's help I solved it but installing the waveshare demo that is on @mendhak website too and used the epd test.py from that. I then transferred all my files to the examples folder that the installation creates. running the epd from there it worked. I needed to delete display from run.sh and replace it with a sudo epd...py to run the python file directly. I did need to put a command in run.sh to cp the bmp made to the pic folder because that is where the demo version expects to find the file. Then it worked. Maybe there are better ways. 
Re the inkscape error I did not check closely enough but initially I did not get any error. I did see that with the error the weather data seen in terminal on running ./run.sh was not the same as the icon displayed on the screen. I am still xhecking that out.

mendhak, 2020-12-12:
Thanks for describing the workaround @feh123 

I'll tell you one more thing you can try.  Suppose the `sudo display/display screen-output.bmp 1` isn't working.  

Go into the display folder and recompile it.  

```
cd display
# Make a backup juuust in case
cp display display.backup
# Now remove the current display
rm display
# Now try to rebuild it
make
```

Just running the make command should be enough to recreate the `display` executable.   Then go back and try running the display/display ... again.

jobreu, 2020-12-12:
Recompiling the display folder did not solve the issue for me. I tried to follow the approach suggested by @feh123. However, I think I did something wrong in editing the epd test.py file.
Could you maybe share your code for that file and your version of the run.sh file?

jobreu, 2020-12-12:
Or maybe I can just share what I put in the edited run.sh and the python script. I probably made one or more obvious mistakes there...

Here are the contents of the run.sh:
`
. env.sh
figlet Weather info
sudo -E python3 screen-weather-get.py

current_hour=`date +"%H"`

figlet Pihole info
sudo -E python3 screen-pihole-get.py show

figlet Calendar info
sudo -E python3 screen-calendar-get.py

figlet Export
# Inkscape can't export to BMP, so let's export to PNG first.
sudo inkscape  screen-output-weather.svg --without-gui -e screen-output.png -w640 -h384 --export-dpi=150

# Convert to a black and white, 1 bit bitmap
sudo convert -colors 2 +dither -type Bilevel -monochrome screen-output.png screen-output.bmp
sudo cp ./screen-output.bmp ../pic

SHOULD_REFRESH=0
current_minute=`date +"%M"`

if [ $current_minute -eq 0 ] ; then
   SHOULD_REFRESH=1
fi

figlet Display
sudo python3 display_test.py $SHOULD_REFRESH
`

And here is the Python script which I named display_test.py:
`
#!/usr/bin/python
# -*- coding:utf-8 -*-
import sys
import os
import time
libdir = os.path.join(os.path.dirname(os.path.dirname(os.path.realpath(__file__))), 'lib')
if os.path.exists(libdir):
    sys.path.append(libdir)

from waveshare_epd import epd7in5_V2
import time
from PIL import Image,ImageDraw,ImageFont

try:
    epd = epd7in5_V2.EPD()
    epd.init()
    Himage = Image.open('../pic/screen-output.bmp')
    epd.display(epd.getbuffer(Himage))
    epd.sleep()
except KeyboardInterrupt:
    epd7in5_V2.epdconfig.module_exit()
    exit()
`

feh123, 2020-12-13:
Hi @jobreu here's my run.sh

. env.sh
figlet Weather info
sudo -E python3 screen-weather-get.py

current_hour=`date +"%H"`

figlet Calendar info
sudo -E python3 screen-calendar-get.py

figlet Export
# Inkscape can't export to BMP, so let's export to PNG first. 
inkscape  screen-output-weather.svg --without-gui -e screen-output.png -w640 -h384 --export-dpi=150

# Convert to a black and white, 1 bit bitmap
convert -colors 2 +dither -type Bilevel -monochrome screen-output.png screen-output.bmp
convert screen-output.bmp -rotate 90 screen-output.bmp
convert screen-output.png -rotate 90 screen-output.png

# Moving bmp

cp screen-output.png /home/pi/waveshare-epaper-sample/'RaspberryPi&JetsonNano'/python/pic

SHOULD_REFRESH=0
current_minute=`date +"%M"`

if [ $current_minute -eq 0 ] ; then
   SHOULD_REFRESH=1
fi

sudo ./epd_screen_output.py $SHOULD_REFRESH

I have to rotate the image (thanks to @mendhak for the code) as my picture was protrait. I have to move it to the pic file bacause that is where wavepaper defined it to be. I only move the png now but I did move both. Finally I delete display and replace with the sudo code.

To make the examples folder a bit cleaner I moved all the other py files to a temp folder I created. I remained the one I used and copied all my files (from waveshare-epaper-diplay) to this directory. This idea of using the demo was @mendhak's (thanks!).

My epd_screen_output.py that I created from the examples folder py files is:

#!/usr/bin/python
# -*- coding:utf-8 -*-
import sys
import os
picdir = os.path.join(os.path.dirname(os.path.dirname(os.path.realpath(__file__))), 'pic')
libdir = os.path.join(os.path.dirname(os.path.dirname(os.path.realpath(__file__))), 'lib')
if os.path.exists(libdir):
    sys.path.append(libdir)

import logging
from waveshare_epd import epd7in5_V2
import time
from PIL import Image,ImageDraw,ImageFont
import traceback

logging.basicConfig(level=logging.DEBUG)

try:
    logging.info("epd_screen-output.py")

    epd = epd7in5_V2.EPD()
    logging.info("init and Clear")
    epd.init()
    epd.Clear()

    font18 = ImageFont.truetype(os.path.join(picdir, 'Font.ttc'), 18)
    font12 = ImageFont.truetype(os.path.join(picdir, 'Font.ttc'), 12)

    logging.info("1.read png file on window")
    Himage2 = Image.new('1', (epd.height, epd.width), 255)  # 255: clear the frame
    bmp = Image.open(os.path.join(picdir, 'screen-output.png'))
    Himage2.paste(bmp, (50,10))
    epd.display(epd.getbuffer(Himage2))
    time.sleep(3550)

    logging.info("Clear...")
    epd.init()
    epd.Clear()

    logging.info("Goto Sleep...")
    epd.sleep()
    epd.Dev_exit()
    
except IOError as e:
    logging.info(e)
    
except KeyboardInterrupt:    
    logging.info("ctrl + c:")
    epd7in5.epdconfig.module_exit()
    exit()

It has a time.sleep function that I set to 3550 - so it powers down and then run.sh starts it via a cron job that runs very hour. This wrecks the clock of course but I have lots of clocks, 

I can send you more files if you need. I could have tried to download a waveshare master git but I have not done.

jobreu, 2020-12-13:
Thx a lot! Got it to work with your solution @feh123.
Does anything speak against reducing the time in the time.sleep function and increasing the frequency of the CRON job? If I, e.g., let it run every 3 or 5 minutes, the clock would only be off by a few minutes.

feh123, 2020-12-13:
Hi @jobreu no I think you could try changing the times of both. You could also delete the time.sleep code too I guess and only use the cron. Once it was working I left it as is so I have never investigated! 

Glad it's going - I should say that @mendhak gave me the outline of what I needed to do to get it working (and the code for my run.sh) - without that I could not have done it.

mendhak, 2020-12-13:
I think I know why the original display isn't working - you're both on V2 of the displays!  
I'm on the 'original' and I noticed in the examples there's a separate example script for the V2 displays vs the originals.  

I don't have a V2 display so I really can't test this, but I've gotten the Python script displaying the image in about 35 seconds for my V1 version.  I'm applying some of those tweaks here to your version @feh123.  This displays the image and then exits right away.  This should help you @jobreu if you want more frequent updates.  

```
#!/usr/bin/python
# -*- coding:utf-8 -*-
import sys
import os
picdir = os.path.join(os.path.dirname(os.path.dirname(os.path.realpath(file))), 'pic')
libdir = os.path.join(os.path.dirname(os.path.dirname(os.path.realpath(file))), 'lib')
if os.path.exists(libdir):
    sys.path.append(libdir)

import logging
from waveshare_epd import epd7in5b_V2
import time, datetime
from PIL import Image,ImageDraw,ImageFont
import traceback

logging.basicConfig(level=logging.DEBUG)

try:
    logging.info("epd7in5b_V2 Demo")

    epd = epd7in5b_V2.EPD()
    logging.info("init and Clear")
    epd.init()

    if datetime.datetime.now().minute==0:
        epd.Clear()

    logging.info("3.read bmp file")
    Himage = Image.open(os.path.join(picdir, 'screen-output.png'))
    epd.display(epd.getbuffer(Himage))
    epd.sleep()
    epd.Dev_exit()

except IOError as e:
    logging.info(e)

except KeyboardInterrupt:
    logging.info("ctrl + c:")
    epd7in5b_V2.epdconfig.module_exit()
    exit()
```

A few differences to note.  

* I'm only doing an epd.Clear() on the hour, not each time.  
* I'm reading the image directly rather than using the .paste() function.  
* No time.sleep, just epd.sleep and exit.  The time sleep was only for sample purposes.

feh123, 2020-12-13:
Hi @mendhak thanks for this. So now the image remains until the cron updates? How often do you update with cron?
It isn't a typo - I used the png because up to now I only have my cron updating every hour. I think my +3B takes about the same time to load the png (around 20-30s). I may flip back to the bmp and update at your rate.

mendhak, 2020-12-13:
Yeah the image will remain until the cron runs again.  I've got the cron running once a minute. 

    * * * * * cd /home/pi/waveshare-epaper-display && bash run.sh > ./run.log 2>&1

feh123, 2020-12-13:
Thanks - I am just trying your code for the weather descriptor. It should run in about 10 minutes. I will let you know how it goes!

jobreu, 2020-12-13:
Thx again for all of your suggestions! While trying out a few of the things you suggested, my display "got stuck". No matter what Python script I run now, I always get the debug logging info "e-Paper busy" and the image (from the bmp) is frozen on the screen (even after powering off the Raspberry Pi). Any idea how I can fix this (i.e., reset/clear the screen)? My Google search for the issue turned up no solution (but maybe my search was just not optimal).

mendhak, 2020-12-13:
Just restarting should have been enough, that should reset the pins. 

You could try to disconnect it from the rpi maybe, I'm not sure that'll help though. You could also try running one of the epaper samples again.

jobreu, 2020-12-13:
Thx a lot! I will try those things out.

mendhak, 2021-01-31:
@jobreu  did you find out what's wrong, was it your hardware? There was another user with a similar problem: https://github.com/mendhak/waveshare-epaper-display/issues/7#issuecomment-758883045  and I directed them to [this thread](https://github.com/waveshare/e-Paper/issues/92).

lifeofbrian, 2021-02-01:
For what it's worth -- I didn't see this with my integration/developer effort over the weekend with a V2 connected to a Raspberry Pi 4.

jobreu, 2021-02-01:
@mendhak Thanks for the link! TBH, as I did not find a solution after several attempts, I have stopped trying. I will check out the thread and maybe also try it with a Pi 4 instead of a Zero as suggested by @lifeofbrian.

closed 2021-02-04 by mendhak