mendhak/waveshare-epaper-display #7: Error running "run.sh"

pdavid-muc, 2020-12-21:
I'd like to built a screen displaying current or future meetings, i don't need the PiHole or Weather integration.
I followed your instructions, apart from installing python-imaging (which was not available), i installed "python-pil" (apt recommended that).

In Google, i enabled the Google Calendar API and instead for "Desktop App" i went for "TVs and limited input devices" (does that make a difference? Selecting desktop app did not work for me). 
After using 'curl' for the localhost-URL, this is was the script outputs:

`ickle is stale, calling the Calendar API
(calendar entry of the day)
(next calendar entry)
Traceback (most recent call last):
  File "screen-calendar-get.py", line 80, in <module>
    output = codecs.open(template , 'r', encoding='utf-8').read()
  File "/usr/lib/python3.7/codecs.py", line 898, in open
    file = builtins.open(filename, mode, buffering)
FileNotFoundError: [Errno 2] No such file or directory: 'screen-output-weather.svg'`

Did i miss something?
I have a raspberry pi zero WH, the Display you've mentioned and accessories.

pdavid-muc, 2020-12-21:
Weid. I cannot run examples:

>~/waveshare-epaper-sample/RaspberryPi_JetsonNano/python/examples $ sudo python3 epd_7in5_V2_test.py
>INFO:root:epd7in5_V2 Demo
>INFO:root:init and Clear

The wifi connection drops then.. there is nothing displayed.

mendhak, 2020-12-21:
The first problem is easy enough - it happens because the screen-weather-get script is taking `screen-template.svg` and renaming it to `screen-output-weather.svg` and some of the subsequent scripts are expecting that svg to exist.  You could try making a copy of the screen-template.svg and call it `screen-output-weather.svg` so that the calendar script can find the file.

mendhak, 2020-12-21:
I think try to get the samples working first.  If those don't work, this project won't.  

You'll later run into another problem - which is this project was written for the v1 screens and you probably have the v2 screens.  I'm just working on a script for that right now that might help.  

But in the meantime try to get the demo working.

pdavid-muc, 2020-12-22:
Thank you for your reply. Yes, i have the V2 screen. Trying to run the examples results in a device reboot. I'm running buster liste, does that make a difference?

mendhak, 2020-12-22:
I really don't know why even the samples would result in a reboot, you'll have to troubleshoot this.  It could be not enough power being supplied (try a higher Amp charger, power from mains instead of computer, something like that), it could be the OS (sure you can try Raspbian Stretch), just to eliminate variables.

feh123, 2020-12-22:
Hi @pdavid-muc I used the Raspberry Pi Imager and downloaded the recommended version. I followed @mendhak's advice re using the demo. It worked almost straightaway - I never saw a reboot error.

pdavid-muc, 2021-01-12:
I have really no idea what i am doing wrong. Tried different power source (2.4A). Did exactly what is described in the readme. When i try to run the demo:
```
~/e-Paper/RaspberryPi_JetsonNano/python $ sudo python3 examples/epd_7in5_V2_test.py
INFO:root:epd7in5_V2 Demo
INFO:root:init and Clear
```

The raspberry pi just restarts.

mendhak, 2021-01-12:
@pdavid-muc  I see an issue similar to yours on the waveshare repo, have a look :  https://github.com/waveshare/e-Paper/issues/92

It also looks like it's worth emailing Waveshare directly since what you're experiencing doesn't seem to be normal - their screen with their own sample code isn't working.

pdavid-muc, 2021-01-18:
I'll have a look there - thank you very much!

closed 2021-01-18 by pdavid-muc