mendhak/waveshare-epaper-display #8: Climacell V3 ends July 2021

feh123, 2020-12-28:
Hi @mendhak I just received this update: https://developer.climacell.co/v3/docs/deprecation-notice. It sounds more than just a new API key. These weather apps seem to constantly change!

mendhak, 2020-12-29:
Thanks for posting it, I didn't get any email from them. 

So it's a bit weird, right now their [deprecation notice page](https://developer.climacell.co/v3/docs/deprecation-notice) links to a migration guide but clicking it results in a 404. 

I also went to their [main page](https://developer.climacell.co/) but another broken link there too. 

Looks like they've moved everything over to https://docs.climacell.co but didn't test their own links properly.  I'll go through this link soon to see what's involved.

feh123, 2020-12-30:
Hi @mendhak  I use V3 in another project. We had the same problem you faced - Dark Sky was no longer taking new subscribers. Luckily I have used Dark Sky so I get until end 2021. However we added V3 as an alternative. V4 was checked out too. It's through that I saw the news - nothing directly from Climacell. V3 seems fine but V4 is definitely buggy - they do say it's not ready yet. Good luck with the coding! The epaper-display is running really well.

mattmb, 2021-01-24:
I might PR if I have time but incase I forget I was working on a project largely inspired by this one and I had to update to v4 to make the weather work, the following might help. Are there any wind icons? I've mapped them to clear for now.

```
icon_dict={
    6201: 'freezing_rain',
    6001: 'freezing_rain',
    6200: 'freezing_rain' ,
    6000: 'freezing_rain',
    7101: 'ice_pellets',
    7000: 'ice_pellets',
    7102: 'rain_icepellets_mix',
    5101: 'snow',
    5000: 'snow',
    5100: 'rain_snow_mix',
    5001: 'blizzard',
    8000: 'thundershower_rain',
    4201: 'rain_day',
    4001: 'rain_day',
    4200: 'rain_day',
    4000: 'rain_day',
    2100: 'scattered_clouds_fog',
    2000: 'foggy',
    1001: 'few_clouds',
    1102: 'mostly_cloudy',
    1101: 'few_clouds',
    1100: 'clear_sky_day',
    1000: 'clear_sky_day',
    3000: 'clear_sky_day',
    3001: 'clear_sky_day',
    3002: 'clear_sky_day',
}
```

```
if dt > s["sunset"] or dt < s["sunrise"]:
    icon_dict[1000]="clearnight"
    icon_dict[1101]="partlycloudynight"
    icon_dict[1001]="partlycloudynight"
    icon_dict[4001]="rain_night"
    icon_dict[4200]="rain_night"
    icon_dict[4201]="rain_night"
    icon_dict[4000]="rain_night"
```

```
url = f"https://data.climacell.co/v4/timelines?location={lat}%2C{lon}&fields=temperatureMin%2CtemperatureMax%2CweatherCode&timesteps=1d&apikey={API_KEY}"
data = requests.get(url).json()['data']['timelines'][0]['intervals'][0]
output = {'min': data['values']['temperatureMin'], 'max': data['values']['temperatureMax'], 'icon': icon_dict[data['values']['weatherCode']]}
```

lifeofbrian, 2021-01-31:
I'm going to go ahead and submit a pull request including @mattmb's suggestion and a few other tweaks I've made to best support v4.

lifeofbrian, 2021-01-31:
Also, will be adding an env parameter for format (e.g. CELSIUS or FARENHEIT) for those of us that use Farenheit.

mendhak, 2021-01-31:
That sounds good.  Please default it to Celsius. 

I haven't done any work on this yet, so it'll be helpful if you can move it to v4.  But if you can't find time, that's OK, I'll try to find some time soon. 

After that I'll also find some icons for the windy and any others.  I see a list of codes here: https://docs.climacell.co/reference/data-layers-core

lifeofbrian, 2021-02-01:
Yep - defaulting to Celsius.

lifeofbrian, 2021-02-01:
Quick follow-up:   Last night, I submitted a large pull request that uses guidance that @mattmb provided earlier on the thread here.

mendhak, 2021-02-04:
Thanks everyone, @lifeofbrian PR worked for me, and I found an old wind icon from the DarkSky days in the SVG so just repurposed that.  I've merged the pull request in now so Climacell V4 is in use.

feh123, 2021-02-05:
Thanks a lot @mendhak. I guess if I keep copies of my files - run.sh, weather-get.py, calendar-get.py and my svg I can add back my specific preferences I created after installing the new version?

lifeofbrian, 2021-02-05:
@mendhak - nice. good call on repurposing that icon.

lifeofbrian, 2021-02-05:
@feh123 - that sounds arduous. What are your 'specific preferences'? We may be able to better accommodate that through configuration.

feh123, 2021-02-06:
Hi @lifeofbrian thanks a lot for the offer. 
My setup is based on the workaround that @mendhak suggested to me for the V1 program on a V2 epaper screen. 
I only activate, in env.sh, climacell and google calendar and I have removed PiHole completely from my screen-template.svg file. I do not use train times.
My svg file has been re-designed to take seven diary entries and my own description of the weather icons (thanks to @mendhak again). I have altered the svg file via inkscape and the xml file (to define the positions). I altered the screen size here to give me more space too. 
In run.sh I use sudo ./epd_screen_output.py and not display. In run.sh I have to flip the image to landscape (don't know why) and move it to the examples folder (where the examples program requires it to be) as part of the workaround. Finally I display the png (I still create the bmp) as I prefer the better image resolution. 
So even though I have a +3B RPi this display takes at least 30 secs. However I do not need the clock so I have a cron job running at 60 minutes. I still display the clock - it does show the system is working and it's right every hour. I alter the refresh times in epd_screen_output.py so that the diary/calendar are always out of date for a 60 minute cron job.
I have not looked at how the new program works in comparison. 
I was thinking of,just installing the new software on a different sd card from scratch and adding back my modifications as best I could. Keeping my old system ready to re-use if things went badly wrong. But if you have any thoughts from the above on how to adapt the current software (and maybe help others who want to make their own hacks) then that would be great. 
But my edits are quite numerous - so I can fully understand if it's best for me just to start from a fresh install - thanks anyway.

mendhak, 2021-02-06:
Yeah that does sound like a lot.  What you could do is visually inspect the changes made by lifeofbrian [here](https://github.com/mendhak/waveshare-epaper-display/pull/11/files).  And apply it to your own changes.  The problem there may be if you put a change in the wrong place or there's some typo, it'll be hard for you to recover, so definitely make a backup of what you already have.   

It might also help you longer term if you pushed your changes up to your github account - minus the env.sh, apiresponse, pickle files, etc - so that you've got a running 'commentary' and backup of what you're doing.

For what it's worth I'm slowly working on modifying the layout a bit, I'm removing the train and pihole sections. I'll move the calendar portion 'up' a bit and add a few more entries (but not 7!), and maybe few other adjustments.  I'll post updates in the [pihole issue](https://github.com/mendhak/waveshare-epaper-display/issues/10).

feh123, 2021-02-06:
Hi @mendhak. Thanks for the ideas - I too thought it rather a lot. I will look at the changes - I am waiting to get more sd cards so I can keep a working card safe.
If you are doing some calendar mods - is it possible to get the diary entry time added as well as the date? Currently I add the time to the description line - not too difficult but many events are added automatically and they all need to be edited.
I will keep checking the pihole issue page. Good luck.

closed 2021-02-07 by mendhak