mendhak/waveshare-epaper-display #10: I want to remove the PiHole section

mendhak, 2021-01-31:
Out of everything on the dashboard, the only thing I don't look at is the PiHole section.  I would like to remove it. 

I also noticed this from other issues and other users like @feh123 who have modified the SVG to show more calendar entries, that the PiHole section is not that useful.  

Considering the tasks for this: 

- [ ] Remove pihole from SVG
- [ ] Make the calendar section taller OR rearrange the dashboard: For example, time + date together, bottom left section for weather and weather description. 
- [ ] Remove pihole section from run.sh
- [ ] Modify the README and code.mendhak.com article to remove references to PiHole
- [ ] Modify the env.sh, it doesn't need PiHole address anymore
- [ ] Take a new photo

lifeofbrian, 2021-02-01:
I agree - in my local branch, I have commented out that code.

What are your thoughts on a LayoutManager / Widget framework for these modules? This might afford more flexibility for varying use cases (e.g. a "Countdown to X event", Sports scores, breaking news, etc.) and the ability to alternate layouts at intervals. 

If interested, I may take a stab at it this weekend.

mendhak, 2021-02-04:
Hmm interesting idea  but to be honest not very interested in that, the current setup is hacky but is seldom changed, so I'm happy to manually modify it for now.

lifeofbrian, 2021-02-05:
Sounds good -- appreciate the feedback. 

I still may give it a try later this month. I like the idea of a flexible layout for a couple of these e-Paper dashboards around the house.

mendhak, 2021-02-06:
@feh123 

Just to update, I'm still playing around.  I was able to add the time to the date line (it stays empty for all day events).  Anything beyond that was too crowded and unable to pass the Significant Other test. 

[image]

feh123, 2021-02-06:
Hi @mendhak thanks looks very nice. Interesting that Climacell V4 gives a better weather forecast. My V3 justs says snow!

mendhak, 2021-02-06:
[image]

Added today's day name, and moved day name + date to top right area.  Moved weather icon a bit higher, giving it some space.

mendhak, 2021-02-06:
[image]

Added weather description.

mendhak, 2021-02-06:
[image]

It feels like the space is being better used now. 

I've also been rendering the PNG directly rather than converting to BMP, because the time it takes to draw them on screen is the same.  Visually the PNG one looks good, but when I look very closely I see all this jagged edges, I don't know what that is (but I suspect nothing much I can do about it)

mendhak, 2021-02-06:
Climacell provide some weather icons corresponding to most (not all) of their weather codes: https://github.com/climacell-api/weather-code-icons

I've added some and referenced them in the icon dictionary, and attributed them in the readme. 

The icon for light freezing rain (my current weather) is a bit odd, but I'll get used to it. https://github.com/ClimaCell-API/weather-code-icons/blob/master/black/freezing_rain_light.svg

mendhak, 2021-02-07:
I think it's done, planning on merging the changes today into the master brnach. 

[image]

[image]

mendhak, 2021-02-07:
Merged into master.  

- The latest version now only does date, time, weather, calendar. No more pihole or trains. 
- 4 calendar entries instead of 2
- Calendar entries have a time next to the date
- The log output is a bit neater now, no more figlet. 
- Rearranged the dashboard so that the date/time are at the top right
- Weather icon slightly refreshed, used some new ones from Climacell, and added weather description
- Also I'm not converting to BMP anymore since the time it takes to draw on screen is the same using the Python script, so now the output is PNG drawn to screen.  I was previously converting to BMP which was fast when combined with the precompiled 'display' binary.

closed 2021-02-07 by mendhak