mendhak/waveshare-epaper-display #13: Severe weather alert API needed

mendhak, 2021-02-07:
Previously the DarkSky API used to provide an alert as part of its API response.  The dashboard template has a section for an ALERT_MESSAGE right at the top for severe weather alerts.  I can't seem to find a good API endpoint or API service to provide severe weather alerts.  

Currently (Feb 7 2021) in London there is a Yellow Alert for snow, and in Scotland there is an Amber alert

The Met Office feed has the alert present

[image]

The Climacell API is a confusing mess, it took me a very long time to figure out the call. But it has no alerts present.  

    https://data.climacell.co/v4/events?location=XX,YY&insights=winter&insights=floods&insights=temperature&insights=fog&insights=tornado&insights=air&insights=fires&insights=wind&buffer=1&apikey=XXXXXXXXXXXXXXXXXXX

[image]

I tried the OpenWeatherMap API, no alerts section is present. 

    https://api.openweathermap.org/data/2.5/onecall?unites=metric&exclude=minutely,hourly,current&lat=XX&lon=YY&appid=XXXXXXXX

[image]

I tried the WeatherBit API, also nothing. 

    https://api.weatherbit.io/v2.0/alerts?lat=XX&lon=YY&key=xxxxxxxxxxxxx

[image]

mendhak, 2021-07-29:
In the branch [alertmessage](https://github.com/mendhak/waveshare-epaper-display/tree/alertmessage) I have implemented a Met Office Alert.  

I need to find a few more Alert APIs or feeds, it's not easy to find those.  I found one from AccuWeather but it's not part of the trial package.

mendhak, 2021-08-09:
I got a severe weather warning today, based on the Met Office feed.  It cuts off since the line is too long. 

[image]

feh123, 2021-08-09:
Hi @mendhak this looks really nice. Do I need to re-install the software to get the alerts? Thanks.

mendhak, 2021-08-09:
Hey @feh123  I've just merged it into the master branch.  So yes you'd need to update to the latest version. 

Notice this new line in the env.sh:

https://github.com/mendhak/waveshare-epaper-display/blob/e95cd1faf2e7dbf2d5d5ba15c4a71bf6622486f0/env.sh.sample#L38

You need to add that to your env.sh (uncommented), and change the URL to match one for your region.  Get the URL from this page:  https://www.metoffice.gov.uk/weather/guides/rss

[image]

Do let me know whether it works or not.  Or if you know of any other severe alert providers.  They're really hard to find.

jmason, 2021-08-10:
this is awesome. I've been meaning to add a weather provider for Met Eireann (the Irish weather service) and I think they offer a severe weather alert API too.....

mendhak, 2021-08-17:
@jmason  I had a look at Met Eireann APIs out of curiosity.  Their license specifically says that if you use their weather forecast API, you also need to show their weather warnings.  So it's basically a 2-for-1.  

The Weather Forecast API is:  http://metwdb-openaccess.ichec.ie/metno-wdb2ts/locationforecast?lat=51.8959843;long=-8.533089

It appears the forecasts are hourly, so there would be a bit of work to go over the next _x_ hours of forecasts and pick the high and low values.  What is not clear is a good way to pick the right weather symbol to use.  

The Warnings Feed is:  https://www.met.ie/warningsxml/rss.xml

All the warnings for the nation appears in there, which means that the code would need to filter by searching for a user-provided keyword in the title.

jmason, 2021-08-18:
also ideally I could get it to ignore the "Blight Advisories" as I'm not a potato farmer :)

mendhak, 2022-03-06:
Found an alert provider for the US.  api.weather.gov can provide alerts for a given lat long. 

https://api.weather.gov/alerts?point=41.7456,-120.0892

[apiweathergov_example_alert_1.json.txt](https://github.com/mendhak/waveshare-epaper-display/files/8192449/apiweathergov_example_alert_1.json.txt)

Within this output the first `NWSheadline` property seems to be the most relevant headline to present.  

Weather.gov also does forecasts.  How did I miss this?  

The user (or code) would need to get their grid points by first looking at the lat/long endpoint: https://api.weather.gov/points/39.7456,-97.0892

Then from there, the forecast:  https://api.weather.gov/gridpoints/TOP/31,80/forecast

The output also includes a `shortForecast` which has a nice string to display.

jmason, 2022-03-28:
as noted in https://github.com/mendhak/waveshare-epaper-display/pull/34#issuecomment-1080465225 , I've the start of support for Met Eireann's severe weather alert API (which is required to use their weather data feed).

jmason, 2022-04-06:
FYI, I have rewritten the Met Eireann severe weather alert support -- rather than using the country-wide XML file, there's another, more succinct JSON API which can be targeted down to the county level.  Example output:

mendhak, 2022-04-07:
Hey @jmason could you link to the JSON API which targets county level?  

As for the conditional alertbox... so I assume you mean if there's an alert, it should push the other contents a little bit higher?
One way could be by adding a `display="none"` attribute to the box, which hides it by default, then some extra Python code to see whether an alert was returned from Met Eireann, and set `display="block"` so that it is rendered.

jmason, 2022-04-08:
> Hey @jmason could you link to the JSON API which targets county level?

Better than that, here's the PR :)  https://github.com/mendhak/waveshare-epaper-display/pull/35

BTW just for reference, here's a diff against my main branch - https://github.com/mendhak/waveshare-epaper-display/compare/master...jmason:prod?expand=1 -- there's a few other "improvements" in there which I haven't yet got around to tidying up into useful PRs yet but feel free to take a look :)

> As for the conditional alertbox... so I assume you mean if there's an alert, it should push the other contents a little bit higher? One way could be by adding a `display="none"` attribute to the [...]

Yeah something like that.  One idea I had was to have a kind of "popup" box look and feel for the alert, so if there is an alert in progress, it could appear as a boxed chunk of text, covering the bottom of the display -- although I have no idea how feasible that is in SVG.  here's a mockup:

[image]

mendhak, 2022-04-10:
Goodness me you won't believe how hard it is to get a box around some text in SVG! So there isn't really a concept like HTML where a container box can expand along with its inner contents.  Or at least, not one I could find easily, what little I could find often involved JavaScript. 

The best way I could do it was by drawing a `<rect>` with borders around it, taking up the entire bottom space. Then adding the text in that area.  And then I [conditionally show](https://github.com/mendhak/waveshare-epaper-display/blob/layout_refactor/screen-weather-get.py#L164) the box if an alert message is present. 

[image]

mendhak, 2022-04-10:
Gave it a bit more space.  

[image]