mendhak/waveshare-epaper-display #21: Suddenly not working.

mrdanjohnson, 2021-02-18:
Im not sure if this is where I can post this silly question, but I had this working the other day. Now when I run the run.sh script I get;

File "run.sh", line 2
    . env.sh
    ^
SyntaxError: invalid syntax

My env.sh file seems totally fine.

mendhak, 2021-02-18:
If you try that yourself outside run.sh, does it still fall?

It could indicate bad syntax inside env.sh

You can test each line in env by copy pasting and running it. Make sure there are no spaces between the key, equal and values in the env

mrdanjohnson, 2021-02-18:
Even when using the env.sh directly from your repo, I still get the error. It just started happening yesterday. I can run waveshare's example files without error.

mendhak, 2021-02-18:
Something doesn't look right.  See,

[image]

Bash doesn't produce errors like this.  How are you running run.sh?  It should just be `./run.sh`

mendhak, 2021-02-18:
Ah are you running it as a Python script like `python3 run.sh`?  I just tried it arbitrarily and got the same 'error'

[image]

Since it's a `.sh` you should just run it as a normal bash script.  Just do `./run.sh`

mrdanjohnson, 2021-02-18:
That totally worked. Did something change or am I loosing my mind. Has run always been a shell script?

While I have you here. Is there a way to keep it from flashing so many times when clearing or writing. I was expecting it to work more like a kindle.

mendhak, 2021-02-18:
It's always been a shell script!  

Right, the image flashes several times when rendering to screen.  The Waveshare in general isn't as fast as the Kindles and other more commercial screens (and they're also different technologies, eink is a specific brand, it's a commercial patent encumbered technology, epaper is a generic term). You'll have noticed that the Waveshare Python samples are slow to render stuff to screen, I think about 20-30s. 

But if you've tried the C samples they're a bit faster, although still not Kindle speed, I think it takes about 8s.  So a way to write to screen faster would be to write a C version of the [display script](https://github.com/mendhak/waveshare-epaper-display/blob/master/display.py).  I do have a [C version from a while ago](https://github.com/mendhak/waveshare-epaper-display/blob/8811525a0223120507472a891addd965d4418c25/display/display), but it quickly became clear that it wasn't working for most people who were getting the V2 screens.  I don't have a V2 screen so I've had to switch to a basic Python script.   

I also used to have a project ages ago which basically used a Kindle screen to display the dashboard:  https://github.com/mendhak/Kindle-Time-and-Weather - it was definitely very fast.  Only problem, the Kindle's wifi radio burnt out after a year or two, I guess it's not meant to do very frequent http requests.

mrdanjohnson, 2021-02-20:
Wow, I have no idea where my mind went. I have no idea how I moved from working with the shell script and suddenly switching to python. I guess because I was jumping back and forth between waveshare and your project.  I actually had bought their 5" display for a project that I no longer needed it for, and got it working with your script. I really love your project, and even my 10 year old is pretty impressed. She's been learning coding and thinks she's pretty smart.

Thanks for all your hard work, and for responding to my dumb question.

closed 2021-02-20 by mrdanjohnson