mendhak/waveshare-epaper-display #24: Accuweather error service unavailable number of requests exceeded

feh123, 2021-04-19:
Hi @mendhak I noticed this error when I first set up accuweather and I thought it was because in messing around setting up the software I had manually started it with ./run.sh many times and had probably hit the 50 limit. I had also reduced the caching time (not a good idea!). However I found what I thought was a way round it by edting and re-saving my accuweather app and deleting weather-cache.json.
How it's happened again - yesterday at 9:00 pm it stopped getting weather updates - this stops all updates so the png shows no update to time or calendar as well.
Run.log: [image].
I have copies of the analytics from the accuweaher app portal but they do not seem to record the number of calls made?
As of this morning my attempts to fix this by editing the app have failed and I get the same error. Incidentally my caching time is 3500 secs - is this correct? Any thoughts appreciated!

feh123, 2021-04-19:
Hi @mendhak Just a quick update - my workaround was successful. Maybe it took some time for the app update to filter through the servers but the Pi updated on my cronjob at 9:01 am (or maybe of course the 24 hrs is up and a new 24 hrs begins?). Anyway I guess my concern now is it will happen again at 9:01 pm.

mendhak, 2021-04-19:
OK I've [just pushed a bit of code](https://github.com/mendhak/waveshare-epaper-display/commit/3d9eead29e92c249f95032028617d0f4cef94fb2) that can help you figure this out.  It'll enable debug logging down to the python requests level.  

You'll need to get latest, then in your `env.sh` set `LOG_LEVEL=DEBUG`
Then remove weather-cache.json so that the code is forced to call Accuweather again. 
Then, run the run.sh again. 

This time, in the run.log you'll see the call being made to Accuweather, and right after that a bunch of headers get printed.  And among those headers, you'll see something like this:

> header: RateLimit-Limit: 50
> header: RateLimit-Remaining: 43

Accuweather returns the remaining rate limit in its response headers so I'm logging that to the run.log file now.  

---

The caching time should be 3600, you can see the default in env.sh.sample:

```
# How long, in seconds, to cache weather for
export WEATHER_TTL=3600
# How long, in seconds, to cache the calendar for
export CALENDAR_TTL=3600
```

feh123, 2021-04-19:
Hi @mendhak I have just seen the same error at 10:01 pm - the display is now stuck at 09:01 and the calls exceeded error is back. Thanks for the extra code I will git clone the new version tomorrow.

mendhak, 2021-04-19:
Do be sure to check your WEATHER_TTL value.  If by chance it's lower than 1800 then you'll easily hit the 50 limit. 

Another possibility - have you reused your API Key somewhere? In another project?

You could eliminate that possibility by deleting and recreating your AccuWeather app, or creating a new one, and getting a new API Key.

feh123, 2021-04-20:
Hi @mendhak thanks for the ideas. It is odd because I only update every hour. The weather_ttl was set at 3500 and I am sure I only use accuweather api's for the calendar.
I will try deleting and recreating a new api. Your call program for debug does sound good so I may update anyway.

feh123, 2021-04-22:
Hi @mendhak Everything is working fine now. You were right - I had two RPI's running Home Assistant with an Accuweather feed - Oops! I just deleted these feeds. Thanks for adding the call debug code - it will be useful when I try to re-instate a feed on a RPI. Thanks again for all your help.

mendhak, 2021-04-22:
OK good you found it. How many raspberry pis are you running lol!

I'm actually liking accuweather now, but I wish they weren't so stingy the rate limits!

mendhak, 2022-04-28:
Closing as it's been 1 year since any activity, and I've added debug logging and caching to help.

closed 2022-04-28 by mendhak