mendhak/waveshare-epaper-display #25: calendar - show start/end of meeting

crazy-raptor, 2021-08-09:
Hi,
I love this project but wanted to ask whether its possible to show start - stop meeting times?

also do you have a guidance on how I can increase the number of events ?

Thanks!

mendhak, 2021-08-09:
Hello @crazy-raptor 

Start-stop meeting times, you should already be seeing start times, right? So all you're asking for is the end time as well?

To add more events you'll have to edit the SVG file and edit some Python code as well, so it's not a straightforward tweak.  In turn it means if there are any updates to this repo that you need to apply, you'll have to redo the tweaks. 

So, how to do it:  In the SVG get rid of the calendar icon, [which is here](https://github.com/mendhak/waveshare-epaper-display/blob/master/screen-template.svg?short_path=ee7ca41#L958-L962).  

And then you'll see a series of [calendar entries](https://github.com/mendhak/waveshare-epaper-display/blob/master/screen-template.svg?short_path=ee7ca41#L964-L979).  Experiment by changing the `y="..."` values which moves it further up.  Do this for each entry.  You should start creating space at the bottom.  

Then copy paste one of the entries and make a _5 version.  

Then [edit this Python line](https://github.com/mendhak/waveshare-epaper-display/blob/master/screen-calendar-get.py#L16) and make it 5 (or 6 if you managed to squeeze that in)

It should automatically populate the _5 and _6 entries that you added in the SVG

crazy-raptor, 2021-08-09:
Thanks for the tips.

Yes, I'd also like to show the end time of the meeting (or duration). any
thoughts on that one?

mendhak, 2021-08-09:
I think it sounds useful, so I'll take it as a feature request, I could work on it this or next weekend probably.  

It's a bit more complicated as it goes across [multiple files](https://github.com/mendhak/waveshare-epaper-display/blob/master/screen-calendar-get.py), so a bit of reformatting required.

mendhak, 2021-08-10:
@crazy-raptor  So anyway the weekend came a bit early.  If you get the latest version of this repo's code, the calendar start and end times are now shown.  

If the end date is the same as the start date, then it just shows the end time.  If the end date is on another day then both dates are shown with the times.  

[image]

crazy-raptor, 2021-08-10:
Great I’ll test it out shortly . I’ll have a go over my weekend to include the number of items  also

[image]<https://user-images.githubusercontent.com/746276/128905784-105039b0-d3a9-4a02-931c-bd0fadfe9878.png>

mendhak, 2021-12-27:
I've been using this for a while as well so I'm closing this one

closed 2021-12-27 by mendhak