mendhak/waveshare-epaper-display #27: illegal

crazy-raptor, 2021-12-20:
started a fresh install and when I get try to run python3 screen-calendar-get.py   I get a return that just says "illegal operation"

worked through the installation guide, used a fresh raspberry pi image on a pi zero w (got it previously to work with the outlook calendar but wanted to try the google one (got the credentials.json file made and in the waveshare directory)

let me know if you seen this before and what other info is needed?

mendhak, 2021-12-20:
Never seen that before except [in a meme](https://user-images.githubusercontent.com/746276/146824294-3121b9b1-4d33-4c3b-9555-5db8cec0f16a.png).  Are you getting any other info aside from the message, any stack trace?  

The main thing I'd want to find out is which line of code is causing that problems.  

In the `env.sh` you can set `export LOG_LEVEL=DEBUG` and that should hopefully show a bit more info about what's happening just before the error occurs.  

I'll also have a go in a fresh VM to see if something has changed in recent releases of some library.  I'm assuming you're having the problem with Google Calendar setup.

crazy-raptor, 2021-12-20:
not clear where to find the log. Changed to debug

if i run ./run.sh under calendar info it prints
---------------------------------------
GET CALENDAR INFO
---------------------------------------
./run.sh: line 14:   610 Illegal instruction     python3
screen-calendar-get.py
---------------------------------------
EXPORT TO PNG
---------------------------------------
Warning: Option --without-gui= is deprecated
Unable to init server: Could not connect: Connection refused
Unknown option -e

mendhak, 2021-12-20:
Oh illegal instruction, that is an important distinction. That's indicating that your python3 installation is not compatible with your hardware. For example it could be an x64 python trying to run on an ARM raspberry pi.  What os did you install, and did you do any special steps to install anything else?

crazy-raptor, 2021-12-20:
installed it all through command line . had a few issues with some packages in the instructions being depreciated and found newer versions . let me try to reinstall tomorrow.

crazy-raptor, 2021-12-21:
used this to install python on the pi through ssh
this image:
2021-10-30-raspios-bullseye-armhf-lite.zip

installed 3.9.2-3

sudo apt update
sudo apt install python3 idle3

crazy-raptor, 2021-12-21:
did a little python test and that seems to run
***@***.***:~/waveshare-epaper-display $ python3 -c "print('ok')"
ok

mendhak, 2021-12-27:
OK after you reinstalled, did the problem go away?  As in were you able to run the `run.sh` script without a problem this time?

crazy-raptor, 2021-12-27:
hi,
no, unfortunately i still get the error with the screen calendar script .  will do a fresh pi image in a bit. i did notice that some packages specified in the instructions  were depreciated ( incl the one on the piwiring, if i recall wel )

crazy-raptor, 2021-12-28:
hey,

use the full image (2021-10-30-raspios-bullseye-armhf-full) and went
through the install guide again. I got the screen-calendar.py to work now.
I also got the google auth set-up.

I did ran into another "pickle" with the generation of the png file. Any
suggestions here?

EXPORT TO PNG
---------------------------------------
Warning: Option --without-gui= is deprecated
Unable to init server: Could not connect: Connection refused
Unknown option -e
---------------------------------------
DISPLAY ON EPAPER
---------------------------------------
2021-12-28:14:56:12 DEBUG    [display.py:22] Initialize screen
2021-12-28:14:56:12 DEBUG    [epd7in5_V2.py:69] e-Paper busy
2021-12-28:14:56:12 DEBUG    [display.py:32] Read image file:
screen-output.png
2021-12-28:14:56:12 ERROR    [display.py:40] [Errno 2] No such file or
directory: 'screen-output.png'
Traceback (most recent call last):
  File "/home/pi/waveshare-epaper-display/display.py", line 33, in <module>
    Himage = Image.open(filename)
  File "/home/pi/.local/lib/python3.9/site-packages/PIL/Image.py", line
2975, in open
    fp = builtins.open(filename, "rb")
FileNotFoundError: [Errno 2] No such file or directory: 'screen-output.png'

******
As part of the install i found some packages in the guide to be depreciated
/unavailable.

***@***.***:~ $ sudo apt install git ttf-wqy-zenhei ttf-wqy-microhei
python3                            python3-pip python-imaging
libopenjp2-7-dev libjpeg8-dev inkscape figlet wiringp
     i
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'fonts-wqy-zenhei' instead of 'ttf-wqy-zenhei'
Note, selecting 'fonts-wqy-microhei' instead of 'ttf-wqy-microhei'
Package libjpeg8-dev is not available, but is referred to by another
package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  libjpeg9-dev libjpeg62-turbo-dev

E: Unable to locate package python-imaging
E: Package 'libjpeg8-dev' has no installation candidate
E: Unable to locate package wiringpi

used this as alternative to wiringpi--> wget
https://github.com/joan2937/pigpio/archive/master.zip
unzip master.zip
cd pigpio-master
make
sudo make install

crazy-raptor, 2021-12-28:
update:
got it to work editing the inkscape command abit :

inkscape screen-output-weather.svg  --export-filename=screen-output.png
--export-width=800 --export-height=480 --export-dpi=300

and copying /usr/share/fonts -> /usr/share/inkscape/fonts

the display updates now but doesnt show the time of the appointment!

day month year
meeting title

(x4)
any thoughts on this one?

mendhak, 2021-12-28:
Ah ok so looks like there have been some updates, and inkscape isn't taking the `e` argument anymore.  And some additional problems you've found.  I think I'll have to reimage my RPi and try it again from scratch. 

> the display updates now but doesnt show the time of the appointment!

Are you connecting to a calendar, what happens when you remove 'calendar.pickle', then run

    python3 screen-calendar-get.py

crazy-raptor, 2021-12-28:
it’s definitely connecting to the calendar.

it’s a copy of my main calendar where all events are called ‘meeting’. it’s showing 4 ‘meetings’ on the display with the dates but no time !

i don’t understand enough on the google calendar formats in the scripts to see where it might throw an issue. when i run the script it prints exactly what it then shows on the display. ( so it doesn’t look like it’s an inkscape issue on my end with the updates i managed to do)

mendhak, 2021-12-28:
If you're not seeing any times then those might be all day events in Google Calendar.  Any events with times against them should appear like this

```
[screen-calendar-get.py:176] main() - {'CAL_DATETIME_1': 'Wed Dec 29, 8:00 AM - 12:00 PM', 'CAL_DESC_1': 'Some timed event', 'CAL_DATETIME_2': 'Thu Dec 30', 'CAL_DESC_2': 'Some all day event'}

```

crazy-raptor, 2021-12-28:
could be! it's not updating for some reason at the moment (added a few
short test meetings).

Will check tomorrow

mendhak, 2021-12-28:
Yeah it won't update right away, there's a cache file (otherwise you'll hit a rate limit and get blocked).  If you delete the calendar.pickle file, then the code should get the meetings fresh from Google Calendar.

mendhak, 2021-12-29:
I've been able to recreate the 'illegal operation' error.  It's very odd.  I did a brand new card, flashed Raspberry Pi OS Lite.  Installed things.  And then I started getting 'illegal operation' on normal commands, not just related to this project.  Just running `pip3` would give me that error.  When I re-flashed, and ran the install steps again,things started working again.  So the problem did appear in front of me, just not reliably reproducible.  I bet if I flash the card again, it'll start failing again. Intermittent problems, that's the worst kind of problem!

I'll adjust instructions in the documentation as well, to includ pigpiod.  And see if there's a simpler way to do some of the steps.

mendhak, 2021-12-29:
Good news.  Well it's probably a bit late for you so it's "OK News".  

I've redid the instructions with Raspberry Pi OS.  It's muuuuuch simpler now.  `pigpio` is already available via apt, so there's no more `make` instructions required, I've removed a bunch of unused dependencies.  I've also replaced inkscape with cairosvg as well so need for the fonts workaround.  

The [setup section](https://github.com/mendhak/waveshare-epaper-display#setup-dependencies) is much simpler now.  

If you do decide to get latest you'll want to rerun the dependencies instructions but it should be shorter this time.

crazy-raptor, 2021-12-29:
so.. re-imaged and followed the new instructions, got:
***@***.***:~/waveshare-epaper-display
$ python3 screen-calendar-get.py
Illegal instruction

mendhak, 2021-12-29:
I've reflashed about 10-15 times today and only got that problem once. :disappointed:  I really think this has something to do with the Raspberry Pi Bullseye image.  

I don't know if it makes a difference, you could try using the Raspberry Pi imager here:  https://www.raspberrypi.com/software/  

[image]

When you open this software press Ctrl+Shift+X and it will open an advanced menu where you can enable SSH and set a username password and enable WIFI too.

crazy-raptor, 2021-12-29:
that worked! use balena etcher before

mendhak, 2022-01-07:
@crazy-raptor in case you run into the error again I found the solution. 

https://github.com/mendhak/waveshare-epaper-display/issues/29#issuecomment-1007796898

I've also updated the instructions in the README to download the fixed package version so that should take care of future installers.

closed 2022-01-08 by mendhak

crazy-raptor, 2022-01-08:
thanks for looking into this !

saw another user mentioning smaller screens. would be great to collect
versions for those ( eg just the next meetings ) also . eg for something
like this
https://www.waveshare.com/product/displays/e-paper/2.13inch-e-paper-b.htm
to work with a pi zero w . I might get one to try something like this

mendhak, 2022-01-08:
That's a good idea and I think it's not too difficult. Would appreciate if you did that, share the SVG.  

In the coding side it would be a matter of letting the user specify some environment variable, which causes that 'calendar only' SVG to be picked.

crazy-raptor, 2022-01-08:
have you ever looked into the raspberry pico ? mentions it runs micropython
?

mendhak, 2022-01-08:
I think I've heard of it but never looked into it.  It seems interesting in that it would only run the application you put on it, no OS extras.  I'm not familiar with how microcontroller types would be set up. I'm guessing for a project like this, you'd need to find an eink display for it + networking module + battery module.