mendhak/waveshare-epaper-display #29: I'm getting Illegal instruction when running python3 screen-calendar-get.py

mrdanjohnson, 2022-01-07:
This is my second time setting up this project. Had the previous one working fine, but this time, after importing the credentials.json file, I get "Illegal instruction" when running 'python3 screen-calendar-get.py.

note: the json file I downloaded from google was not called credentials.json but a name that started with "client_secret_..." Contents of the file look like this;

{"installed":{"client_id":"17292850-eqmq8jnabvtnvruc0ee.apps.googleusercontent.com","project_id":"flash-clover-337523","auth_uri":"https://accounts.google.com/o/oauth2/auth","token_uri":"https://oauth2.googleapis.com/token","auth_provider_x509_cert_url":"https://www.googleapis.com/oauth2/v1/certs","client_secret":"GX-iwNEA_k-CS7c17MV","redirect_uris":["urn:ietf:wg:oauth:2.0:oob","http://localhost"]}}

I removed characters  in the secret vars before posting.

Thanks

mrdanjohnson, 2022-01-07:
Tried again using a different google account. Same issue.

mrdanjohnson, 2022-01-07:
I am getting the same error for outlook_util.py

mendhak, 2022-01-07:
Yes I think once you start getting "illegal instruction" you'll start getting it everywhere.  There's something in the new Raspberry Pi OS that causes this to happen and then it happens everywhere. 

I've only been able to reproduce this once, just once, and during that I couldn't even run "pip --version" I was still getting that error.  

What did you use to write the image to SD card?  In this thread there seems to be better success using the Raspberry Pi Imager?  https://github.com/mendhak/waveshare-epaper-display/issues/27#issuecomment-1002670476

Although, that itself should make no difference but worth trying?

mrdanjohnson, 2022-01-07:
I did use the pi imager, and the demo python scripts work from waveshare (python not python3). Do I really need to start over, or can you think of a way through it?

Also, I would like to contribute the fact that I am using this project on a smaller screen, and it works just great. I simply changed the driver call to use the appropriate driver for my screen size, and the size of image that it creates when the script is run. Could be an option you add for those using different screen sizes.

mendhak, 2022-01-07:
Only thing comes to mind is troubleshooting.  Are you any good with `gdb` or Python virtual environments? 

I think the last time I got the error I was getting it just doing `python3 --version`.  

I did 

    gdb --args python3 --version

And then when it failed I think I used the `tb` command and got some output pointing at a problem in the Python crypto library??? I can't remember well now. 

So the other thing to try could be to create a virtual environment in the waveshare folder and do the package installations there.  

But that will depend if your pip is still working.

mrdanjohnson, 2022-01-07:
No, not good at any of this. Just learning really.

I did try;

python3 -m pip install Pillow

And it worked just fine.

mrdanjohnson, 2022-01-07:
I was also able to run 'python3 utility.py'
It didn't output anything, but I didn't get the error.

Screen-weather-get.py works as well. (Using: python3 screen-weather-get.py)

mendhak, 2022-01-07:
I can try tonight after work - reflash an SD card until I get the problem. Then find a way around.

My difficulty is I'm not getting a problem frequently enough or knowing what's gone wrong - we're just installing normal Python packages - so it's hard to find that workaround or root cause.

OK if you're not familiar with gdb/venv. 

If you have time and willing, you can also try narrowing down something.  Try to uninstall the python packages then reinstalling them. 

This is what the instructions said: 

```
sudo pip3 install python-dateutil astral spidev RPi.GPIO Pillow google-api-python-client google-auth-httplib2 google-auth-oauthlib msal
```

I'd suggest uninstalling the last few.  Do a `sudo pip3 uninstall google-api-python-client google-auth-httplib2 google-auth-oauthlib msal` 

Then try running the screen-calendar-get.  Hopefully it gives an error other than illegal instruction.  

Then one by one, reinstall the packages.  I hope that can help narrow down which package the problem is centered around.

mendhak, 2022-01-07:
Finally! It took 7 tries and I finally got the error.  Each time I reflashed the card, followed the README instructions exactly, but the code always worked.  On the 7th try I got an 'illegal instruction' and I think I have the solution and cause.  

## First, the solution 

The simplest way to recreate the error is

    python3 -c "from cryptography.fernet import Fernet"

You should see "Illegal instruction"

Now downgrade the cryptography package:

    sudo pip3 install cryptography==36.0.0

This should remove the current 36.0.1 version you have, and put 36.0.0 in its place.  

[image]

Now run the test command again. 

        python3 -c "from cryptography.fernet import Fernet"

You should see no output.  That means it works, and the rest of the application should work too.  You should be able to do the normal screen-calendar-get.py run.  

## Observations

Some of the packages involved require the pip `cryptography` package.  I don't know why but the latest version of this package, 36.0.1, is causing Illegal Instruction on Raspberry Pi.  

During pip install this wheel gets downloaded

>   Downloading https://www.piwheels.org/simple/cryptography/cryptography-36.0.1-cp39-cp39-linux_armv6l.whl (2.7 MB)

I tried to run a gdb of the test command. It pointed at a problem in `/usr/local/lib/python3.9/dist-packages/cryptography/hazmat/bindings/_rust.abi3.so`.  

    pi@epaper:~ $ gdb --args python3 -c "from cryptography.fernet import Fernet"
    GNU gdb (Raspbian 10.1-1.7) 10.1.90.20210103-git
    Copyright (C) 2021 Free Software Foundation, Inc.
    License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
    This is free software: you are free to change and redistribute it.
    There is NO WARRANTY, to the extent permitted by law.
    Type "show copying" and "show warranty" for details.
    This GDB was configured as "arm-linux-gnueabihf".
    Type "show configuration" for configuration details.
    For bug reporting instructions, please see:
    <https://www.gnu.org/software/gdb/bugs/>.
    Find the GDB manual and other documentation resources online at:
        <http://www.gnu.org/software/gdb/documentation/>.

    For help, type "help".
    Type "apropos word" to search for commands related to "word"...
    Reading symbols from python3...
    (No debugging symbols found in python3)
    (gdb) run
    Starting program: /usr/bin/python3 -c from\ cryptography.fernet\ import\ Fernet
    [Thread debugging using libthread_db enabled]
    Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".

    Program received signal SIGILL, Illegal instruction.
    0xb645356c in register_tm_clones () from /usr/local/lib/python3.9/dist-packages/cryptography/hazmat/bindings/_rust.abi3.so
    (gdb) bt
    #0  0xb645356c in register_tm_clones () from /usr/local/lib/python3.9/dist-packages/cryptography/hazmat/bindings/_rust.abi3.so
    #1  0xb6fdd328 in call_init (l=<optimized out>, argc=argc@entry=3, argv=argv@entry=0xbefff6b4, env=env@entry=0xbefff6c4) at dl-init.c:72
    #2  0xb6fdd430 in call_init (env=<optimized out>, argv=<optimized out>, argc=<optimized out>, l=<optimized out>) at dl-init.c:30
    #3  _dl_init (main_map=0x4c8290, argc=3, argv=0xbefff6b4, env=0xbefff6c4) at dl-init.c:119
    #4  0xb6e64a0c in __GI__dl_catch_exception (exception=0x5, operate=0x0, args=0xbeffcd18) at dl-error-skeleton.c:182
    #5  0xb6fe1514 in dl_open_worker (a=<optimized out>) at dl-open.c:758
    #6  0xb6e649a8 in __GI__dl_catch_exception (exception=0xb668d280, operate=0xbeffcec0, args=0xbeffcf40) at dl-error-skeleton.c:208
    #7  0xb6fe0d3c in _dl_open (file=0xb668ee00 "/usr/local/lib/python3.9/dist-packages/cryptography/hazmat/bindings/_rust.abi3.so", mode=-2147483646, caller_dlopen=0x1f1948, nsid=-2, argc=3, argv=0xbefff6b4, env=0xbefff6c4) at dl-open.c:837
    #8  0xb6f79d98 in dlopen_doit (a=0xbeffd12c) at dlopen.c:66
    #9  0xb6e649a8 in __GI__dl_catch_exception (exception=0x58, exception@entry=0xbeffd0e0, operate=0xbeffd0e8, args=0x0, args@entry=0xb6f8c000) at dl-error-skeleton.c:208
    #10 0xb6e64a90 in __GI__dl_catch_error (objname=0x505394, errstring=0x505398, mallocedp=0x505390, operate=<optimized out>, args=0xbeffd12c) at dl-error-skeleton.c:227
    #11 0xb6f7a87c in _dlerror_run (operate=<optimized out>, args=0xbeffd12c, args@entry=0xbeffd124) at dlerror.c:170
    #12 0xb6f79e5c in __dlopen (file=<optimized out>, mode=<optimized out>) at dlopen.c:87
    #13 0x001f1948 in ?? ()
    Backtrace stopped: previous frame identical to this frame (corrupt stack?)
    (gdb) quit
    A debugging session is active.

        Inferior 1 [process 1173] will be killed.

    Quit anyway? (y or n) y

I then did some guesswork and tried downgrading to version 36.0.0 and the code started working. 

I had a look at the cryptography package's [changelog](https://github.com/pyca/cryptography/blob/main/CHANGELOG.rst#3601---2021-12-14) and and the only change is that this 36.0.1 has been compiled with a new version of openssl.  I don't know what that means for us.  

I think I should probably raise an issue in the cryptography Github repo but it's a bit intimidating :stuck_out_tongue_closed_eyes:

mendhak, 2022-01-07:
> Also, I would like to contribute the fact that I am using this project on a smaller screen, and it works just great. I simply changed the driver call to use the appropriate driver for my screen [...]

This could be useful.  Which waveshare screen are you using.  Can you show what you added to the display.py.  Maybe I could add some environment variable and point it at the right driver version.

mrdanjohnson, 2022-01-08:
It seems that it's not until the google-python-client is installed that I get that error. 

I did what you suggested. Uninstalled that group, and generally got a different error until I added that client.

mrdanjohnson, 2022-01-08:
Wait a second. But the script to get outlook is doing the same thing... So it can't be the google api.

Does that help?

When I run ./run.sh everything else works just fine.

mendhak, 2022-01-08:
Did you:

    sudo pip3 install cryptography==36.0.0

mrdanjohnson, 2022-01-08:
I think that worked! Thanks so much! What would cause it to randomly load the wrong cryptography?

mendhak, 2022-01-08:
I did a bit of investigation in my [long comment above](https://github.com/mendhak/waveshare-epaper-display/issues/29#issuecomment-1007796898), but best I can tell is that screen-calendar-get references msal package, msal references cryptography package.  

The cryptography library that Raspberry Pi uses is provided by a group called [PiWheels](https://www.piwheels.org/).  I think their version 36.0.1 may have been compiled just [4 days ago](https://github.com/piwheels/packages/issues/267) so it might be related. I'm not sure.  But I've raised an [issue there anyway](https://github.com/piwheels/packages/issues/273).

mrdanjohnson, 2022-01-08:
Is this error from the crypto thing we did, or is it a problem with my developer app. I know I had to do something that wasn't in the instructions.

Had to complete the OAuth consent screen / settings before I could get a client ID

Error 400: redirect_uri_mismatch

You can't sign in to this app because it doesn't comply with Google's OAuth 2.0 policy.

If you're the app developer, register the redirect URI in the Google Cloud Console.
Learn more
Request Details
The content in this section has been provided by the app developer. This content has not been reviewed or verified by Google.
If you’re the app developer, make sure that these request details comply with Google policies.
redirect_uri: http://localhost:8080/

mrdanjohnson, 2022-01-08:
Seems to be working great now. Before I close this, could I run something by you?

Because my screen is smaller, I was always able to resize the output using the settings in run.sh. For some reason, when I change them now, it draws a full black screen. Did something change in your recent update that would effect that?

Also, do you know of a graphical editor for the SVG file? Because of the size of my panel and the way the panel refreshes, I would like to remove the time, and move the date to the left.

Thanks so much for your time on my issue, and for maintaining your project. I really enjoy it.

Dan

mendhak, 2022-01-08:
Sure, I can give you a few things to try to troubleshoot.

Regarding recent changes that will depend on the last time you ran the setup.  I've got a [Changelog](https://github.com/mendhak/waveshare-epaper-display/blob/master/CHANGELOG.md) if you know when you last did a setup.  The most recent change has been to use 'cairosvg' instead of 'inkscape' to convert the SVG to PNG.  

Nothing has changed in the code to display the image on screen itself.  

What you could do is try to verify the image that's generated.  In the waveshare directory you will see a file called `screen-output.png`.  This is the file that gets generated from the SVG.  You can try copying that to your main computer, for example through `scp` and have a look at it there. Is it all blank or is it a normal looking image.  

You've also mentioned that you change the driver path in the `display.py` script.  You'll want to make sure you're pointing at the right path.  Try the [e-Paper repo](https://github.com/waveshare/e-Paper) demo itself, ensure that you can write an image using their sample code.  

You can even try copying over some sample image to the Pi and use display.py to draw it on screen.  

What you're doing is trying to figure out which thing could be the problem - driver, screen, or generated image. 

----

So to edit the SVG, I think Inkscape is the most popular one, you can try that.  But for most of my editing I've found the fastest, simplest way is to edit the `screen-template.svg` and rerun `./run.sh`.  It's a bit crude but directly shows how it'd appear.  

Based on what you've described.  Removing time is a matter of finding `TIME_NOW` in the SVG and deleting it. Moving the date to the left requires finding DAY_NAME and DAY_ONE and setting their `x="..."` values to make them shift left.

mendhak, 2022-03-03:
It's been a while I'll assume everything is OK now, so I'm going to close this issue.  We can reopen it if there are still problems.

closed 2022-03-03 by mendhak