mendhak/waveshare-epaper-display #30: cairosvg: error: no such option: --output-width

DominicBoisvert, 2022-01-10:
Hi,

With cairoSVG (pyhton3-cairosvg package), I got the following error message: cairosvg: error: no such option: --output-width

I changed "--output-width" to "-W" and "--output-height" to "-H" (line 26 of the run.sh file).

The error message is gone, and the display is good.

mendhak, 2022-01-10:
Thanks for reporting that, did you install cairosvg using apt or pip?  What version appears when you run `cairosvg --version` ?

I wonder if the [arguments are newer](https://cairosvg.org/documentation/) maybe.

mendhak, 2022-01-10:
Strangely for me when I use `-W` and `-H` the image that's generated is the wrong dimensions and doesn't display on the epaper screen (it errors with 'wrong dimensions'). 

I had to use --output-width and --output-height to get it working.

DominicBoisvert, 2022-01-11:
Thanks for the quick return.

The package was python3-cairosvg, installed via apt (Raspbian Buster). The version was 1.0.20.

I removed the package (via apt) and installed it with pip3. Now it shows version 2.5.2, and it's working with either combination of `--output-width` and `--output-height` or `-W` and `-H`.

DominicBoisvert, 2022-01-11:
Installing cairosvg (version 2.5.2) with pip3 solved the issue.

closed 2022-01-11 by DominicBoisvert