mendhak/waveshare-epaper-display #40: missing file

crazy-raptor, 2022-09-09:
Hi, tried installing the latest version to get the different display version (more entries) but the "lib" folder is empty and when runnign env.sh erro is shown:
Traceback (most recent call last):
  File "/home/pi/waveshare-epaper-display/display.py", line 18, in <module>
    from waveshare_epd import epd7in5_V2 as epd7in5
ModuleNotFoundError: No module named 'waveshare_epd'

Do i manually copy edp7in5_V2 into the lib folder (whats the best way of doing that through SSH?)

crazy-raptor, 2022-09-09:
was able to git clone the e-ink files into lib folder.  Calendar doesnt seem to update yet but i'll see about that later

crazy-raptor, 2022-09-10:
REgarding the calendar not syncing while it did with an earlier install i am getting this error: 2022-09-10:08:26:39 INFO     [screen-calendar-get.py:186] Fetching Google Calendar Events
Traceback (most recent call last):
  File "/home/pi/waveshare-epaper-display/screen-calendar-get.py", line 197, in <module>
    main()
  File "/home/pi/waveshare-epaper-display/screen-calendar-get.py", line 187, in main
    google_events = get_google_events(max_event_results)
  File "/home/pi/waveshare-epaper-display/screen-calendar-get.py", line 101, in get_google_events
    service = build('calendar', 'v3', credentials=get_google_credentials(), cache_discovery=False)
  File "/home/pi/waveshare-epaper-display/screen-calendar-get.py", line 89, in get_google_credentials
    credentials = flow.run_local_server()
  File "/usr/local/lib/python3.9/dist-packages/google_auth_oauthlib/flow.py", line 457, in run_local_server
    local_server = wsgiref.simple_server.make_server(
  File "/usr/lib/python3.9/wsgiref/simple_server.py", line 154, in make_server
    server = server_class((host, port), handler_class)
  File "/usr/lib/python3.9/socketserver.py", line 452, in __init__
    self.server_bind()
  File "/usr/lib/python3.9/wsgiref/simple_server.py", line 50, in server_bind
    HTTPServer.server_bind(self)
  File "/usr/lib/python3.9/http/server.py", line 138, in server_bind
    socketserver.TCPServer.server_bind(self)
  File "/usr/lib/python3.9/socketserver.py", line 466, in server_bind
    self.socket.bind(self.server_address)
OSError: [Errno 98] Address already in use

mendhak, 2022-09-10:
For the first one, did you follow the README command to use `--recursive`, if you do that then the lib folder would get populated

    git clone --recursive https://github.com/mendhak/waveshare-epaper-display.git

For the calendar error, I think what's happening is the Google Authentication workflow is trying to set up a server to listen on locally for completion of the workflow.  `Address already in use` indicates something is already listening on port 8080.  If you have something on port 8080 can you kill it while you do the Google Authentication.  

If you don't have anything deliberately listening on 8080 then you can do `sudo lsof -i:8080` and it should show you the process ID of what's listening on that port to give you a clue.  You can kill it.  Or just reboot.

crazy-raptor, 2022-09-10:
Hi, yes, i killed the process and ran they screen-calendar-get.py but getting this and no token.pickle is being created:
 INFO     [flow.py:522] "GET /?state=tCdIRvnH583V2j7rjcpr9Ww3v2drGI HTTP/1.1" 200 65
Traceback (most recent call last):
  File "/home/pi/waveshare-epaper-display/screen-calendar-get.py", line 197, in <module>
    main()
  File "/home/pi/waveshare-epaper-display/screen-calendar-get.py", line 187, in main
    google_events = get_google_events(max_event_results)
  File "/home/pi/waveshare-epaper-display/screen-calendar-get.py", line 101, in get_google_events
    service = build('calendar', 'v3', credentials=get_google_credentials(), cache_discovery=False)
  File "/home/pi/waveshare-epaper-display/screen-calendar-get.py", line 89, in get_google_credentials
    credentials = flow.run_local_server()
  File "/home/pi/.local/lib/python3.9/site-packages/google_auth_oauthlib/flow.py", line 505, in run_local_server
    self.fetch_token(authorization_response=authorization_response)
  File "/home/pi/.local/lib/python3.9/site-packages/google_auth_oauthlib/flow.py", line 300, in fetch_token
    return self.oauth2session.fetch_token(self.client_config["token_uri"], **kwargs)
  File "/usr/lib/python3/dist-packages/requests_oauthlib/oauth2_session.py", line 186, in fetch_token
    self._client.parse_request_uri_response(authorization_response,
  File "/usr/lib/python3/dist-packages/oauthlib/oauth2/rfc6749/clients/web_application.py", line 203, in parse_request_uri_response
    response = parse_authorization_code_response(uri, state=state)
  File "/usr/lib/python3/dist-packages/oauthlib/oauth2/rfc6749/parameters.py", line 274, in parse_authorization_code_response
    raise MissingCodeError("Missing code parameter in response.")
oauthlib.oauth2.rfc6749.errors.MissingCodeError: (missing_code) Missing code parameter in response.

i'm going to try a fresh install

crazy-raptor, 2022-09-10:
different SD card, fresh install (the full image this time, rather than lite).  followed all steps. got the google calender set-up, where I then pasted the code into the browser -> localhost... -> copy and "curl" it into a duplicate terminal but again this error:

Traceback (most recent call last):
  File "/home/pi/waveshare-epaper-display/screen-calendar-get.py", line 197, in                                 <module>
    main()
  File "/home/pi/waveshare-epaper-display/screen-calendar-get.py", line 187, in                                 main
    google_events = get_google_events(max_event_results)
  File "/home/pi/waveshare-epaper-display/screen-calendar-get.py", line 101, in                                 get_google_events
    service = build('calendar', 'v3', credentials=get_google_credentials(), cach                                e_discovery=False)
  File "/home/pi/waveshare-epaper-display/screen-calendar-get.py", line 89, in g                                et_google_credentials
    credentials = flow.run_local_server()
  File "/usr/local/lib/python3.9/dist-packages/google_auth_oauthlib/flow.py", li                                ne 505, in run_local_server
    self.fetch_token(authorization_response=authorization_response)
  File "/usr/local/lib/python3.9/dist-packages/google_auth_oauthlib/flow.py", li                                ne 300, in fetch_token
    return self.oauth2session.fetch_token(self.client_config["token_uri"], **kwa                                rgs)
  File "/usr/lib/python3/dist-packages/requests_oauthlib/oauth2_session.py", lin                                e 186, in fetch_token
    self._client.parse_request_uri_response(authorization_response,
  File "/usr/lib/python3/dist-packages/oauthlib/oauth2/rfc6749/clients/web_appli                                cation.py", line 203, in parse_request_uri_response
    response = parse_authorization_code_response(uri, state=state)
  File "/usr/lib/python3/dist-packages/oauthlib/oauth2/rfc6749/parameters.py", l                                ine 274, in parse_authorization_code_response
    raise MissingCodeError("Missing code parameter in response.")
oauthlib.oauth2.rfc6749.errors.MissingCodeError: (missing_code) Missing code par                                ameter in response.

crazy-raptor, 2022-09-10:
all good , forgot the " " again doing the curl command.

for some reason, the text is very faint after refresh. but maybe this will resovle

mendhak, 2022-09-10:
I'm not sure what you mean by very faint, do you mean it's visible, just thin?    If so, that's how that view is, in order to cram as much as possible the fonts need to be non-bold, whereas the other views you will see have bolder fonts, to favor readability.  If that isn't it you could post a comparison photo of the good versus faint. Are you using the Waveshare 7.5" screen or something else?

crazy-raptor, 2022-09-10:
no the letters are actually quite large but very faint , like a shadow . i’ll take a pic tomorrow

the views 1,3 and 4, letters are full black. view 2 , like sometbing barely transparent is in front of it

crazy-raptor, 2022-10-11:
using wavescreen 7.5 single colour

in the other views the font is much darker , like in your screenshots. only
this view has an issue .

mendhak, 2022-11-30:
I haven't heard back in a while and I don't see any photos so I'll close this, we can reopen if there's still a problem.

closed 2022-11-30 by mendhak