mendhak/waveshare-epaper-display #46: Calendar not updating

feh123, 2023-01-18:
Today at 13:00 my calendar stopped updating. If I run ./run.sh I get the following:

EXPORT TO PNG
---------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python3.9/xml/etree/ElementTree.py", line 1720, in feed
    self.parser.Parse(data, False)
xml.parsers.expat.ExpatError: not well-formed (invalid token): line 13, column 66

I have truncated the output - the weather entries it downloads and the diary entries look good so I don't think that is the issue. It's at the export to png that things seem to fail. I have just pasted the first lines of the traceback. Any thoughts on what is wrong much appreciated. As far as I am aware I have not altered anything. The Calendar software was last updated Apirl 2022.

mendhak, 2023-01-18:
Nothing's jumping out from that, I think I'd need to see more to know what's going on. 

Have a look at some of the calendar entries, it's happened to me that a weird character sometimes messed up the SVG and the PNG generation fails.  If it turns out to be some weird character I could investigate.   At least that's what the error is indicating, something has been written to the file and it's considered a bad SVG.  

If you want you could send me the SVGs in that directory `gpslogger@mendhak.com` and I could look and see what goes wrong while rendering it, but you'll have to not modify any of the contents, which means it's going to contain personal data.

mendhak, 2023-01-18:
To do the first bit - figure out the bad 'thing' causing SVG corruption I think it should be enough to just open the file in something like Firefox or VSCode or anything that has an XML Viewer and it should point out an error.  Somewhere near that, will be an offending 'character' or token or something.

feh123, 2023-01-18:
Hi @mendhak yes, solved, you were right. This morning I added to my diary an entry title contaning &. When I opened the svg in VScode this was highlighted in red. I changed it to "and", checked ./run.sh and the Calendar has updated properly. Thanks a lot for your fast response and the answer to my problem.

closed 2023-01-18 by feh123

mendhak, 2023-01-19:
Ah just so you know I did run into that myself in September, https://github.com/mendhak/waveshare-epaper-display/issues/39

There's a fix for it but if you have any version after September, you had mentioned yours is from April

feh123, 2023-01-19:
Thanks. Yes you are right I last updated in Apr so I must put this on my todo list. The calendar is quite key for our house now!