mendhak/waveshare-epaper-display #47: E-Paper black and white swapped

Karllebolla, 2023-01-20:
Hello,

first of all thanks for that fantastic project.
I got everythinkg working except that black and white on the e-paper is swapped - means the background is black and the text is white.

The generated .svg looks fine - white background and black text.

Do you know whats happened? 

Thanks for your help.

mendhak, 2023-01-21:
Never heard of this happening before! 

You'll want to make sure you have the [right version of the display set here](https://github.com/mendhak/waveshare-epaper-display/blob/master/env.sh.sample#L33).  The code here only knows about v1 and v2, do you know if your version is different?

I guess you could try to isolate where the problem is.  Can you try the waveshare demo repository: https://github.com/waveshare/e-Paper

There are some examples in there, you can match the example with your screen and see whether it produces a swapped image as well.  https://github.com/waveshare/e-Paper/tree/master/RaspberryPi_JetsonNano/python

Karllebolla, 2023-01-23:
Oh damn i got the V3-Version ...

you plan to expand your code for V3 or do you know what is to adapt for the new version?

mendhak, 2023-01-23:
I don't have a V3 but I'd be happy to include its code if you can help with some bits, it shouldn't be a lot of effort. The easiest way to do this is to figure out which of these demos work for your version of the display:  

https://github.com/waveshare/e-Paper/tree/master/RaspberryPi_JetsonNano/python/examples

If you can let me know which of the samples worked for you I should be able to figure out the module name.  However in that list I don't see a V3, is the product called something else? Could you link to it? 

If you are curious you can try it yourself too.  Inside the sample file, there will be the name of the exact module that it is using to work with your screen.  That module, in turn would be placed into this block in `display.py`.  You could verify if for yourself by replacing this block: 

https://github.com/mendhak/waveshare-epaper-display/blob/690adc773cccbba87ba1970be125b20a63012a17/display.py#L15-L18

with (substitute the xxxxxxxx ):

```python
from waveshare_epd import xxxxxxxx as epd7in5 
```

Karllebolla, 2023-01-23:
I got i working.
For V3 you have to import **epd7inb_V2**.

epd.display needs now an additional parameter:

` def display(self, imageblack, imagered):
        self.send_command(0x10)
        # The black bytes need to be inverted back from what getbuffer did
        for i in range(len(imageblack)):
            imageblack[i] ^= 0xFF
        self.send_data2(imageblack)

        self.send_command(0x13)
        self.send_data2(imagered)
        
        self.send_command(0x12)
        epdconfig.delay_ms(100)
        self.ReadBusy()`

I added the line from waveshare example code and it worked well for me but i let you comment if this i correct or not, you are more expert on this.

`Limage_Other = Image.new('1', (epd.height, epd.width), 255)  # 255: clear the frame
 epd.display(epd.getbuffer(Himage), epd.getbuffer(Limage_Other))
`
Thanks for you help.

mendhak, 2023-01-27:
Sorry for delay in this, it'll be a few more days before I can get to this and add it to the codebase. 

So to be clear do you mean `epd7in5b_V2`, I couldn't find `epd7inb_V2`.  And I assume the screen is [this one](https://www.waveshare.com/product/displays/e-paper/epaper-1/7.5inch-e-paper-hat-b.htm), it has red color capability?  Or if you could point me at the [sample code file in this list](https://github.com/waveshare/e-Paper/tree/master/RaspberryPi_JetsonNano/python/examples) that would be useful, for comparison.

Thanks for sorting the code out, I should be able to add it into an if-else statement to get that extra Limage_other

mendhak, 2023-01-29:
OK I've merged your code into the main branch.  I've tested my own screen to make sure it's still working.  You can give it a try, set 

    export WAVESHARE_EPD75_VERSION=2B

If I end up adding more screens in the future I'll consider breaking this variable to make the names include more info.  For now hopefully this should work.

Karllebolla, 2023-02-02:
Sorry for delay ...
I tested your fix and it worked well for me. Thanks

mendhak, 2023-02-02:
🍜 nice!

closed 2023-02-02 by mendhak