mendhak/waveshare-epaper-display #52: Discussion: 24 hour time.

hnnweb, 2023-02-05:
How much work would it be to display the 24 hour timeformat?
The most correct way would be to rely on the LC_TIME format.

mendhak, 2023-02-05:
Well I started playing around with this, and I'm so glad you asked this because I had a ton of fun trying things out. I thought it would be simple, but ended up in an unexpected place.  tl;dr: It's possible but fonts are a problem. 

It seems there's a way in Python to tell it to pick up the system's locale settings which includes the date and time formatting.  

```
$ LC_TIME=ko_KR.utf8 python3
>>> import locale, datetime
>>> locale.setlocale(locale.LC_ALL, '')
'LC_CTYPE=en_GB.UTF-8;LC_NUMERIC=en_GB.UTF-8;LC_TIME=ko_KR.utf8;LC_COLLATE=en_GB.UTF-8;LC_MONETARY=en_GB.UTF-8;LC_MESSAGES=en_GB.UTF-8;LC_PAPER=en_GB.UTF-8;LC_NAME=en_GB.UTF-8;LC_ADDRESS=en_GB.UTF-8;LC_TELEPHONE=en_GB.UTF-8;LC_MEASUREMENT=en_GB.UTF-8;LC_IDENTIFICATION=en_GB.UTF-8'
>>> datetime.datetime.now().strftime('%X')
'18시 30분 09초'
```

So the first problem is the seconds, because the 'time' string isn't the same across cultures it isn't as easy as cutting off the last part.   It's not end of the world though, there's a library called Babel which can help. It has a format=short which just does hour and minutes.  In some cases it was just the hour and minute, and in some cases it includes a suffix like 'pm' or 'PM' or 'μ.μ.' or 'م'

```
LC_TIME=el_GR.utf8 python3
Python 3.10.6 (main, Nov 14 2022, 16:10:14) [GCC 11.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from datetime import datetime
>>> from babel.dates import format_time
>>> format_time(datetime.now(), format='short')
'6:56 μ.μ.'
```

This is where I ran into an issue, the font! 

When I try zh_TW, and I'd imagine for some other locales, the font I've chosen in the SVG doesn't have the right symbols. 

This was zh_TW

[image]

This was ko_KR

[image]

The left side time is from the Babel library and the right side date is the normal Python formatting. 

Similar issues with Korean, though in fairness most European languages didn't have a problem. 

[image]

---

OK some thoughts.  So one option is (what's explored above) is just completely respecting what the OS says and letting Python pick it up and display it however it should - that includes whether to show 12 or 24 hour format.  And the main task then is to figure out if the missing character problem, can a different font be picked somehow, or does the system tell you what its "default" font is. 

Another option is to completely ignore all of that, don't respect anything, just have an option for 12 or 24 hour time. 

Of course _I really like_ the first option, because it's not just about 24 hour format but localization.  Wouldn't it be nice if this weren't English centric! And I bet that the calendar entries could be localized too, with a bit of help from the `humanize` package. 

If you've read this far let me know your thoughts on it.  I will be playing around with font/display options next week anyway and see if there's a simple solution.

hnnweb, 2023-02-06:
Ok, this kind of exactly what i have in mind.

I was intentionally skipping the translation part because i can live with Monday, Tuesday and so on... 
But i can't figure out the whole AM/PM thing... :) ( Don't bother try to explain it, i have the Wikipedia article in my browser toolbar )

Both options work for me and for my spouse. The time is the big thing.

The font, is that TTF? I know that the limitation is in the display/e-ink. To small it becomes blurry och pixelated. To big and the stuff doesn't fit. Maybe play around with Google Fonts or similiar if that works.

mendhak, 2023-02-06:
OK cool I'll continue playing.  

The font default in the SVG is 'Nimbus' which is decent but wouldn't display non-Latin. Also I am not sure if it's truetype or what truetype means, but its file extension is '.pfb'.  

I have also tried using the font ['Noto'](https://fonts.google.com/noto) (which has .ttf xtension). It seems to be a family of fonts that is more Unicode friendly than others, but for C/J/K you have to pick specific font names, so it wasn't that useful.  It is narrower than the system default which I believe is 'DejaVu Sans' (ttf).  DejaVu Sans already seems to be pretty good.  

A simpler way I found, when I leave the font at the web-safe value `sans-serif` then the system figures out what to use and pick the right thing. 

```
$ LC_ALL=th_TH.UTF-8 fc-match sans-serif
FreeSerif.ttf: "FreeSerif" "ปกติ"
$ LC_ALL=sv_SE.UTF-8 fc-match sans-serif
DejaVuSans.ttf: "DejaVu Sans" "Book"
LC_ALL=zh_TW.UTF-8 fc-match sans-serif
DroidSansFallbackFull.ttf: "Droid Sans Fallback" "Regular"
```

Thai

[image]

zh_TW

[image]

sv_SE

[image]

ko_KR

[image]

mendhak, 2023-02-13:
Finally I have merged the changes.  

By default the display will be in the Raspberry Pi's locale (en_GB).  I've added instructions on using a different locale: https://github.com/mendhak/waveshare-epaper-display#how-to-use-a-different-display-language

Worth noting, the setup instructions are changed! It's using a [Python virtual environment](https://github.com/mendhak/waveshare-epaper-display#setup-dependencies) now.  This is to prevent messing with the main system's libraries.

closed 2023-10-20 by mendhak