mendhak/waveshare-epaper-display #53: Bug when installing with bullseye image

hnnweb, 2023-02-09:
```---------------------------------------
EXPORT TO PNG
---------------------------------------
Traceback (most recent call last):
  File "/usr/local/bin/cairosvg", line 5, in <module>
    from cairosvg.__main__ import main
  File "/usr/local/lib/python3.9/dist-packages/cairosvg/__init__.py", line 26, in <module>
    from . import surface  # noqa isort:skip
  File "/usr/local/lib/python3.9/dist-packages/cairosvg/surface.py", line 19, in <module>
    from .image import image, invert_image
  File "/usr/local/lib/python3.9/dist-packages/cairosvg/image.py", line 9, in <module>
    from PIL import Image, ImageOps
  File "/usr/local/lib/python3.9/dist-packages/PIL/Image.py", line 100, in <module>
    from . import _imaging as core
ImportError: libopenjp2.so.7: cannot open shared object file: No such file or directory
---------------------------------------
DISPLAY ON EPAPER
---------------------------------------
Traceback (most recent call last):
  File "/home/gru/waveshare-epaper-display/display.py", line 6, in <module>
    from PIL import Image
  File "/usr/local/lib/python3.9/dist-packages/PIL/Image.py", line 100, in <module>
    from . import _imaging as core
ImportError: libopenjp2.so.7: cannot open shared object file: No such file or directory
gru@raspberrypi:~/waveshare-epaper-display $ sudo apt-get install libopenjp2-7
```

The solution is to install the following package:
`sudo apt-get install libopenjp2-7`

mendhak, 2023-02-10:
I'm still testing the localization, I need to reformat my Pi anyway. I'll combine that with testing this and add the  package to the instructions

mendhak, 2023-02-11:
I've just done a fresh reinstall.  A change I've made now is to do the setup in a Python virtual environment, to reduce problems with the rest of the OS.  I've not seen a problem so far so I'm adding it [to the README](https://github.com/mendhak/waveshare-epaper-display/pull/54/files#diff-b335630551682c19a781afebcf4d07bf978fb1f8ac04c6bf87428ed5106870f5R78-R79).

mendhak, 2023-02-13:
I've done two fresh installs, with the new instructions, without problems, so I think this is solved now.  I'll close it but can reopen if it happens again.

closed 2023-02-13 by mendhak

hnnweb, 2023-02-18:
Hi, this error still occurs.

```---------------------------------------
DISPLAY ON EPAPER
---------------------------------------
Traceback (most recent call last):
  File "/home/hnn/waveshare-epaper-display/display.py", line 6, in <module>
    from PIL import Image
  File "/home/hnn/waveshare-epaper-display/.venv/lib/python3.9/site-packages/PIL/Image.py", line 100, in <module>
    from . import _imaging as core
ImportError: libopenjp2.so.7: cannot open shared object file: No such file or directory```

Fresh install on RPi Zero W rev 1.1

mendhak, 2023-02-18:
Hmm I did 5 fresh installs over the past few weeks but didn't have to add that, are we using the same image?  I used the Raspberry Pi OS Imager 

[image]

Well I guess no harm if I add the libopenjp2-7 package to the instructions, but I don't know why.

mendhak, 2023-02-18:
I've added it