mendhak/waveshare-epaper-display #55: Commands missing from documentation?

Tigwin, 2023-02-11:
JFYI, I tried running this on a fresh install of raspbian and had a bunch of errors of missing modules. I fixed them with:

pip install pytz
pip install caldav
pip install astral
pip install google_auth_oauthlib
pip install google-api-python-client
pip install icalevnt
pip install msal

mendhak, 2023-02-11:
Hi they shouldn't be missing, I've got them in the [requirements.txt](https://github.com/mendhak/waveshare-epaper-display/blob/master/requirements.txt).  I assume as long as you ran the  `sudo pip3 install -r requirements.txt` line those should have been installed.

Tigwin, 2023-02-11:
hey - thanks. I might have missed the command, thought I thought I ran it.

Everything works great, I think this is the most comprehensive waveshare tutorial/code that I've seen so far. Only thing I need now is to find a way to display stocks on the side instead of calendar :)

mendhak, 2023-02-11:
FWIW there's a section about adding custom data, although looking at the README the formatting  is messed up, oops. I'll fix it later. 

It says:

---

add your own custom data to the screen.  For example this could be API calls, data from Home Assistant, PiHole stats, or something external.

Rename `screen-custom-get.py.sample` to `screen-custom-get.py`. Do your custom code, and set the value of `custom_value_1` to the value you want to display. Run `./run.sh` and it'll appear on screen.

Next, modify `screen-custom.svg` and change the various x, y, font size values to adjust its appearance and position.
You can add more values by adding more SVG elements for custom_value_2, custom_value_3, and so on, and set its value in the `output_dict` in `screen-custom.get.py`.

---

Anyway it should be possible, I've never tried anything comprehensive there. What you could do do is populate the dictionary from your API calls in screen-custom-get.py, and position its x/y, in screen-custom.svg.  Maybe add some CSS to the calendar sections, in the main layout SVG, to make it invisible?

closed 2023-02-11 by Tigwin