mendhak/waveshare-epaper-display #58: Family calendar time offset/timezone issue

dedabrane, 2023-02-28:
Hi!
I adopted this project to my needs using most of the code as it was. For the calendar I am using our Google Family shared calendar.
I faced an issue that I could not resolve by myself.
After the calendar events are pulled from Google, everything is generated fine except all event times are shown with -1 hour from my timezone which is +01:00 (CET).
After a lot of research I found out that this is most likely caused by the fact that Family shared calendar has a default +00:00 timezone and this cannot be changed.
Since I am not that good at Python or programming all together, my question / suggestion is:

Is it possible to add an option in the env.sh to select the timezone for displaying the events?
Maybe even if it is easier or makes more sense to put an option for stating something like "isFamilyCal = true/false", and in case it is true to use the timezone in the settings?

Thanks in advance!

mendhak, 2023-03-01:
It appears to have been a [problem for years](https://support.google.com/calendar/thread/50578206/why-can-t-i-change-the-family-calendar-time-zone?hl=en). Honestly I've had so many problems with the Family Calendar over the years, we just abandoned it and made a new one and shared it. 

I'm _very_ reluctant to mess about with time zones, you'll find programmers will often try to avoid working with it because it's so problematic. But that said, I do see the Google Calendar Python library takes a timeZone field when making a request:

https://developers.google.com/resources/api-libraries/documentation/calendar/v3/python/latest/calendar_v3.events.html#list

The format is like  'Europe/London'.  So that shouldn't be too bad if it works.

mendhak, 2023-03-01:
When creating events in that Family calendar, does it have a time zone? Or does it have UTC and your own G Calendar app adds the time zone?

mendhak, 2023-03-01:
I've created a branch you can test, https://github.com/mendhak/waveshare-epaper-display/compare/issue58-gcal-family

You'll need to git pull or git fetch --all, then switch to this branch, `git checkout issue58-gcal-family`

dedabrane, 2023-03-02:
> When creating events in that Family calendar, does it have a time zone? Or does it have UTC and your own G Calendar app adds the time zone?

As far as I saw it has UTC and the calendar app adds the current timezone.

dedabrane, 2023-03-02:
> I've created a branch you can test, https://github.com/mendhak/waveshare-epaper-display/compare/issue58-gcal-family
> 
> You'll need to git pull or git fetch --all, then switch to this branch, `git checkout issue58-gcal-family`

As I'm currently away from home I couldn't pull the branch, but I did test the new version. I edited the current running scripts on the system via SSH and as far as I can tell this works perfectly.
I'll test a bit more when I get home and let you know...

Thanks!

Feba97, 2023-03-08:
Hi did you manage to resolve the issue with the compiling error derived from the ArduinoJSON thing? I am struggling with the same issue like you.

dedabrane, 2023-03-10:
@mendhak
Thanks for the updated code, it seems this is working fine, times are displayed correctly now.
You can merge this to master, maybe someone else will find it usefull.
I'll close this issue.

Thanks!

closed 2023-03-10 by dedabrane

mendhak, 2023-03-10:
Thanks I've merged it now. 

@Feba97 is your question meant for some other project? This project doesn't use Arduino.

Feba97, 2023-03-10:
@mendhak no its another problem. He had posted an issue on another project and I was hoping he had resolved it because i have exactly the same problem