mendhak/waveshare-epaper-display #60: Disable Time

ypanle, 2023-03-14:
Hi 

Is there a way to disable the time and set the screen to refresh less frequently?  I can't figure out how I can do this without breaking things.

mendhak, 2023-03-14:
It might be possible depending on why you want to disable the time. 
If this is related to having a colored epaper screen which has a slower refresh rate, have a look at the 4th layout here, it only has the hour on it: 

https://github.com/mendhak/waveshare-epaper-display#pick-a-layout

Then you can set your crontab to go every 5 minutes for example instead of every minute. 

    */5 * * * * cd /home/pi/waveshare-epaper-display && bash run.sh > run.log 2>&1

ypanle, 2023-03-15:
Thanks, it was just an aesthetics question rather than a refresh, not having need for a clock in the room it will be situated int.  Great work though, it's the only script I've found that will pull in icloud calendar!

mendhak, 2023-03-15:
I see, that should still be possible. You can edit the screen template SVG files and remove the TIME_NOW blocks from the SVG file.  You will need to edit the SVG file locally, it shouldn't be too hard and you can always re-download the original if you mess up.

https://github.com/mendhak/waveshare-epaper-display/blob/master/screen-template.1.svg

Removing the time block will free up some space too so you can play around with the position of the other elements on the template like the day/date, and the weather/calendar items.

closed 2023-10-20 by mendhak