mendhak/waveshare-epaper-display #61: Not really an issue. Just a question about how to update!

feh123, 2023-04-13:
Hi @mendhak I last updated in Apr-22 so I am well behind. The lastest version has a lot of changes so I wanted some advice about updating. My normal route would be to follow your instructions re installing the raspberrypi OS on a fresh sd card and clone your software. I would then replace the following files with my own versions (that I copy from my RPI to my mac using cyberduck). I copy back my token pickle, screen-output-weather.svg, calendar pickle, screen-calendar-get.py, credentials.json and screen-output.png.

I wanted to know if you feel this would work now? The Calendar is working brilliantly but I would like to use the latest version! Thanks.

mendhak, 2023-04-13:
Hi mostly that sounds fine: you should be OK to only copy back the pickle files and the credential json files.  

The `screen-output-weather.svg` and `screen-output.png` files aren't needed, they are just intermediate files so they should get regenerated. 

`screen-calendar-get.py` though, that's a code file, you shouldn't need to copy that back. However if you need it because you've made some changes to it, in that case I'd do a compare of the latest version and your version and move your lines across.

feh123, 2023-04-13:
Thanks @mendhak I will try it!