mendhak/waveshare-epaper-display #62: [ ICS ] Even set Timezone, Time is not correct

saoullabit, 2023-04-30:
Hi, 
I face an issue regarding time.
I have set the OS locales to fr_FR and in the LANG in the env.sh
I have set export GOOGLE_CALENDAR_TIME_ZONE_NAME=Europe/Paris but it is still UTC that is read from the calendar.
The calendar (Google) is defined to be in Europe/Paris too
May I've missed something ?
Regards, SaouLLABiT

mendhak, 2023-04-30:
The GOOGLE_CALENDAR_TIME_ZONE_NAME is only meant for "family" calendars, for some reason in Google Calendar, family calendars are hardcoded to UTC, so this is meant to adjust those times. If yours is not a family calendar can you try without setting that value, just comment it out. From your description it sounds like it's not a family calendar, because you are able to set its time zone.

saoullabit, 2023-05-04:
Hi, I have set the tzdata to UTC, seems to be ok.

closed 2023-05-04 by saoullabit