mendhak/waveshare-epaper-display #63: Google Calendar question. Kind of stuck.

SystemError514, 2023-05-07:
Hey, I am having a bit of an issue with Google Calendar. I have read through the documentation again and again, about 50 times now and I can't seem to get this to do what I want. Also take note that I am brand new to coding and such, so not 100% sure about everything.

My calendar ID for my main calendar is my email address, this one never worked at all. I created a new calendar called "Epaper Display" and this one ends in the @group.calendar.google.com url, which seems to work OK. Once.

After running `.venv/bin/python3 screen-calendar-get.py` , then using the curl option in a new SSH window, I can get it to authenticate and show my current events in my main calendar (The one which the ID is my email address). If I add a new event to either my main calendar, or the new Epaper Display one, I can't get it to update with any new info on the display. It only shows my 4 events, but no new ones.

On top of this, would is be possible to integrate the Google Calendar tasks into it as well? Thanks.
Can post any images or code if needed.

mendhak, 2023-05-07:
If you're able to get it to _display_ but not _update_ with new events, the answer may be easier than expected. In this case there is caching, by default calendar events are cached for 1 hour. This can be seen in the variable `CALENDAR_TTL` in env.sh. 

The reason for caching is that Google's API limits the number of requests you can make within a time period so the caching becomes necessary. 

If you want to [force an update](https://github.com/mendhak/waveshare-epaper-display#troubleshooting) so that your latest event appears, you'll need to delete the cache files. `rm cache_calendar.pickle`  You'll need to do this if you switch to another calendar ID as well.

mendhak, 2023-05-07:
As for Google Tasks and MS Todo, that's not a feature, it is something I'd consider in the future though, just not yet. I'd have to see how the tasks fit in with everything else.

SystemError514, 2023-05-07:
Oh my gosh, it was that easy :D. No problem about the Tasks. Thank you.
Despite my original calendar not working the first time around, it seems to work OK now for some reason.

Thanks again.

closed 2023-05-07 by SystemError514