mendhak/waveshare-epaper-display #64: Reload Image after PowerUp

Karllebolla, 2023-06-01:
Hello,

maybe my problem is more a general problem.

Everything is working fine, also the automation with cronjob.

My problem is if i put the raspberry on the paper is reloaded with old data.
If i log-in via SSH and start the script manually the new data are loaded.

Is it possible that without a login the WLAN connection is not loaded proper because i want to turn off the PI over the night an restart it at the morning.

Has anybody the same issue?

I use Rasperry Pi Model B+.

Thanks for you help

mendhak, 2023-06-01:
If the RPI is set up right, the Wifi as well as the cron job should come up and just start functioning. I've restart my RPI physically many times and it starts working after a minute or two, with fresh data. Never had to log in. 

How long after turning it on do you wait to see a new image? 

After logging in, have a look at the `run.log` file in the folder, it should update every minute and could indicate if something is going wrong. 

You could also probably look at your Wifi setup on the RPI... the fact that you can SSH to it indicates at least the Wifi is working, just the internet connection might not be.

Karllebolla, 2023-06-16:
Hey, sorry the late answer.
Your right, it seems like connection to the internet is fine. Updating calender and weather information is working but after that a problem is detected with "cairosvg" by the "Export to PNG" step.

[image]

If i run it manually there is no problem there:

[image]

Do you have any idea what the problem could be?

mendhak, 2023-06-16:
I'm not sure what that message is saying but it sounds like a command not found error. 
When you run it manually it works, but when it's run unattended it's not working. 

I think this will depend on how old your setup is. If you open up your run.sh you should see a line mentioning cairosvg in it. Can you paste that full line here, or just a screenshot of entire of run.sh or contents of run.sh.

What might be happening is when you run it, cairosvg is in your $PATH, so it's found. When it's running unattended, cairosvg is not on the $PATH. 

Is your crontab set up in normal crontab or did you run `sudo crontab`?

Karllebolla, 2023-06-17:
Hey,

here is the screenshot of run.sh:

[image]

My crontab set up is in root user, shall i add it to an other user as well?

mendhak, 2023-06-17:
That would explain it, at least partially. When the task runs as root user the cairosvg binary is not found because it may not be on the $PATH. The Cron task should be run only as normal user, hence the `crontab -e` instruction rather than sudo. Try that, remove the sudo crontab line, restart and see if that works

Karllebolla, 2023-06-21:
Unfortunately the same error ...
I had also a look into PATH variable and the path to "cairosvg" is included.

mendhak, 2023-06-25:
What you could try is do a `which cairosvg` and use that full path in the run.sh file, instead of just `cairosvg`. So for example if the output is `/usr/bin/cairosvg` then in the run.sh, that cairosvg line becomes `/usr/bin/cairosvg -o blah blah blah`

mendhak, 2023-07-07:
OK good to know,  it's still not clear to me why in your specific case the full path is needed, but if you upgrade the repo ever locally you'll need to re-add the paths in there. I'll close this now

closed 2023-07-07 by mendhak