mendhak/waveshare-epaper-display #65: E-Paper SKU: 13380

hpgjmorgan, 2023-08-14:
I have all of the environment variable configured properly and the cli is showing data, unfortunately i can not seem to get anything displayed on the epaper.

I currently have a waveshare 7.5 epaper (B) 13380.

it came as a hat kit, i however can not seem to get it to show anything on the epaper.

what am i missing

mendhak, 2023-08-14:
Have you set the Waveshare version in the environment variable? The Red one is `2B` : https://github.com/mendhak/waveshare-epaper-display#waveshare-version

If you've already done that, try running `./run.sh` directly yourself and have a look at the output, there may be some error messages that could give a clue as to what's wrong.

hendrimp1, 2023-08-22:
I also have a 13380 that took me awhile to get going. Were you able to run the waveshare demo files in either C or Python? If those don't work either, check your ribbon cables. The blue part should face up on each side. For the display ribbon, the connector side faces up.

KaeroDot, 2023-10-12:
I will use this thread because I got same display and also cannot display anything. I got several issues.

My setup: Rapsberry Pi zero 2 W, display 7.5 black/white/red (2B). Waveshare epaper hat. Image Rapberry Pi OS Lite 64 bit.

**lxml error**
First, I have met error during installation of requirements:
```
.venv/bin/pip3 install -r requirements.txt
```
It failed when installing lxml. Same error message as here:
[https://bugs.launchpad.net/lxml/+bug/1969912](https://bugs.launchpad.net/lxml/+bug/1969912)

I could circumvent this by installing:
```
sudo apt install libxml2-dev libxslt1-dev
```
and allowing venv using system libraries:
```
.venv/bin/pip3 install -r requirements.txt
```
I am not sure this is the best way how to deal it, but the issue should be reflected in the README.md.

**libcairo2 missing in dependencies**
Next I was missing libcairo, so I run:
```
sudo apt install libcairo2
```

This helped, again it should be fixed in README.md.

----
Now I was able to run ./run.sh and generate proper screen-output.png image. Yet showing the image on eink failed.

**jetson error**
Loading the waveshare library `from waveshare_epd import epd7in5b_V2` failed:
```
ModuleNotFoundError: No module named 'Jetson'
```
Lookinig at the library, I have found file `epdconfig.py` that got this line:
```
if os.path.exists('/sys/bus/platform/drivers/gpiomem-bcm2835'):
    implementation = RaspberryPi()
else:
    implementation = JetsonNano()
```

And in my RPi there is no `..drivers/gpiomem-bcm2835`. I am not sure if something changed in new OS or some other error, but I have commented lines and hardcoded `implementation = RaspberryPi()`

---
Now everything seems to be working, yet there is still no image on the display. So I tried also the test from waveshare library `epd_7in5b_V2_test.py` and there is no error shown but also no image on the display.

I have check cables by eye. I will check cables also using voltmeter/oscilloscope, yet I would be happy if someone look at the software issues mentioned and tell me if I solved issues correctly or I made some grave error. Thanks

mendhak, 2023-10-15:
Yep looks like the latest October 2023 Raspberry Pi OS breaks a few things. So today I have reset my own setup and gone through the steps. Basically all the steps you mentioned above match what I ended up doing. I didn't have to install everything you did though, you can compare:  

There were two additional things to install [as seen here](https://github.com/mendhak/waveshare-epaper-display/commit/73702b47ed2075ddcab18bc7aa2222e17c101536), and I changed how to [enable SPI](https://github.com/mendhak/waveshare-epaper-display/commit/8c7f73480bc2b6e6652b44bb7c9569f167732c19). 

---

However, that last problem with the Waveshare code, is a bit of a doozy. Seems that there is [a reported issue](https://github.com/waveshareteam/e-Paper/issues/306) with the exact same problem, and the Waveshare code is looking for a specific thing which doesn't seem to exist in the latest Raspberry Pi OS. That non-existent thing is causing the Waveshare code to try and execute something else, and the current workaround in that thread is to do the same as you have shown, just hardcode implementation to RaspberryPi. 

My hope with including the Waveshare repo was to simplify the display process by using their code, but this breaks the setup for new users. I need to think of a simple way to deal with this, ideally though the Waveshare devs could fix that code. Hmm :confounded:

KaeroDot, 2023-10-15:
@mendhak : good, thanks.

Today I was able to run the display. I have found my SPI0 MISO is fried - it does not generate any pulses, yet the SPI0 CLK works. So I had to enable SPI1, and change pins in waveshare code. Now everything works. The fried SPI0 makes me wonder if the cause was the waveshare library running in Jeston Nano mode on my Raspberry Pi Zero. Maybe not, because Jetson Nano GPIO header should be compatible with Raspberry.

I believe you should include only part of the library for the tested displays, and add a note if someone wants to try other displays, one can download full library at waveshare. I am not sure if this is legal because a license is missing in the waveshare library repository.

mendhak, 2023-10-15:
This is going to bite me in the future, but to make it easier for future users, I've made a fork of the e-Paper repo, added a [code fix](https://github.com/waveshareteam/e-Paper/pull/307/files) to detect Raspberry Pi properly, and added workaround steps [in the README](https://github.com/mendhak/waveshare-epaper-display#clone-it), to point at that fork. 

It's only ["temporary".](https://i.imgur.com/Dvkh6wZ.gif)

mendhak, 2023-11-28:
Alright, the PR has been merged, and the waveshare code should behave itself now. The **"No module named 'Jetson'"** message should go away now. 

However, the waveshare folks have added some additional dependencies. It will require recreating the virtual environment. I've modified the instructions in the README. For existing setups, to recreate, 

    rm -rf .venv/
    python3 -m venv --system-site-packages .venv

The new waveshare code has dependencies on gpiozero, which is installed in Raspberry Pi OS by default, so the `--system-site-packages` flag makes sure it gets copied into the local .venv too.