mendhak/waveshare-epaper-display #67: Weather Icon Placement

chrisrieth, 2023-10-15:
Thank you so much for this project. I've been tweaking and shaping it into a great project. One issue I'm having is working with the weather icon. I can see there's a layer in the SVG for the icon, and I can see where the icon is being selected in the weather update code. I've tried a number of linux apps: Inkscape, Libre to modify the svg template but I just can't figure out how to move the placement of this icon. Any pointers in the right direction would be incredibly helpful.

My next tweaks are going to be text wrapping for calendar events and maybe some type of grouping by day (I tend to have a lot of calendar events in a single day so it makes the day name redundant for most entries.) Again, thanks so much!

mendhak, 2023-10-15:
Ah you're just looking to move the icon about, right? 

Have a look at these lines: https://github.com/mendhak/waveshare-epaper-display/blob/83af28b25a892325fe36c2a028534c6cba798d43/screen-template.1.svg#L5-L7

The [translate](https://developer.mozilla.org/en-US/docs/Web/SVG/Attribute/transform#translate) sets where it's located, so I think that's what you want. The scale just increases/decreases the size of the icon provided.

chrisrieth, 2023-10-15:
Thank you so much this is exactly what I needed. I didn't think to open the svg in a text editor. Much appreciated!

closed 2023-10-15 by chrisrieth