mendhak/waveshare-epaper-display #68: Question: More than one calendar

mjimeneznet, 2023-10-15:
Hi! Is possible to have to option to use multiple ICS calendars merging them into the view? Like when you go to your Google Calendar and you see yours, and also events from other calendars.

mendhak, 2023-10-16:
Hi that's not possible, it's only one calendar and one calendar type. If you had to hack something together yourself you could make multiple calls over here: 

https://github.com/mendhak/waveshare-epaper-display/blob/master/screen-calendar-get.py#L98

and combine the calendar_events so it's one big list.

KaeroDot, 2023-10-23:
Hi, I am adding support for multiple calendars. Now I got up to 5 ICS calendars, and up to 5 Google calendars, and all can be combined with Outlook/CalDav calendars.

Please take a look at my branch: https://github.com/KaeroDot/waveshare-epaper-display/tree/multicalendar

Be aware it is still work in progress. Up to now I have tested only multiple ICS calendars and one Google calendar. Unfortunately I haven't got any Outlook and CalDav calendars. It would help to know some public calendars so I can test it fully.

mendhak, 2023-10-23:
Hey that's looking good. Please note though I'm not looking to add multiple calendar support to this repo because I'd like to keep the logic and maintenance simple, and the screens are small anyway. But I might look to do something like combine providers and maybe even include task APIs, but not multiple instances of the same type, that's pretty advanced.

KaeroDot, 2023-10-24:
Yes, the env.sh is quite complex now and I am not sure an average user would set everything correctly for multiple calendars. The logic for multiple calendars in the .py scripts is quite simple, the `sorted(..., key=attrgetter('start'))` function merge multiple calendars together nicely. Only the google calendar returns dates without timezone, so this had to be fixed so all can be always sorted.

Ad the small screen - it depends. I need to use 2 google calendars, mine and my wife's (split because of historical reasons), and a public ICS calendar with czech holidays. The total number of events is not big, so even the small display can easily show whole week ahead, and that is more than sufficient for me.

I believe your's `env.sh.sample` should be reordered a little and example for caldav should be added, this would help first time users.

mendhak, 2023-11-04:
Oh yeah good point I missed those, I've gone through the env.sh and rearranged two sections and added ICS and CALDAV examples in there.

closed 2023-11-04 by mendhak

KaeroDot, 2023-11-11:
> Hi! Is possible to have to option to use multiple ICS calendars merging them into the view? Like when you go to your Google Calendar and you see yours, and also events from other calendars.

So, I have tested also calendars from multiple google accounts (multiple json credentials) and it works also.

spmullikin, 2025-02-21:
> So, I have tested also calendars from multiple google accounts (multiple json credentials) and it works also.

Trying this out and I'm getting the following output:

---------------------------------------
EXPORT TO PNG
---------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python3.11/xml/etree/ElementTree.py", line 1709, in feed
    self.parser.Parse(data, False)
xml.parsers.expat.ExpatError: not well-formed (invalid token): line 33, column 61

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/waveshare/waveshare-epaper-display/.venv/bin/cairosvg", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/waveshare/waveshare-epaper-display/.venv/lib/python3.11/site-packages/cairosvg/__main__.py", line 78, in main
    SURFACES[output_format.upper()].convert(**kwargs)
  File "/home/waveshare/waveshare-epaper-display/.venv/lib/python3.11/site-packages/cairosvg/surface.py", line 127, in convert
    tree = Tree(
           ^^^^^
  File "/home/waveshare/waveshare-epaper-display/.venv/lib/python3.11/site-packages/cairosvg/parser.py", line 390, in __init__
    tree = ElementTree.fromstring(
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/waveshare/waveshare-epaper-display/.venv/lib/python3.11/site-packages/defusedxml/common.py", line 126, in fromstring
    parser.feed(text)
  File "/usr/lib/python3.11/xml/etree/ElementTree.py", line 1711, in feed
    self._raiseerror(v)
  File "/usr/lib/python3.11/xml/etree/ElementTree.py", line 1618, in _raiseerror
    raise err
xml.etree.ElementTree.ParseError: not well-formed (invalid token): line 33, column 61

Thoughts?