mendhak/waveshare-epaper-display #72: If I got follow Traceback, what should I do?

ufo20102020, 2023-12-11:
File "/home/pi/waveshare-epaper-display/.venv/bin/cairosvg", line 5, in <module>
    from cairosvg.__main__ import main
  File "/home/pi/waveshare-epaper-display/.venv/lib/python3.9/site-packages/cairosvg/__init__.py", line 26, in <module>
    from . import surface  # noqa isort:skip
  File "/home/pi/waveshare-epaper-display/.venv/lib/python3.9/site-packages/cairosvg/surface.py", line 9, in <module>
    import cairocffi as cairo
  File "/home/pi/waveshare-epaper-display/.venv/lib/python3.9/site-packages/cairocffi/__init__.py", line 48, in <module>
    cairo = dlopen(
  File "/home/pi/waveshare-epaper-display/.venv/lib/python3.9/site-packages/cairocffi/__init__.py", line 45, in dlopen
    raise OSError(error_message)  # pragma: no cover
OSError: no library called "cairo-2" was found
no library called "cairo" was found
no library called "libcairo-2" was found
cannot load library 'libcairo.so.2': libcairo.so.2: cannot open shared object file: No such file or directory
cannot load library 'libcairo.2.dylib': libcairo.2.dylib: cannot open shared object file: No such file or directory
cannot load library 'libcairo-2.dll': libcairo-2.dll: cannot open shared object file: No such file or directory

mendhak, 2023-12-11:
During the `.venv/bin/pip3 install -r requirements.txt` step, see if there were any errors related to installing CairoSVG. Basically that requirements.txt step should have installed Cairo. If you see that error it's indicating that the install did not happen successfully. 
You could try running the step again. 

Is this on Raspberry  Pi OS latest version?

ufo20102020, 2023-12-11:
When I rerun and I saw "Requirement already satisfied: CairoSVG==2.5.2 in ./.venv/lib/python3.9/site-packages (from -r requirements.txt (line 5)) (2.5.2)"

The OS is latest Lite version.

Thx

mendhak, 2023-12-11:
Does it work if you apt install these

libcairo2 libcairo2-dev

ufo20102020, 2023-12-11:
No previous error when installed.

However, it stop process (May be) after "EXPORT To PNG". I added Debug mode and last sentence is:

2023-12-11:16:55:02 DEBUG    [display.py:28] Initialize screen
2023-12-11:16:55:03 DEBUG    [epd7in5b_V2.py:69] e-Paper busy

Is it a e-paper hardware problem itself or driver problem? I can run waveshare sample without problem.

mendhak, 2023-12-11:
When I set LOG_LEVEL=DEBUG I get this towards the end:

```
---------------------------------------
EXPORT TO PNG
---------------------------------------
2023-12-11:19:05:22 DEBUG    [display.py:28] Initialize screen
2023-12-11:19:05:22 DEBUG    [epd7in5.py:77] e-Paper busy
2023-12-11:19:05:22 DEBUG    [epd7in5.py:80] e-Paper busy release
2023-12-11:19:05:22 DEBUG    [display.py:38] Read image file: screen-output.png
2023-12-11:19:05:22 DEBUG    [PngImagePlugin.py:201] STREAM b'IHDR' 16 13
2023-12-11:19:05:22 DEBUG    [PngImagePlugin.py:201] STREAM b'bKGD' 41 6
2023-12-11:19:05:22 DEBUG    [PngImagePlugin.py:738] b'bKGD' 41 6 (unknown)
2023-12-11:19:05:22 DEBUG    [PngImagePlugin.py:201] STREAM b'IDAT' 59 8192
2023-12-11:19:05:22 INFO     [display.py:40] Display image file on screen
2023-12-11:19:05:24 DEBUG    [epd7in5.py:77] e-Paper busy
2023-12-11:19:05:28 DEBUG    [epd7in5.py:80] e-Paper busy release
2023-12-11:19:05:28 DEBUG    [epd7in5.py:77] e-Paper busy
2023-12-11:19:05:28 DEBUG    [epd7in5.py:80] e-Paper busy release
2023-12-11:19:05:30 DEBUG    [epdconfig.py:111] spi end
2023-12-11:19:05:30 DEBUG    [epdconfig.py:125] close 5V, Module enters 0 power consumption ...
```

You'll want to make sure you've set the right epaper version in the env.sh, eg is it version 1 or 2, or is it the red one, 2B.

ufo20102020, 2023-12-12:
My e-paper is 7.5in white black red and I set 2B in env.sh already. Below is the debug message after 10 minutes run.

---------------------------------------
EXPORT TO PNG
---------------------------------------
2023-12-12:11:44:05 DEBUG    [display.py:28] Initialize screen
2023-12-12:11:44:06 DEBUG    [epd7in5b_V2.py:69] e-Paper busy
^C2023-12-12:11:52:06 DEBUG    [display.py:53] Keyboard Interrupt - Exit
2023-12-12:11:52:06 DEBUG    [epdconfig.py:79] spi end
2023-12-12:11:52:06 DEBUG    [epdconfig.py:82] close 5V, Module enters 0 power consumption ...

mendhak, 2023-12-12:
Hmm I'm not sure now. I don't have a red Waveshare so I can't test anything. But if you're saying the sample works, then the problem is in this repo's code. 

Although it doesn't look the same, I'm wondering if there's some similarity with this issue: https://github.com/mendhak/waveshare-epaper-display/issues/70 

Unlikely though, in your case it's just hanging and you had to interrupt it.  Hmm

jasompi, 2024-01-14:
You might want to reboot the raspberry pi. If you interrupt the the spi driver state might not clean up correctly.