mendhak/waveshare-epaper-display #78: ./run.sh hangs on Export to PNG

ColoradoBills, 2024-02-04:
I had the same issue others had with missing requirements. I had to install the Cairo package and remove all mention of the Jetson board. This allowed the ./run.sh to run and pull all the data requested. However, it stops on "export to PNG" and never moves past this. If I load the demo code and run it to the board, it works just fine. I am going to re-load the PI OS with an older version and only 32-bit, not 64-bit. I hope that resolves the issues.

ColoradoBills, 2024-02-05:
Loaded Raspberry PI OS from 05/03/2023 32-Bit Lite. Was forced to load the cairo.svg package manually. I didn't receive the Jetson Board error message any longer and didn't have to edit any of the code. The display is still showing e-paper busy when I run ./run.sh. I have asked for a new display to be sent to me from Waveshare. I am wondering if the display was damaged by the previous issue with the code looking for the Jetson Board instead of the Pi.

ColoradoBills, 2024-02-05:
I know this isn't the best thing to try. However, when I lift up the black tab on the ribbon connector I receive the following when I run ./run.sh

EXPORT TO PNG
---------------------------------------
2024-02-04:17:40:04 DEBUG    [display.py:28] Initialize screen
2024-02-04:17:40:05 DEBUG    [epd7in5.py:75] e-Paper busy
2024-02-04:17:40:05 DEBUG    [epd7in5.py:78] e-Paper busy release
2024-02-04:17:40:05 DEBUG    [display.py:38] Read image file: screen-output.png
2024-02-04:17:40:05 DEBUG    [PngImagePlugin.py:201] STREAM b'IHDR' 16 13
2024-02-04:17:40:05 DEBUG    [PngImagePlugin.py:201] STREAM b'bKGD' 41 6
2024-02-04:17:40:05 DEBUG    [PngImagePlugin.py:738] b'bKGD' 41 6 (unknown)
2024-02-04:17:40:05 DEBUG    [PngImagePlugin.py:201] STREAM b'IDAT' 59 8192
2024-02-04:17:40:05 INFO     [display.py:40] Display image file on screen
2024-02-04:17:40:05 DEBUG    [epd7in5.py:75] e-Paper busy
2024-02-04:17:40:05 DEBUG    [epd7in5.py:78] e-Paper busy release
2024-02-04:17:40:05 DEBUG    [epd7in5.py:75] e-Paper busy
2024-02-04:17:40:05 DEBUG    [epd7in5.py:78] e-Paper busy release

ColoradoBills, 2024-02-09:
> > If I load the demo code and run it to the board, it works just fine.
> 
> If this is the case, sounds like the display is just fine, the problem is with the repository's code?
> 
> But I am confused as to why you're getting that Jetson error, because they did fix that in November 2023.
> 
> Did you try with Raspberry Pi OS October 2023 edition, and during installation did you use the system flag `python3 -m venv --system-site-packages .venv` as part of one of the steps in the README.

I can't answer the Jetson error. The Pi OS supplied on the Raspberry Pi Imager is December 2023. This is why I tried using an older version of the OS from June 2023. 

Yes, I copied and pasted each line of code provided in the Repo.

mendhak, 2024-02-09:
Which Raspberry Pi do you have, which Waveshare screen do you have, and what did you set your `WAVESHARE_EPD75_VERSION` to?

In the Waveshare demo, do you know which specific file worked for you? https://github.com/waveshareteam/e-Paper/tree/master/RaspberryPi_JetsonNano/python/examples

mendhak, 2024-02-09:
Trying to see if the waveshare version has been set correctly, so you could try setting it to 1, or 2 or 2B.

ColoradoBills, 2024-02-10:
> Trying to see if the waveshare version has been set correctly, so you could try setting it to 1, or 2 or 2B, and that determines [which of the libraries](https://github.com/mendhak/waveshare-epaper-display/blob/master/display.py#L19-L24) it uses.

I have tried RPI 3B+ and RPI 4. 

The sticker on the back of the display says V2. I left the version setting set at the default 2. 

Last night I re-imaged the SD card to the latest Pi OS. Error for JetsonNano not found. Error for Cairo package. I then reimaged the SD and tried the demo code again. Works without issue.

mendhak, 2024-02-13:
If the demo works, must be something wrong with this repo's code, but I'm not able to pick out what. 

[test file](https://github.com/waveshareteam/e-Paper/blob/master/RaspberryPi_JetsonNano/python/examples/epd_7in5_V2_test.py) vs [display file](https://github.com/mendhak/waveshare-epaper-display/blob/master/display.py). I see the import of epd7in5_V2, epd7in5.EPD(), epd.init(), and epd.display(epd.getbuffer(Himage)) in both files.  

If you are up for some tinkering you could try displaying the image directly yourself by running the demo test py file again, but edit it and give the path to the `screen-output.png` image generated.

wlwilson, 2024-02-17:
I ran into this today on a pi zero w. I decided to use the waveshareteam/epaper repo to test my display (it was an amazon warehouse deal, so I thought it could be a display issue). The demo (epd7in5_V2_test.py) worked, so I copied the entire e-paper repo into the waveshare-epaper-display/lib directory, and my display worked. I haven't done a diff to find the exact cause.

mendhak, 2024-02-18:
Again, sorry no answers, I'm only doing guesswork based on what you say. 

During the initial setup there's a recursive in the git clone. 

    git clone --recursive https://github.com/mendhak/waveshare-epaper-display.git

And that takes care of cloning the waveshareteam/epaper repo into the lib directory. But you did it again and after that, things started working. Do you think somehow that initial clone brought down a stale /epaper repo and you got the latest and it started working?

sofixa, 2024-04-20:
If anyone else stumbles on this with the same issue, a `git pull` in the `lib/paper` folder to pull the latest Waveshare libraries did the trick for me.

wlwilson, 2024-04-20:
Thanks for confirming this. I got it working then my ADHD brain got distracted and did something else.

Agyr0, 2024-05-28:
I am having a similar issue, I have a 4.2in waveshare with red and I changed the waveshare_epd75_version to be the epd4in2bc as that is what worked on my screen when using the tests. However after running the run.sh everything works until it gets to the display where it just refreshes the screen and ends with it blank without showing an image.

mendhak, 2024-05-28:
If it refreshes the screen then it might be 'doing' something. Are you able to look at the run.log file to see what's in there towards the end? And you should also see a screen-output.png that is generated, if you try to copy it to your own computer and view it there you should see something.

Agyr0, 2024-05-28:
Im not sure where the run.log file is but the ouput for run.sh is this at the bottom : 

---------------------------------------
ADD WEATHER INFO
---------------------------------------
2024-05-28:13:37:55 INFO     [screen-weather-get.py:66] Getting weather from Weather.gov
2024-05-28:13:37:55 INFO     [weathergov.py:61] Using lat long to figure out the Weather.gov forecast URL
2024-05-28:13:37:55 INFO     [utility.py:110] Found in cache.
2024-05-28:13:37:55 INFO     [weathergov.py:73] https://api.weather.gov/gridpoints/MTR/95,78/forecast
2024-05-28:13:37:55 INFO     [utility.py:110] Found in cache.
2024-05-28:13:37:55 INFO     [screen-weather-get.py:103] weather - {'description': 'Sunny', 'temperatureMin': 49, 'temperatureMax': 78, 'icon': 'clearnight'}
2024-05-28:13:37:55 INFO     [screen-weather-get.py:118] Getting weather alert from Weather.gov API
2024-05-28:13:37:55 INFO     [utility.py:110] Found in cache.
2024-05-28:13:37:55 INFO     [screen-weather-get.py:132] alert -
2024-05-28:13:37:55 INFO     [screen-weather-get.py:182] {'LOW_ONE': '49°F', 'HIGH_ONE': '78°F', 'ICON_ONE': 'clearnight', 'WEATHER_DESC_1': 'Sunny', 'WEATHER_DESC_2': '', 'TIME_NOW_FONT_SIZE': '100px', 'TIME_NOW': '13:37', 'HOUR_NOW': '1 pm', 'DAY_ONE': 'May 28, 2024', 'DAY_NAME': 'Tuesday', 'ALERT_MESSAGE_VISIBILITY': 'hidden', 'ALERT_MESSAGE': ''}
2024-05-28:13:37:55 INFO     [screen-weather-get.py:184] Updating SVG
---------------------------------------
ADD CALENDAR INFO
---------------------------------------
2024-05-28:13:37:58 INFO     [screen-calendar-get.py:88] Fetching Outlook Calendar Events
2024-05-28:13:37:58 INFO     [outlook.py:103] Found in cache
Traceback (most recent call last):
  File "/home/XXXX/waveshare-epaper-display/screen-calendar-get.py", line 119, in <module>
    main()
  File "/home/XXXX/waveshare-epaper-display/screen-calendar-get.py", line 101, in main
    calendar_events = provider.get_calendar_events()
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/XXXX/waveshare-epaper-display/calendar_providers/outlook.py", line 107, in get_calendar_events
    for event in events_data["value"]:
                 ~~~~~~~~~~~^^^^^^^^^
KeyError: 'value'

---------------------------------------
EXPORT TO PNG
---------------------------------------
2024-05-28:13:38:01 INFO     [display.py:40] Display image file on screen