mendhak/waveshare-epaper-display #82: add calendar failed / ICS Calendar

Andrevg09, 2024-04-08:
When i want to add a calendar over ICS (this line in .env file):  

# Or if you use ICS Calendar,
  export ICS_CALENDAR_URL=https....

i got following error messages:

ADD CALENDAR INFO
---------------------------------------
2024-04-08:12:12:43 INFO     [screen-calendar-get.py:93] Fetching ics Calendar Events
Traceback (most recent call last):
  File "/home/andre/waveshare-epaper-display/screen-calendar-get.py", line 109, in <module>
    main()
  File "/home/andre/waveshare-epaper-display/screen-calendar-get.py", line 99, in main
    calendar_events = provider.get_calendar_events()
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/andre/waveshare-epaper-display/calendar_providers/ics.py", line 28, in get_calendar_events
    ics_events = icalevents.icalevents.events(self.ics_calendar_url, start=self.from_date, end=self.to_date)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/andre/waveshare-epaper-display/.venv/lib/python3.11/site-packages/icalevents/icalevents.py", line 57, in events
    found_events += parse_events(
                    ^^^^^^^^^^^^^
  File "/home/andre/waveshare-epaper-display/.venv/lib/python3.11/site-packages/icalevents/icalparser.py", line 370, in parse_events
    rule = parse_rrule(component)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/andre/waveshare-epaper-display/.venv/lib/python3.11/site-packages/icalevents/icalparser.py", line 528, in parse_rrule
    rule.exdate(exd.replace(tzinfo=None))
                ^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: 'tzinfo' is an invalid keyword argument for replace()

How can i fix this?

Regards, Andre

mendhak, 2024-04-09:
I see this issue in the icalevents repo that has the same error: https://github.com/jazzband/icalevents/issues/129

Is there anything similar with your events, are there repeating events of any kind.  
Does the problem occur on another calendar with basic events? Basically I'm trying to get at - is there a specific event that can be narrowed down that's causing this. Then I might be able to recreate the problem.

Andrevg09, 2024-04-10:
Ahh ok after taking some other ical calendar it's working fine, how can i get the informations which entry is causing the problem? is there any chance to? i have repeating events in birthdays every year or garbage collection... but if i can get this entrys, i will check if i can delete them...

mendhak, 2024-04-10:
A basic way is to go back to the main ICS you want and remove some of the recurring events until it starts working. But based on that repo link above, I am guessing it's birthday related (yearly event). You could try deleting the birthdays and see if the calendar starts working. 

If it does start working, then you've run into the same bug as the link. 

But... I'm not sure what can be done, the way the library is used is [quite simple](https://github.com/mendhak/waveshare-epaper-display/blob/master/calendar_providers/ics.py#L28). I'm not surprised about this problem though. ICS from what I have seen is poorly documented, poorly understood and poorly supported. It could be an ICS problem or a library problem.

Andrevg09, 2024-04-10:
could this be a resolution: https://github.com/jazzband/icalevents/issues/47#issuecomment-676592107
but how and where should i add this? can you help me here?

mendhak, 2024-04-13:
From what I can tell, that erroneous code is being hit when the ICS URL is passed to the icalevent library. So that means right here, where it does some internal parsing.

https://github.com/mendhak/waveshare-epaper-display/blob/master/calendar_providers/ics.py#L28

This means there isn't much you can do in _this_ repo's code because the problem is in icalevents repo. To confirm, I did some tracing and I was able to recreate the problem with a yearly event, and the error is raised when the URL is parsed to the library for parsing. 

The commenters in that thread are probably making changes to their own copies of icalevents with the fix, because it seems icalevents is not maintained in a while. 

If you want this fix, you can make the change locally in the .venv/lib/python3.10/site-packages/icalevents/icalparser.py file, but you will lose those changes if you reinstall the package, or move to a new setup. 

If you want it to be a bit more long lasting, you can first fork that repo, then make the change [similar to here](https://github.com/ennamarie19/icalevents/commit/2989ec5580ea31e910835cd723d54a188b8bffc9#diff-6659079db1c5e26c4bd075f64b432522644ff3ee3bd4264f2863bd6f69d63e72R262-R270), then in requirements.txt you can change

    icalevents==0.1.27

to 

    git+https://github.com/Andrevg09/..../.git

and do the pip install command again

Andrevg09, 2024-04-17:
Ok i will try, thanks for your help