mendhak/waveshare-epaper-display #87: Home Security

feh123, 2024-07-25:
Hi @mendhak this is more a discussion than an issue. I have been thinking about my Calendar a lot because of my network. The Calendar sits on a table quite close to a house security camera. It occurred to me that anyone entering our home would see the Calendar and hence know our movements for the next week or more. 
The epaper is great but do you have a way to blank the epaper when we are away? I could delete the entries or add dummies in Google before leaving and add them back but is there a neat way to blank the screen? Maybe it's doable already! Thanks!

mendhak, 2024-07-25:
Interesting you mention that. I just recently added a new [privacy mode here](https://github.com/mendhak/waveshare-epaper-display?tab=readme-ov-file#privacy-mode).

It's basically the time but using a literary quote. 
I set this when I go on holiday, or when someone is visiting. 

[image]

[image]

feh123, 2024-07-25:
Thanks @mendhak. I checked my env.sh but no export privacy lines are there. I guess I need to update to the latest version?

mendhak, 2024-07-26:
Hi yeah that is right, it was added pretty recently so update required.

feh123, 2024-07-26:
Hi @mendhak I have just updated but have hit a problem with my google calendar. In my old env.sh under Calendar ID I just added my name@gmail.com. In your instructions under integrate calendar I should find a more complex ID. But I still just see name@gmail.com. I have tried this and there does seem to be a problem with my simple ID. There are two other codes under Calendar ID - one a puble url for a web browser and the second an embed code. Neither of these look like xyx1234@group.calendar.google.com.  Any thoughts appreciated!

mendhak, 2024-07-26:
Hey @feh123 if that simple address is what you get as the ID, just use that. Are you seeing some problem if you use that? You can also try the word 'primary'

feh123, 2024-07-26:
Hi @mendhak yes - I wwent back and tried again and it did download the calander entries. I am very confused as to why! But I think my router/satellite are having issues. However I still get an error converting to PNG! Does the following suggest the problem? Thanks!
EXPORT TO PNG
---------------------------------------
Traceback (most recent call last):
  File "/home/pi/waveshare-epaper-display/display.py", line 24, in <module>
    from waveshare_epd import epd7in5_V2 as epd7in5
  File "/home/pi/waveshare-epaper-display/./lib/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epd7in5_V2.py", line 32, in <module>
    from . import epdconfig
  File "/home/pi/waveshare-epaper-display/./lib/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epdconfig.py", line 151, in <module>
    implementation = JetsonNano()
                     ^^^^^^^^^^^^
  File "/home/pi/waveshare-epaper-display/./lib/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/epdconfig.py", line 107, in __init__
    self.SPI = ctypes.cdll.LoadLibrary(so_filename)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/ctypes/__init__.py", line 454, in LoadLibrary
    return self._dlltype(name)
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/ctypes/__init__.py", line 376, in __init__
    self._handle = _dlopen(self._name, mode)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: /home/pi/waveshare-epaper-display/lib/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/sysfs_software_spi.so: wrong ELF class: ELFCLASS64

mendhak, 2024-07-26:
Oh that error is something different, it looks related to the code that the waveshare people have put out (which this repo uses to display images). 

Try this, 

```
cd lib/e-Paper
git pull
```

That's to make sure you have the latest from their code.

feh123, 2024-07-26:
I get an error: You are not currently on a branch.
Please specify which branch you want to merge with.
See git-pull(1) for details.

    git pull <remote> <branch>

mendhak, 2024-07-26:
OK try this, this should clean up your e-Paper module first: 

    cd lib/e-Paper
    git reset --hard HEAD

Then go back to the top of the repo, and get the latest e-Paper module again:  

    git submodule update --recursive --remote

You're now thinking, why did I embark on this journey 😆  but FWIW [this problem did hit many people](https://github.com/waveshareteam/e-Paper/issues/306) this year and should have been fixed now.

feh123, 2024-07-26:
I have kept my old calendar on another RPI! I'm afraid this code did not seem to work: cd lib/e-Paper
git reset --hard HEAD
HEAD is now at 6ef4a7d Merge pull request #132 from SSYYL/master
pi@Calendar:~/waveshare-epaper-display/lib/e-Paper $ cd ..
pi@Calendar:~/waveshare-epaper-display/lib $ cd ..
pi@Calendar:~/waveshare-epaper-display $ git submodule update --recursive --remote
error: The following untracked working tree files would be overwritten by checkout:
	RaspberryPi_JetsonNano/python/lib/waveshare_epd/__pycache__/__init__.cpython-311.pyc
	RaspberryPi_JetsonNano/python/lib/waveshare_epd/__pycache__/epd7in5_V2.cpython-311.pyc
	RaspberryPi_JetsonNano/python/lib/waveshare_epd/__pycache__/epdconfig.cpython-311.pyc
Please move or remove them before you switch branches.
Aborting
fatal: Unable to checkout 'a824b4f8f34dee7e721183c0154788dcde41c460' in submodule path 'lib/e-Paper'

mendhak, 2024-07-26:
I don't know why that failed. You could try to remove the submodule folder and get it again

```
rm -rf lib
git submodule update --recursive --remote
```

[image]

feh123, 2024-07-26:
Brilliant - that worked! Thanks a lot for working through this for me.

mendhak, 2024-07-26:
[image]

feh123, 2024-07-26:
I know the feeling!

closed 2024-09-21 by mendhak