mendhak/waveshare-epaper-display #88: Missing CALDAV_CALENDAR_ID sample in env.sh

msawatzky75, 2024-08-29:
Missing required config sample to setup a CALDAV calendar in the `env.sh.sample

msawatzky75, 2024-08-29:
Despite this being part of my CALDAV url, it was still required to get calender events to show up.

mendhak, 2024-09-05:
Hi there, I didn't understand. There is a caldav section in the sample file: 

https://github.com/mendhak/waveshare-epaper-display/blob/60076e6281c0f71bed47d792e1b73999fe080471/env.sh.sample#L36C1-L39C34

(although just spotted the typo, "caldave" lol)

msawatzky75, 2024-09-05:
Yes, but that did not fetch my calendar events until i specified `CALDAV_CALENDAR_ID` as referenced [here](https://github.com/mendhak/waveshare-epaper-display/blob/60076e6281c0f71bed47d792e1b73999fe080471/screen-calendar-get.py#L26).

mendhak, 2024-09-07:
Ah now I get it, didn't understand it from your earlier comments. OK that's been added now thanks, hopefully that helps other people.

closed 2024-09-07 by mendhak