mendhak/waveshare-epaper-display #89: Wrong ELF class: ELFCLASS64

Silvanfelix, 2024-11-10:
Every time I want to run the program for the display with the command ./run.sh it takes a while to process everything.
But at the end I always get this error message:

OSError: /home/Silvan/waveshare-epaper-display/lib/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd/sysfs_software_spi.so: wrong ELF class: ELFCLASS64

What does this mean exactly, because I followed the instructions step by step and I can't get what I did wrong.
Do I have to install the 32-Bit OS instead of the 64-Bit OS for the raspberry Pi Zero W?

mendhak, 2024-11-10:
Hi I'm looking at the Raspberry Pi OS page, and their Raspberry Pi OS (which their imager installs) is 32 bit: https://www.raspberrypi.com/software/operating-systems/

[image]

So I think what you're seeing would make sense, you'll need a 32 bit OS on the RPi Zero W to run this and the epaper modules. I think that's what I have as well, 

Did you use the imager to set up the SD card, just using the lite OS version?

Silvanfelix, 2024-11-10:
[image]

I downloaded the 64-Bit version. 

So you think I will have to download the 32-Bit one?

mendhak, 2024-11-10:
I am also confused. I think I will need to try to recreate the setup then, or perhaps even re-get the e-Paper module in case they've changed something recently. 

You're doing a fresh install so I expected it to work right away. This error has come up before, see [this thread](https://github.com/mendhak/waveshare-epaper-display/issues/87) for example, but all that fixed it was [getting the submodule again](https://github.com/mendhak/waveshare-epaper-display/issues/87#issuecomment-2252800555).

mendhak, 2024-11-10:
Alright, I flashed the latest Raspberry Pi. I got the same error. 

But then I did this

```bash
rm -rf lib
git submodule update --recursive --remote
```

And it started working! 

Can you try that?

mendhak, 2024-11-11:
I've updated the submodule to use the latest code. I believe that was the problem, submodule was on an old version. It should be working now. 

You shouldn't need to reformat anything, just run the command above and it should forcefully get latest.

Silvanfelix, 2024-11-12:
[image]

Now I'm getting this error.
What should I do here?

mendhak, 2024-11-12:
Hey have a look at the errors before what you're seeing on screen. It's possible that something failed before this, like the weather information fetching could have failed, so the screen-output-weather.svg never got written, and the subsequent steps failed to see it. 

If you're not sure you could try posting the full error logs here.

Silvanfelix, 2024-11-12:
ADD WEATHER INFO
---------------------------------------
2024-11-12:20:42:04 INFO     [screen-weather-get.py:74] Getting weather from Accuweather
2024-11-12:20:42:04 INFO     [utility.py:96] Cache file is stale. Fetching from source.
2024-11-12:20:42:05 ERROR    [utility.py:105] 401 Client Error: Unauthorized for url: http://dataservice.accuweather.com/forecasts/v1/daily/1day/314678?apikey=FDsNmq1MSzBowZ48xV2IP9JHIBnG9Ey2&details=true&metric=true
2024-11-12:20:42:05 ERROR    [utility.py:106] {"Code":"Unauthorized","Message":"Api Authorization failed","Reference":"/forecasts/v1/daily/1day/314678?apikey=FDsNmq1MSzBowZ48xV2IP9JHIBnG9Ey2&details=true&metric=true"}
2024-11-12:20:42:05 ERROR    [utility.py:107] {'Date': 'Tue, 12 Nov 2024 19:42:05 GMT', 'Content-Type': 'application/json', 'Content-Length': '171', 'Connection': 'keep-alive'}
Traceback (most recent call last):
  File "/home/Silvan/waveshare-epaper-display/screen-weather-get.py", line 192, in <module>
    main()
  File "/home/Silvan/waveshare-epaper-display/screen-weather-get.py", line 150, in main
    weather = get_weather(location_lat, location_long, units)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/Silvan/waveshare-epaper-display/screen-weather-get.py", line 102, in get_weather
    weather = weather_provider.get_weather()
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/Silvan/waveshare-epaper-display/weather_providers/accuweather.py", line 74, in get_weather
    response_data = self.get_response_json(url)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/Silvan/waveshare-epaper-display/weather_providers/base_provider.py", line 63, in get_response_json
    return get_json_from_url(url, headers, "cache_weather.json", self.ttl)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/Silvan/waveshare-epaper-display/utility.py", line 99, in get_json_from_url
    response.raise_for_status()
  File "/usr/lib/python3/dist-packages/requests/models.py", line 1021, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 401 Client Error: Unauthorized for url: http://dataservice.accuweather.com/forecasts/v1/daily/1day/314678?apikey=FDsNmq1MSzBowZ48xV2IP9JHIBnG9Ey2&details=true&metric=true
---------------------------------------
ADD CALENDAR INFO
---------------------------------------
2024-11-12:20:42:16 INFO     [screen-calendar-get.py:88] Fetching Outlook Calendar Events
2024-11-12:20:42:22 INFO     [screen-calendar-get.py:112] main() - {'CAL_DATETIME_1': 'Do Jan 23  ', 'CAL_DATETIME_START_1': 'Do Jan 23  ', 'CAL_DESC_1': 'Mein Geburtstag', 'CAL_DATETIME_2': 'Mo Feb 24  ', 'CAL_DATETIME_START_2': 'Mo Feb 24  ', 'CAL_DESC_2': 'Geburtstag Mama', 'CAL_DATETIME_3': 'Sa Mai 31  ', 'CAL_DATETIME_START_3': 'Sa Mai 31  ', 'CAL_DESC_3': 'Geburtstag Silvan Bachmanb', 'CAL_DATETIME_4': 'Mo Jun 16  ', 'CAL_DATETIME_START_4': 'Mo Jun 16  ', 'CAL_DESC_4': 'Geburtstag Paps', 'CAL_DATETIME_5': 'So Jul 27  ', 'CAL_DATETIME_START_5': 'So Jul 27  ', 'CAL_DESC_5': 'Geburtstag Maurus', 'CAL_DATETIME_6': 'Di Okt 7  ', 'CAL_DATETIME_START_6': 'Di Okt 7  ', 'CAL_DESC_6': 'Geburtstag Flurina', 'CAL_DATETIME_7': '', 'CAL_DESC_7': '', 'CAL_DATETIME_8': '', 'CAL_DESC_8': '', 'CAL_DATETIME_9': '', 'CAL_DESC_9': '', 'CAL_DATETIME_10': '', 'CAL_DESC_10': ''}
2024-11-12:20:42:22 INFO     [screen-calendar-get.py:114] Updating SVG
Traceback (most recent call last):
  File "/home/Silvan/waveshare-epaper-display/screen-calendar-get.py", line 119, in <module>
    main()
  File "/home/Silvan/waveshare-epaper-display/screen-calendar-get.py", line 115, in main
    update_svg(output_svg_filename, output_svg_filename, output_dict)
  File "/home/Silvan/waveshare-epaper-display/utility.py", line 58, in update_svg
    output = codecs.open(template_svg_filename, 'r', encoding='utf-8').read()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen codecs>", line 906, in open
FileNotFoundError: [Errno 2] Datei oder Verzeichnis nicht gefunden: 'screen-output-weather.svg'
---------------------------------------
EXPORT TO PNG
---------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python3.11/urllib/request.py", line 1505, in open_local_file
    stats = os.stat(localfile)
            ^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/home/Silvan/waveshare-epaper-display/screen-output-weather.svg'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/Silvan/waveshare-epaper-display/.venv/bin/cairosvg", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/Silvan/waveshare-epaper-display/.venv/lib/python3.11/site-packages/cairosvg/__main__.py", line 78, in main
    SURFACES[output_format.upper()].convert(**kwargs)
  File "/home/Silvan/waveshare-epaper-display/.venv/lib/python3.11/site-packages/cairosvg/surface.py", line 127, in convert
    tree = Tree(
           ^^^^^
  File "/home/Silvan/waveshare-epaper-display/.venv/lib/python3.11/site-packages/cairosvg/parser.py", line 386, in __init__
    bytestring = self.fetch_url(
                 ^^^^^^^^^^^^^^^
  File "/home/Silvan/waveshare-epaper-display/.venv/lib/python3.11/site-packages/cairosvg/parser.py", line 243, in fetch_url
    return read_url(url, self.url_fetcher, resource_type)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... cut ...]
2024-11-12:20:42:37 ERROR    [display.py:50] [Errno 2] No such file or directory: 'screen-output.png'
Traceback (most recent call last):
  File "/home/Silvan/waveshare-epaper-display/display.py", line 39, in <module>
    Himage = Image.open(filename)
             ^^^^^^^^^^^^^^^^^^^^
  File "/home/Silvan/waveshare-epaper-display/.venv/lib/python3.11/site-packages/PIL/Image.py", line 3131, in open
    fp = builtins.open(filename, "rb")
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: 'screen-output.png'

This is the complete log

mendhak, 2024-11-12:
Looks like the Accuweather call is failing. 

> requests.exceptions.HTTPError: 401 Client Error: Unauthorized for url: http://dataservice.accuweather.com/forecasts/v1/daily/1day/314678?apikey=XXXXXXXXXXXXXXXX&details=true&metric=true

I think you will need to try getting the API key again from AccuWeather, and you can 'test' it by substituting it in the URL:

```
http://dataservice.accuweather.com/forecasts/v1/daily/1day/314678?apikey=XXXXXXXXXXXXXXXX&details=true&metric=true
```

Side note - I'm sorry I should have told you to edit out the sensitive details like API key and calendar entries. I've edited your post to remove that information.

Silvanfelix, 2024-11-12:
It worked!!!
Thank you very much for your fast responds
I need to do some adjustments, so maybe I need your help again.

Where do I need to write this command for the automation. Because I wrote it here and it didn't work.
[image]

And can I change the language of the weather message, because everything else is in the right language?

mendhak, 2024-11-12:
Very good, that's fine not a problem if you get stuck. 

So once you've added that command via `crontab -e` (and saved and exited), wait a minute and then go into the waveshare-epaper-display folder and look at the run.log file in there. You should see another indication in the log file about why it failed.

mendhak, 2024-11-12:
If you want to hack the code for now, you could go to the accuweather file and add `&language=de` to the end of the URL here

https://github.com/mendhak/waveshare-epaper-display/blob/ab8a65751deac617fbfb4f2bc41996e46b2838b5/weather_providers/accuweather.py#L71-L72

Silvanfelix, 2024-11-12:
[image]

I can't find any file named run.log??

mendhak, 2024-11-12:
The run.log should appear if the run.sh has been run via the crontab. 

So the run.sh definitely works if you run it **manually**, but if you run `crontab -e` and add the line in there, it's not running automatically? 

When you run `crontab -l` does it list the lines you added in there?

Silvanfelix, 2024-11-13:
[image]
 
This is what it looks like, when I type crontab -l

mendhak, 2024-11-13:
I think you'll need to do some troubleshooting, nothing stands out to me about why the cron is not executing. Is cron working for you at all?

Try this one below, add it to the crontab, remember to edit using `crontab -e`  
Change the path to any simple directory, even the waveshare one will do, even /tmp will do. Wait a minute or so and see if a 'test.log' appears there. 

```bash
* * * * * cd /path/to/directory && echo "[$(date '+%Y-%m-%d %H:%M:%S')] Cron executed" >> test.log
```

Silvanfelix, 2024-11-13:
I found out, why it didn't work.
The problem was, because I wrote the stars **** without a space between them.
Now it works

Is it maybe possible to synchronize the time every minute and the weather and calendar every hour, so it doesn't have to refresh the hole screen, or is it doing this always?

mendhak, 2024-11-13:
Oh I see, that lack of spacing didn't stand out to me at all, just thought it was a font thing heh.

Anyway the weather and calendar are cached by default for 1 hour. You can see the variables CALENDAR_TTL and WEATHER_TTL in the env.sh if you want to adjust it.  So even though run.sh runs every minute, it's just using cached value for an hour. You will see various files that appear in that directory their names starting with cache.

Silvanfelix, 2024-11-13:
Perfect thanks a lot for helping me.
I may ask you some more questions but this already helped me a lot.

mendhak, 2024-11-13:
No problem, enjoy!

closed 2024-11-13 by mendhak