mendhak/waveshare-epaper-display #91: Kalendar ID Outlook

Silvanfelix, 2024-11-15:
Hi it's me agian

I'm trying to link my calendar to the display.

I have two calendar-IDs, but one won't work. Do you know why?
The first one works, but I want the other calendar to be show, but this ID won't work.

AQMkADAwATM3ZmYAZS0yYjg5LWNjAGY5LTAwAi0wMAoARgAAAzL3LrFhcgHAOAXLXNoa4lAvW32BxyY3qUAAAIBBgAAAOAXLXNoa4lAvW32BxyY3qUAAAINSwAAAA==

AAMkAGVmMDEzMTM4LTZmYWUtNDdkNC1hMDZiLTU1OGY5OTZhYmY4OABGAAAAAAKBjgx9rVEURBwAiIsqMbYjsT5e-T7KzowPTAAAAAAEGAAAiIsqMbYjsT5e-T7KzowPTAAABuC35AAA=

This is the error:

---------------------------------------
ADD CALENDAR INFO
---------------------------------------
2024-11-15:12:00:23 INFO     [screen-calendar-get.py:88] Fetching Outlook Calendar Events
Traceback (most recent call last):
  File "/home/Silvan/waveshare-epaper-display/screen-calendar-get.py", line 119, in <module>
    main()
  File "/home/Silvan/waveshare-epaper-display/screen-calendar-get.py", line 101, in main
    calendar_events = provider.get_calendar_events()
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/Silvan/waveshare-epaper-display/calendar_providers/outlook.py", line 107, in get_calendar_events
    for event in events_data["value"]:
                 ~~~~~~~~~~~^^^^^^^^^
KeyError: 'value'
---------------------------------------
EXPORT TO PNG
---------------------------------------

mendhak, 2024-11-15:
You can try setting the `DEBUG=1` value in env.sh and it should output some more information about why it failed. You can post that log here too, but if there's any personal info please XXXXXX it out.

Silvanfelix, 2024-11-17:
[image]

This is the error log from the Degub file run.log
I don't really understand what's the problem

mendhak, 2024-11-17:
Not much different then. 

When you ran the `.venv/bin/python3 outlook_util.py` did it show that calendar ID with its included events?

mendhak, 2024-11-17:
Another thing you could try, in the calendar_providers/outlook.py file, on line 106, add a logging.info(events_data) just before the for loop. That is

```
logging.info(events_data)  # <--- add this line
for event in events_data["value"]:
    ...
```

Then see what it outputs in events_data

Silvanfelix, 2024-11-17:
[image]

This is what I'm getting in the log file if I add logging.info(events_data)

Could it be, that it won't work, because it is the calendar I synchronized with my iPhone, so the Outlook-Calendar-ID is built different? So is it even somehow possible to get the icloud calendar on to the screen?

mendhak, 2024-11-17:
I don't know about iphone calendars, but I am able to synchronize Google calendars into outlook. From Google Calendar I get an ICS url, and then in Outlook Calendar I add a new calendar and paste that ICS URL. The calendar syncs then I'm able to use it in the waveshare display. 

Your other option is to use the ICS URL directly if you can find one for your calendar, see this section: https://github.com/mendhak/waveshare-epaper-display?tab=readme-ov-file#ics-calendar

Silvanfelix, 2024-11-17:
ADD CALENDAR INFO
---------------------------------------
2024-11-17:23:15:58 INFO     [screen-calendar-get.py:95] Fetching ics Calendar Events
2024-11-17:23:15:58 DEBUG    [utility.py:80] is_stale(/home/Silvan/waveshare-epaper-display/cache_ics.pickle) - True
2024-11-17:23:15:58 DEBUG    [ics.py:26] Pickle is stale, fetching ICS Calendar
Traceback (most recent call last):
  File "/home/Silvan/waveshare-epaper-display/screen-calendar-get.py", line 119, in <module>
    main()
  File "/home/Silvan/waveshare-epaper-display/screen-calendar-get.py", line 101, in main
    calendar_events = provider.get_calendar_events()
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/Silvan/waveshare-epaper-display/calendar_providers/ics.py", line 28, in get_calendar_events
    ics_events = icalevents.icalevents.events(self.ics_calendar_url, start=self.from_date, end=self.to_date)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/Silvan/waveshare-epaper-display/.venv/lib/python3.11/site-packages/icalevents/icalevents.py", line 49, in events
    content = ical_download.data_from_url(url, apple_fix=fix_apple)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/Silvan/waveshare-epaper-display/.venv/lib/python3.11/site-packages/icalevents/icaldownload.py", line 66, in data_from_url
    _, content = self.http.request(url)
                 ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/Silvan/waveshare-epaper-display/.venv/lib/python3.11/site-packages/httplib2/__init__.py", line 1541, in request
    connection_type = SCHEME_TO_CONNECTION[scheme]
                      ~~~~~~~~~~~~~~~~~~~~^^^^^^^^
KeyError: 'webcal'
---------------------------------------

I now insearted the URL from the calendar into to ICS URL section
The only error I get is webcal
Is this because the URL starts with this? 
webcal://p159-caldav.icloud.com/published/2/XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX.ics

Because when I enter the URL into the browser it works and if I change the webcal with https: it won't work

mendhak, 2024-11-17:
Yes you are right, it needs to be an `https://` URL so that it can be fetched by the code. Maybe hunt around to see if they provide some https URL for the calendars?

You might have to search discussion threads if it isn't obvious... not sure how helpful this one is: https://discussions.apple.com/thread/5950029?sortBy=rank

Silvanfelix, 2024-11-19:
I'v now tired it by creating an CALDAV-Calendar
And in Outlook it works perfectly, but it won't show it on the display
Do I have to insert the Caldendar ID aswell and where do I find this ID?

export CALDAV_CALENDAR_URL=https://XXXX-caldav.icloud.com/XXXXXXXXXXXXXX/
export CALDAV_USERNAME=XXXXXXXgmail.com
export CALDAV_PASSWORD=xyz
# export CALDAV_CALENDAR_ID=xxxxxxxxxx

I always get this error log:

           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/Silvan/waveshare-epaper-display/.venv/lib/python3.11/site-packages/vobject/icalendar.py", line 1167, in validate
    raise ValidateError(m)
vobject.base.ValidateError: 'VEVENT components cannot contain both DTEND and DURATION                     components'
---------------------------------------
EXPORT TO PNG

Silvanfelix, 2024-11-19:
If I run this script againg .venv/bin/python3 outlook_util.py I only get the ID's from the calendars I don't want, but not the Caldav one. 

With the help of the discussion thread I've got an URL, but it started with webcal://XXX-caldav.icloud.com/ and not with https: and it also didn't end with .ics

So I might have to try the option with the google calendar. But I don't understand what you mean by trying to sync to a G calendar from ICS. How does this work?

mendhak, 2024-11-19:
I believe it's very similar to the way you would sync to Outlook Calendar from an ICS URL. In Google Calendar, click the + next to Other Calendars, then From URL, and in there, paste the ICS URL. 

[image]

Regarding Outlook itself, in my testing I've simply added an ICS URL to the 'Subscribe from Web' dialog, then I've usually had to wait 10-15 minutes to see entries start to appear, and in the outlook_util.py script the Calendar ID appears for me, with calendar entries. 

[image]

Silvanfelix, 2024-11-19:
Now I struggle to get the google calendar on to the display. Even with your instruction https://github.com/mendhak/waveshare-epaper-display/issues/19#issuecomment-780397819
I don't understand this thing with the credentials.json. Do I have the create a new file? where?

mendhak, 2024-11-20:
In the last image in that comment, you'll see a little down arrow. Clicking that will give you a json file. Copy that file to the waveshare project directory and name it credentials.json

Silvanfelix, 2024-11-21:
Ok now I why it didn't worked, I was logged in with the wrong Outlook account so I didn't see the right Outlook ID.
Now I have the right one and it works perfectly fine. 

Sorry for the time waste this was my fault. 
But thanks for helping  me

mendhak, 2024-11-21:
Alright all the best!

closed 2024-11-21 by mendhak