mendhak/waveshare-epaper-display #92: Weather description

Silvanfelix, 2024-11-22:
How can I add another line for the description?
I managed to get the language to german, but now the description is too long.
I changed the max_lines=3 but it still only shoes two lines. Where do I need to change the code aswell?

ef format_weather_description(weather_description):
    if len(weather_description) < 20:
        return {1: weather_description, 2: '', 3: ''}

    splits = textwrap.fill(weather_description, 20, break_long_words=False,
                           max_lines=3, placeholder='...').split('\n')
    weather_dict = {1: splits[0]}
    weather_dict[2] = splits[1] if len(splits) > 1 else ''
    return weather_dict

mendhak, 2024-11-22:
Try this:

```python
def format_weather_description(weather_description):
    if len(weather_description) < 20:
        return {1: weather_description, 2: '', 3: ''}

    splits = textwrap.fill(weather_description, 20, break_long_words=False,
                           max_lines=3, placeholder='...').split('\n')
    weather_dict = {1: splits[0]}
    weather_dict[2] = splits[1] if len(splits) > 1 else ''
    weather_dict[3] = splits[2] if len(splits) > 2 else ''
    return weather_dict
```

You'll need to add the third line here:
https://github.com/mendhak/waveshare-epaper-display/blob/ab8a65751deac617fbfb4f2bc41996e46b2838b5/screen-weather-get.py#L172

And also in the SVG layout file.

mendhak, 2024-11-22:
By default it should be screen-template.1.svg but if you picked any other LAYOUT variable from [here](https://github.com/mendhak/waveshare-epaper-display/tree/master?tab=readme-ov-file#pick-a-layout), then the corresponding number. eg if you did SCREEN_LAYOUT=2, then edit screen-template.2.svg.

Silvanfelix, 2024-11-22:
It worked perfect
thanks a lot

mendhak, 2024-11-22:
Very good!

closed 2024-11-22 by mendhak