mendhak/waveshare-epaper-display #94: screen-output-weather.svg missing

TrantorT, 2024-12-06:
First of all, great job and thanks for putting this together, which is saving me endless hours of research :)

FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/calendarv2/waveshare-epaper-display/screen-output-weather.svg'

I think i did all the setup right, but at the ./run.sh it complains the screen-output-weather.svg is missing, and indeed it is. Has this been renamed?

TrantorT, 2024-12-06:
This is supposed to be generated on the screen-weather-get.py , and my openweather API keys have an issue. They were meant for 2.5 which is now deprecated.

closed 2024-12-06 by TrantorT

mendhak, 2024-12-06:
This happens often enough so I've added an issue - if I can add some error handling then just printing out that an error occurred in the previous step should be useful when troubleshooting. 

https://github.com/mendhak/waveshare-epaper-display/issues/95