mendhak/waveshare-epaper-display #97: &#038  rather than & in display

j0el, 2025-02-01:
Google returns the following 

'CAL_DESC_4': 'February 2025 Wheels &amp;#038; Waves at the Channel Islands Maritime Musuem'

I believe the &amp;#038 is just the ASCII code for an ampersand.  In any case the display shows the &#038 and it should just show an ampersand.  

If you google #038 you will see it is a common problem. but it is ugly.

[image]

Your program is awesome and your instructions worked well for me.  Thank you so much.

mendhak, 2025-02-04:
Hi do you definitely have the latest code? I've just created a Google calendar entry with subject `Testing a & b` and it appears just fine. I also tried with LAYOUT = 5 which matches yours. 

[image]

j0el, 2025-02-05:
I will fetch and try.  Thanks for responding so quickly.

j0el, 2025-02-07:
I have to apologize for not trying to reproduce the error before I
submitted the issue.  Upon investigation it turned out to be that that
particular calendar entry was not UTF-8.  It appeared properly in the
Google calendar because the browser fixed it upon rendering, but the actual
entry in the calendar was as you displayed it.

SO for anyone reading this - Don't be like me and do try to reproduce your
issues before reporting them.

mendhak, 2025-02-07:
Alright fair enough, sounds like you've found the issue and resolved now hopefully. If something still comes up you can always reopen/create new issue. I'll just close this one for now cheers.