mendhak/waveshare-epaper-display #98: Google Calendar API authorization - stuck

selphie8, 2025-03-17:
Hi mendhak, thanks for writing such a detailed tutorial. I really appreciate it and as it is my first Raspberry Pi project, and I have zero coding experience. 

I have been stuck at this section for 3 weeks now, where I can't seem to get the authentication process right. 

When I run .venv/bin/python3 screen-calendar-get.py, I get a URL in the Terminal that starts with "https://accounts.google.com/v3/signin/rejected?access_type=offline&app_domain=http://localhost:8
   080&client_id=.......". 

When I copy pasted that into a Chrome browser, it states "Couldn't sign you in. The browser that you're using doesn't support JavaScript, or has JavaScript turned off. To keep your Google Account secure, try signing in on a browser that has JavaScript turned on. 

Thing is, I have set Javascript to be on for all sites, and even added google.com to the sites allowed for Javascript. I have tried different browsers, or ran Google's quickstart.py to test other programs, but experienced the same error.

I even reformatted the SD card and restarted the project twice with the same results. Let me know if you have any idea or need more information of my settings.

Thank you

mendhak, 2025-03-17:
Hi I assume in general when logging into G services Chrome works fine? 

Assuming yes — I think, looking at that partial URL you posted, this probably doesn't have much to do with the Raspberry Pi/SD card itself, so you probably don't need to reformat. I'm thinking this might have to do with how the application has been set up in Google Cloud Console. 

Could you try redoing the project in Google Cloud Console, I've also got a post with screenshots here: https://github.com/mendhak/waveshare-epaper-display/issues/19#issuecomment-780397819 

Create a new project in Google Cloud Console even.

selphie8, 2025-03-25:
Hi mendhak, yes logging into G services Chrome works fine.

OK, I have created a new Google Cloud console project just to be sure. And retried as per your instructions with Desktop app.

The Google URL link I received is https://accounts.google.com/v3/signin/rejected?access_type=offline&app_
   domain=http://localhost:8080&client_id=297995538841-moirpolb6q2el78d3es
   bufh3cn8a14go.apps.googleusercontent.com&continue=https://accounts.goog
   le.com/signin/oauth/legacy/consent?authuser%3Dunknown%26part%3DAJi8hANL
   yBNX34lGriOVT51TIrFFybPQin4VLnuGweOHXjE7wZ49EXjr2fNMSmVLF9A7vjjTialtn9B
   E8ZsiweV75-jqZCOUV1NHq7J1qVTAEcjVZnbEbpGsVbNmasvkVSlMqqWN3MIkJicFDMqXB7
   9kN01Js321bKd4D5P_cv1yCdtCBhfaSmc5XxeyPIrrRunUZrdux-7IEyelafcLBGBa1Qfb_
   pJUzmFTQlgAZriMJXAxnuspyY_MpxHX4tbC6FdoF_BEZuOat_wyhsRsyIBHgmYKd-KowdmT
   0J7hiAaL9Ita6AwImGiJva-TynU_tTeNO__c8MS7vXz3WGvnnazM9sKDOfQD6Wk9s7tiX8K
   YhMedm6xlqJr4k7rMzp361bZWI2r5a4GwSyllUeDb7T_3DXi26unfnhtayXGE86OifHqkHk
   lOQKp1RKmtMXJB6fiho3CODE8Z%26flowName%3DGeneralOAuthFlow%26as%3DS-15746
   63438%253A1742913756419881%26client_id%3D297995538841-moirpolb6q2el78d3
   esbufh3cn8a14go.apps.googleusercontent.com%23&ddm=1&dsh=S-1574663438:17
   42913756419881&flowName=GeneralOAuthLite&o2v=1&opparams=%253F&rart=ANgo
   xce_7H-epCW9buqGwSPQv3AWoOd8zlOkdEdFp-CiGJcBSkpE1QdPYzY7U8PsDUiYzSrrYtQ
   K4bx9E0XCqXYKPGcItyPBa--gn2c6K8_CCvocqqDegJg&redirect_uri=http://localh
   ost:8080/&response_type=code&rhlk=js&rrk=47&scope=https://www.googleapi
   s.com/auth/calendar.readonly&service=lso&state=X7FfBq79YxOP8yfLwjfQvlJK
   gswOkz

Feel free to anonymise the URL but I thought I'd send you the full details. When I copy pasted it to a browser anyway,  it states the same thing "Couldn't sign you in. The browser that you're using doesn't support JavaScript, or has JavaScript turned off. To keep your Google Account secure, try signing in on a browser that has JavaScript turned on."

mendhak, 2025-03-25:
@selphie8 since so many people have trouble going through GCP's convoluted interface, I've done some setup work today to create an application and I'll see if I could distribute it with this application. 

Could you try this now:

Get the latest code, then switch to my testing branch:

```
git checkout new_google_client_application
```

After you do that, try running:

```
.venv/bin/python3 screen-calendar-get.py
```

And see if it takes you through the process and doesn't complain about javascript. 

[image]

[image]

treff89, 2025-03-27:
I'm having a potentially related issue, after a great and reliable experience for a few months.
Something mucked with my Python install which was creating the error ""ImportError: cannot import name 'sts' from 'google.oauth2'""

I tried uninstaling and reinstaling all the google-oauth related packages in the .venv, using requirements.txt.  Now when I run calendar-get.py, I get the following error:
2025-03-28:00:17:23 INFO     [flow.py:534] "GET /?state=LEKqgAytIX4d6ajvm7EkiuiI1Saz7V HTTP/1.1" 200 65
Traceback (most recent call last):
  File "/home/treffene/waveshare-epaper-display/screen-calendar-get.py", line 119, in <module>
    main()
  File "/home/treffene/waveshare-epaper-display/screen-calendar-get.py", line 101, in main
    calendar_events = provider.get_calendar_events()
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/treffene/waveshare-epaper-display/calendar_providers/google.py", line 61, in get_calendar_events
    service = build('calendar', 'v3', credentials=self.get_google_credentials(), cache_discovery=False)
                                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/treffene/waveshare-epaper-display/calendar_providers/google.py", line 50, in get_google_credentials
    credentials = flow.run_local_server()
                  ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/treffene/waveshare-epaper-display/.venv/lib/python3.11/site-packages/google_auth_oauthlib/flow.py", line 517, in run_local_server
    self.fetch_token(authorization_response=authorization_response)
  File "/home/treffene/waveshare-epaper-display/.venv/lib/python3.11/site-packages/google_auth_oauthlib/flow.py", line 300, in fetch_token
    return self.oauth2session.fetch_token(self.client_config["token_uri"], **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/treffene/waveshare-epaper-display/.venv/lib/python3.11/site-packages/requests_oauthlib/oauth2_session.py", line 244, in fetch_token
    self._client.parse_request_uri_response(
  File "/home/treffene/waveshare-epaper-display/.venv/lib/python3.11/site-packages/oauthlib/oauth2/rfc6749/clients/web_application.py", line 220, in parse_request_uri_response
    response = parse_authorization_code_response(uri, state=state)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/treffene/waveshare-epaper-display/.venv/lib/python3.11/site-packages/oauthlib/oauth2/rfc6749/parameters.py", line 284, in parse_authorization_code_response
    raise MissingCodeError("Missing code parameter in response.")
oauthlib.oauth2.rfc6749.errors.MissingCodeError: (missing_code) Missing code parameter in response.

selphie8, 2025-03-28:
Hi @mendhak thanks for this.

I'm feeling quite daft as I I'm still encountering issues trying to checkout the new_google_client_application branch. After running "git fetch origin", then running the checkout, i get the error of "error: pathspec 'new_google_client_application' did not match any file(s) known to git"

I can see the branch listed on the GitHub website's [branches page](https://github.com/mendhak/waveshare-epaper-display/branches), so what am I missing to try this? 

I tried to use ChatGPT to help me understand what is going on, and tried their solution of "git checkout -b new_google_client_application origin/new_google_client_application" to checkout the branch, but I am still receiving errors, because the branch is not visible to git.

Super grateful if you can dig me out of this probably simple hole.

mendhak, 2025-03-28:
No problem @selphie8  

I just tried it on my rpi, so I'll show you what I did, try this:

```
git fetch --all
git checkout new_google_client_application 
```

I see this output:

[image]

(Worst case, if none of this works, delete the folder and git clone it again then: )

Then, checkout the branch:

```
git checkout new_google_client_application
```

[image]

To eliminate errors, do some cleanup:

```
rm -f cache_*
rm token.pickle
```

Then I ran

```
.venv/bin/python3 screen-calendar-get.py
```

A URL appeared, I clicked it, went to the browser. The brower went through the above screenshots. 
I ended up on a localhost URL, I opened a _new_ session to the Raspberry Pi. Did the 

```
curl "http://localhost:8080/.....
```

Then back in the first session, eventually the calendar entries appear.

selphie8, 2025-03-28:
Thanks so much for the detailed step-by-step instructions. I did 'git fetch all', somehow don't receive the same "*[new branch] ...." line like yours showed.

And when I checked with git branch -r, the output is:
pi@home:~/waveshare-epaper-display $ git branch -r
  origin/HEAD -> origin/master
  origin/master

So I think I'll delete and try again. Will report back

selphie8, 2025-03-29:
OK, I removed the waveshare-epaper-display folder, and repeated the git clone process.

Did 'git branch -r' to double check, but it seems to be the same where I can't receive the new_google_client_application branch.

pi@home:~/waveshare-epaper-display $ git branch -r

  origin/HEAD -> origin/master
  origin/master

Am I missing something here? I don't seem to get the same output of "[new branch] new_google_client_application...." output like you did.

mendhak, 2025-03-29:
If you're still unable to switch to that branch you can also download a zip directly: 

https://github.com/mendhak/waveshare-epaper-display/archive/refs/heads/new_google_client_application.zip

and that zip should contain that specific branch.

selphie8, 2025-03-29:
Hi @mendhak, I managed to switch to the new branch, did everything. Same result unfortunately. Even with new credentials.json, deleting the whole waveshare-epaper-display and starting again.

One thing I don't know if it is worth mentioning is when I run .`venv/bin/python3 screen-calendar-get.py`, I get asked whether to allow some cookies a few times (I chose Y = yes for all of them), then the output URL in the Terminal is similar to my first post above.

I wonder if it is some settings I should check on my end.

selphie8, 2025-03-29:
Hi @mendhak , some better news! I redid everything, made sure I enabled popups and Javascript. 

When I first did `.venv/bin/python3 screen-calendar-get.py`, I got the usual rejected sign in URL (at the end) as per the first post above. Upon interrupting and stopping the process, I noticed the 'correct URL' you were referring to when I first ran that code.

It states (URL changed for anonymity):
2025-03-29:21:00:45 INFO     [screen-calendar-get.py:98] Fetching Google Calendar Events
Please visit this URL to authorize this application: https://accounts.google.com/o/oauth2/auth?response_type=code&client_id=xxxxxxx-yyyyy.apps.googleusercontent.com&redirect_uri=http%3A%2F%2Flocalhost%3A8080%2F&scope=https%3A%2F%2Fwww.googleapis.com%2Fauth%2Fcalendar.readonly&state=xxxxxyyy&access_type=offline

This was actually the first line of the code before it went to show me a rejected sign in URL due to Javascript.

I progressed to getting the http://localhost ... , did the curl on the URL on a separate terminal screen, but somehow (after 5 minutes) I still didn't get the confirmation that the authorisation is completed. I think I must have messed up the order of how things are done, and the second screen is still running the screen-calendar-get.py without much change (i.e. showing my the rejected sign inURL due to Javascript).

I'm not sure what to expect in either screen as both are stuck and feel that I must be missing something. Hope I explained this issue clearly. Not sure if this is normal, I'm using Pi Zero W H and a 128mb SD card.

mendhak, 2025-03-30:
Hi I'm getting notifications for your comments but they don't seem to appear here anymore? 

Anyway you shouldn't need to do anything in Google cloud. You can remove your credentials JSON. Just use the script directly. 

The messages you mentioned about cookies choosing yes, never seen that before could you post a screenshot.

mendhak, 2025-03-30:
So to be clear, you get that good URL to visit, only after you stop the process on first screen? The reason for asking, that first process sets up a listener on localhost 8080, so if that's interrupted, there's nothing listening. 

But what's confusing me is you're saying you get the JavaScript issue again? Even before the correct URL shows? 

Something just feels really off. 
Could you post some screenshots of that beginning bit. I couldn't understand if you're getting two URLs to visit.

mendhak, 2025-03-30:
When you did the second screen to curl localhost, remember to enclose the URL in double quote. Just mentioning it in case.

selphie8, 2025-03-30:
Hi @mendhak , I only _noticed_ the good URL that said "2025-03-29:21:00:45 INFO [screen-calendar-get.py:98] Fetching Google Calendar Events
Please visit this URL to authorize this application: https://accounts.google.com/o/oauth2/auth?response_type=code&client_id=xxxxxxx-yyyyy.apps.googleusercontent.com&redirect_uri=http%3A%2F%2Flocalhost%3A8080%2F&scope=https%3A%2F%2Fwww.googleapis.com%2Fauth%2Fcalendar.readonly&state=xxxxxyyy&access_type=offline" AFTER I interrupted the process as the last bit of the screen I'm seeing is the rejected Google sign in URL. It was there the first time I successfully ran the code.

After that, I thought I could reuse that same URL to get authorisation, but apparently the http://localhost... details generates different links each time which I suspect why the process was stuck with both screens despite doing curl (in double quotes).

In my hope to recreate this issue, I removed the waveshare-epaper-display folder and restarted the whole process again, but unfortunately this time I didn't get the "Please visit this URL to authorize this application ... "

See screenshot below. I found that I needed to use "git checkout -b new_google_client_application" to switch branches.

[image]

selphie8, 2025-03-30:
To add context, in my first run (that was semi successful), the "Please visit this URL for authorisation ... " sentence appeared between the  "Fetching Google Calendar Events" and the purple words "Sign in-Google Accounts".

Unfortunately, this time I couldn't recreate that and I'm back to being stuck here. Hope this makes sense.

selphie8, 2025-03-30:
On your other question on cookies, I managed to capture 2 out of the 3 cookies questions I get asked (which I clicked Yes each time) once I run `.venv/bin/python3 screen-calendar-get.py`.

[image]
[image]

After these 3 permissions, then I get the rejected Google Sign In URL (due to Javascript permission) which you can see above.

mendhak, 2025-03-30:
Oh that `git checkout -b` is actually revealing something - the "b" argument creates a **new** branch, it doesn't actually switch branches to an existing one! So that means to me you still don't have the repo with all its branches (there are in fact 10 branches in the repo currently). 

There's a lot of unusual stuff I'm seeing, that "priority=high allow" bit you've pasted, and the cookie screenshot, I've never seen prompts like those ever! 😲 I don't understand what they are or where they're coming from. 

ok 2 minutes...

mendhak, 2025-03-30:
OK please remove that repo folder, here's another way to directly get that branch down:

```
# remove the old directory
cd ~
rm -rf waveshare-epaper-display

# get that specific branch when cloning
git clone --recursive -b new_google_client_application  https://github.com/mendhak/waveshare-ep
aper-display.git

cd waveshare-epaper-display
git status  #  This should show "On branch new_google_client_application"

# install packages
python3 -m venv --system-site-packages .venv
.venv/bin/pip3 install -r requirements.txt

# moment of truth...
.venv/bin/python3 screen-calendar-get.py
```

mendhak, 2025-03-30:
Do you by any chance have google chrome or chromium installed on the Raspberry Pi itself? Or did you install something browser related on the Rpi? I wonder if that's somehow being picked up? That output you're getting is just strange, like something else is getting in the way of the authentication flow.

mendhak, 2025-03-31:
Have to say your screenshot looks so different, I feel that may be the root of the issue. 

Here's what it looks like for me when I run the script:

[image]

But notice in your screenshot there's a lot of extra stuff, which I've not seen before and I am wondering if something else is 'interfering' in the process. Also notice how different our URLs are. When you get your URL, it seems that something has already browsed to the 'actual' URL and rejected you. You then might be copying what's displayed, which is simply the rejection screen. Notice in my screenshot it's a much shorter URL! 

[image]

Do you have anything extra installed on the RPI? Another browser, some proxy service, anything else?

selphie8, 2025-03-31:
Hi @mendhak , I really appreciate the time you have taken to troubleshoot this with me so far. 

I don't believe I installed anything extra. When I set up the Rpi Zero WH I have used the Imager tool and followed their process, as the wpa_supplicant set up couldn't work for me and I have learned it was supposedly outdated. 

Sorry this is so frustrating though. I retried your process as per your instructions yesterday.

Everything went well until the last part `.venv/bin/python3 screen-calendar-get.py`, which produced the same results. It no longer has the 'Please visit this URL to authorize this application...".

[image]

Still has the cookie permission questions too.

[image]

If it helps, I had reformatted the SD card like 15 times, changed from 64mb to 128mb card as well when it got corrupted. But the rejected sign in URL always appears. 

I'm not sure if this info helps, but since the beginning of the project I couldn't do `git clone --recursive https://github.com/mendhak/waveshare-epaper-display.git
` as I tend to get a RPC curl 92 error. Despite troubleshooting with different internet networks, I finally found the solution [here](https://stackoverflow.com/questions/38618885/error-rpc-failed-curl-transfer-closed-with-outstanding-read-data-remaining) by doing it in stages via: 

- git clone --depth=1 https://github.com/mendhak/waveshare-epaper-display.git
- cd waveshare-epaper-display
- git fetch --depth=N, with increasing N
- and finally, git fetch -all

I did double check the commits I received and it matched 522 on Github.

mendhak, 2025-03-31:
When you used the imager tool, which OS image did you install? It should be "Raspberry Pi OS Lite" is that what you did too or some other?

Everything about your screenshot is indicating there's a weird browser getting in the way, and I also found [this interesting thread with a similar issue](https://www.pythonanywhere.com/forums/topic/14626/).

mendhak, 2025-03-31:
Also note in your git commands, don't forget it needs to be the other branch which has 523 commits

git clone --recursive **-b new_google_client_application**  https://github.com/mendhak/waveshare-epaper-display.git

mendhak, 2025-03-31:
Heh I'm throwing lots of comments. Here's another thing to try. 

If you do 

xdg-open https://example.com

you should get an error, "xdg-open: command not found" 

If you don't get that then I think you might have a Raspberry Pi desktop installed which is getting in the way.

selphie8, 2025-04-01:
Well @mendhak, I installed Pi OS lite and it all worked first time like magic (original instructions, not even trying new_google_client_application branch). Your instructions are so clear and I had NO issues whatsoever. 

Thanks so much for helping to troubleshoot and sorry I missed the detail on Pi OS Lite OS version. Didn't realise it can cause such big issues. And much appreciate the clear instructions you have which is helpful for a complete beginner like me.

Not quite sure how to change to AM/PM as yet as have 24 hour clock timing displays. But other than that, looking to see if I can find a frame and power adapter to hang it on the wall. Thank you!