mendhak/waveshare-epaper-display #101: Error building wheel for lxml

SMack05, 2025-04-13:
This look like a very cool project but I'm having a weird issue no one else seems to have encountered.

I'm running this on a new Pi Zero 2 W with a new SD card.
I flash the OS, you didn't mention exactly which one but Raspberry Pi OS Lile (64bit) no desktop seems like the best fit.
All the steps run fine no errors or warnings.
When I run ".venv/bin/pip3 install -r requirements.txt" all the requirements seem to build perfectly until the last one.
zope.interface==5.5.1

The error looks like this:
```
zope_interface error
Building wheels for collected packages: lxml
  Building wheel for lxml (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [84 lines of output]
      Building lxml version 4.9.1.
      Building without Cython.
      Building against libxml2 2.9.14 and libxslt 1.1.35
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-aarch64-cpython-311
      creating build/lib.linux-aarch64-cpython-311/lxml
```
--a whole bunch of copies that seem to run fine--

```
      running build_ext
      building 'lxml.etree' extension
      creating build/temp.linux-aarch64-cpython-311
      creating build/temp.linux-aarch64-cpython-311/src
      creating build/temp.linux-aarch64-cpython-311/src/lxml
      aarch64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -DCYTHON_CLINE_IN_TRACEBACK=0 -I/usr/include/libxml2 -Isrc -Isrc/lxml/includes -I/home/InfoFrame/waveshare-epaper-display/.venv/include -I/usr/include/python3.11 -c src/lxml/etree.c -o build/temp.linux-aarch64-cpython-311/src/lxml/etree.o -w
      aarch64-linux-gnu-gcc: fatal error: Killed signal terminated program as
      compilation terminated.
      Compile failed: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
      creating tmp
      cc -I/usr/include/libxml2 -I/usr/include/libxml2 -c /tmp/xmlXPathIniti1skvkpd.c -o tmp/xmlXPathIniti1skvkpd.o
      cc tmp/xmlXPathIniti1skvkpd.o -lxml2 -o a.out
      error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for lxml
  Running setup.py clean for lxml
Failed to build lxml
Installing collected packages: lxml, idna, humanize, httplib2, googleapis-common-protos, emoji, defusedxml, charset-normalizer, cffi, cachetools, Babel, astral, vobject, tzlocal, icalendar, google-auth, DateTime, cssselect2, cryptography, cairocffi, icalevents, google-auth-oauthlib, google-auth-httplib2, google-api-core, caldav, CairoSVG, msal, google-api-python-client
  Running setup.py install for lxml ... /
```

The / doesn't spin and the command line never returns, I have to re-boot the Pi to get back into it.
I've tried a bunch of the other OS builds and even tried editing the requirements.txt to use version 6 and 7 or zope.interface, but nothing gets me any farther.
I realize this doesn't seem to be an issue with this project but I'm kind of new to Pi's and I'm not sure where to turn.
Even a solid lead on where to look next would be a huge help.

Thank you in advance both for this cool project and any help you might have to offer

mendhak, 2025-04-13:
Yeah that's pretty unusual I don't think I've seen it before. I did some searching and from other projects this specific part might be memory related: 

> aarch64-linux-gnu-gcc: fatal error: Killed signal terminated program as
      compilation terminated.

If it's memory related that could explain why you have to reboot. 

So you could try a swap file:  

```bash
sudo fallocate -l 1G /swapfile
sudo chmod 600 /swapfile
sudo mkswap /swapfile
sudo swapon /swapfile
```

After that, try the steps again. Raspberry Pi OS Lite (non desktop) is the right one to pick. 

---

Another idea, may or may not work, you could attempt to install the binary

```bash
.venv/bin/pip install --only-binary=:all: lxml

closed 2025-04-14 by SMack05

mendhak, 2025-04-14:
Wonderful thanks for updating, it always makes me happy to hear back when there's a solution!