mendhak/waveshare-epaper-display #106: e-Paper sub git updates from vendor tarball (7z)

jeremymc526, 2025-05-19:
First, thanks or doing this!  It's kinda cool.  ;-)

Using a RPi 3B+ I had to change a few things in the e-Paper [sub git](https://github.com/waveshareteam/e-Paper).   My changes can be found [here](https://github.com/jeremymc526/e-Paper/tree/7_5_v2_7z_updates), which are a few things that were needed to get my 7.5V2 display to output correctly.   Hopefully the commit message is clear describing where the changes came from. 

Given that you are not the original author of the e-Paper support code I was planning on sending a pull request to the vendor git.  Assuming they are even receptive to PR's / changes.  In their defense, everything worked fine from their tarball on the wiki.  (https://www.waveshare.com/wiki/7.5inch_HD_e-Paper_HAT#Working_With_Raspberry_Pi)  Or would you be receptive to taking my changes in your repo as this is where I saw the breakage but not in their demos.

Thoughts, comments, opinions ?

mendhak, 2025-05-20:
I think it makes more sense to introduce the changes to the waveshare vendor so it means no drift over in this repo over time, which is bound to happen. 

But then why is the breakage happening, is the code in their git repo actually out of date, or is the submodule in this repo pointing at an old commit.

mendhak, 2025-05-20:
I do have the submodule pointing at the latest in the master branch, surely that ought to be enough?

```
[submodule "lib/e-Paper"]
        path = lib/e-Paper
        url = https://github.com/waveshare/e-Paper.git
        ignore = dirty
        branch = master
```

jeremymc526, 2025-05-21:
This is enough for your purposes, but since that repo from what I can tell and see, doesn't support the RPi3B+ which is what I am using.   If you are interested in supporting the RPi3B+ and zero we could potentially keep them in separate branches.   Unfortunately I have not heard anything back from waveshare in regards to the dual support in their repo.

jeremymc526, 2025-05-21:
[Here](https://github.com/waveshareteam/e-Paper/compare/master...jeremymc526:e-Paper:7_5_v2_7z_updates) is the diff which shows the RPi3B+ vs. RPiZero.  Unfortunately my Python game is pretty weak so I not really in a position to offer advice on how to support both boards in the same code base.

jeremymc526, 2025-05-22:
I suspect the vendor doesn't really want to support the RPi3B+ anymore as it is old as dirt.   So what I did for simplicity is create a [new branch](https://github.com/jeremymc526/e-Paper/tree/RasBerryPI_3plus) just for the 3B+ .