mendhak/waveshare-epaper-display #108: Google Calendar Quickstart Access

feh123, 2025-05-29:
This is not a bug report. It's a request for help with using my google calendar. I have received a final notice from google that they will delete my Client for my Calendar. As anyone managed to fix this? My experience of doing anything on the Google Cloud Console is quite limited. Apparently I need to do a client update or a token exchange. Does anyone have a method they could share for doing this? Any info very much appreciated. Thanks.

mendhak, 2025-05-29:
I got this email too, about "Automatic deletion of unused OAuth clients", and it listed some of my actively used tokens. I really don't understand why it thinks they're inactive when they're used daily. 

It's possible that the email is in error, and I see there are some confused Reddit threads about it too. [One](https://www.reddit.com/r/googlecloud/comments/1ky7aff/automatic_deletion_of_unused_oauth_clients/) [Two](https://www.reddit.com/r/googlecloud/comments/1ky4gs4/action_advised_manage_your_unused_oauth_clients/) [Three](https://www.reddit.com/r/Firebase/comments/1ky75x3/automatic_deletion_of_unused_oauth_clients/)

But what I think doesn't matter if their automation thinks it's inactive. 

Anyway the way to do a forced refresh would be, 

Go into the waveshare directory, 
Delete the `token.pickle` file, 
Rerun that original Google Calendar [authentication step](https://github.com/mendhak/waveshare-epaper-display?tab=readme-ov-file#google-calendar-setup)... `.venv/bin/python3 screen-calendar-get.py` step and go through the process again where it gives you a URL, open in browser, sign in, invoke the return URL in a new SSH session. 

---

I couldn't help but notice this at the [end of their page](https://developers.googleblog.com/en/usability-and-safety-updates-to-google-auth-platform/): 

[image]

🙄

mendhak, 2025-05-29:
Oh, breaking news, it does look like it might be in error. This is supposedly a comment from a Google Employee:

https://www.reddit.com/r/googlecloud/comments/1ky7aff/automatic_deletion_of_unused_oauth_clients/muwx1qc/

feh123, 2025-05-29:
Thanks @mendhak - maybe I will wait for 1 June and see what happens. When I go to my activity it does show no usage! Do you see anything for your authorisations? This reddit post has a feel of a non-denial denial! Thanks again for the forced refresh info.

mendhak, 2025-05-29:
I see no usage either! I see a few other comments in those three threads where people don't see last usage either.

feh123, 2025-05-30:
Apologises but I have a problem creating a now token pickle. I follow your instructions and everything seems fine. But I get to the final part where I run .venv.... (from waveshare-epaper-display). I copy the https address, open chrome, paste the address and wait. I get to the beware page and agree to go on to quickstsrt. I then get a new address and copy this. I open the RPI again, go to waveshare-epaper-display, paste the address and add curl. I get error (1) 2873, (2) 2874 and then a new prompt with the error message failed to connect to local host. I know sometimes my local host is a bit flakey - can I use an IP address? Or have a missed something? Thanks!

mendhak, 2025-05-30:
OK yes that's it, and for future reference it is also [in the README](https://github.com/mendhak/waveshare-epaper-display?tab=readme-ov-file#google-calendar-setup). 

Also I went back today and I see that they now show information about the clients with warnings next to them. I think if you have a look now probably your client won't have any warning against it.

feh123, 2025-05-31:
Thanks. I will keep watching the API Credetentials page - at present my new credetential shows no activity! It's all something of a mystery.

mendhak, 2025-05-31:
Just got another email from Google, same subject, but now they said none of my clients are going to be deleted. So it does sound like they sent the original email in error.

feh123, 2025-05-31:
Okay. Thanks, I guess we can close this now. It did make me reset a token which is good practice.

mendhak, 2025-05-31:
It was a token effort 🤣 🤣

closed 2025-05-31 by mendhak