mendhak/waveshare-epaper-display #110: Incorrect time on calendar events using Proton ICS.

Leminski, 2025-08-29:
I have an issue that's driving me barmy. I recently swapped to Proton calendar using ICS. Since doing so, my calendar entries are all showing 1 hour earlier than they should. 

I'm based in Dublin, but what's strange is that the issue remains if I set any other time zone! Using timedatectl, setting Europe/Athens or Asia/Tokyo, the calendar or clock doesn't shift. 

To reproduce:
- Be Irish. 
- Clone latest version. 
- Set Weather provider to MET Eireann. 
- Got link from Proton Calendar in ICS format. 
- Set Python Env and got waveshare libraries. 
- Set time to Europe/Dublin: Time correct (2PM), Calendar wrong (-1 Hour)
- Set time to Europe/Athens: Time correct (still 2PM), Calendar wrong (still -1 Hour)
- Set time to Asia/Tokyo: Time correct (still 2PM), Calendar wrong (still -1 Hour)
- Delete cache in case and repeat. Nope
- Check logs. screen-calendar-get.py:112 shows 3AM for Japan. That's weird.
- Scream a bit. 

What am I doing wrong? Its seems screen output does it's own thing and I don't know why my calendar is 1 hour out.

Leminski, 2025-08-29:
OK, just as an idea, added an offset to ics.py. Same result. Set it to 7200 or two hours. Logs are showing correct, but image is still wrong. ?

Leminski, 2025-08-29:
Set an offset in ics.py as a workaround. Now it looks right, but I know it's still a bad workaround.

mendhak, 2025-08-29:
ICS has always been a mess... 

Does it work if you undo the offset, but then comment out these two lines?

https://github.com/mendhak/waveshare-epaper-display/blob/d63d78905f93bfb6d7b3bfb1e2b0489334985d1c/calendar_providers/ics.py#L42-L44

I'm wondering if there is something in the icalevents library which isn't recognizing something about those entries.

mendhak, 2025-08-30:
So, I think looking at this modified code, and the debug log, that indicates to me that the icalevents library itself (the call to icalevents.icalevents.events) is returning a different time or  

Interestingly while trying to troubleshoot this, I ran into a different set of issues, it would fail to parse a Google ICS (even though I have tested it before). I updated the library to a latest version and I was able to recreate the same problem! My events were 1 hour out. 

Now here is some new code to try... 

https://github.com/mendhak/waveshare-epaper-display/pull/111/files

At the very least try modifying the code in screen calendar get, and ics.py as I've shown in that diff. 

The big difference here is that I'm now passing :

tzinfo=get_localzone() so it should pick up local    
strict=True so it reads the timezone from the ICS itself (I would have expected this as default!)
sort=True so it returns the most recent first

Leminski, 2026-01-28:
Apologies for the massive delay. Only getting to this now. 

Is requirements correct here? Got a conflict. 

The conflict is caused by:
    The user requested urllib3==2.5.0
    icalevents 0.3.0 depends on urllib3>=1.26.5
    requests 2.28.1 depends on urllib3<1.27 and >=1.21.1

I'll just pop up requests and recurring-ical-events a bit to a version that overlaps. 

Anyway, with that done:

'CAL_DATETIME_1': 'Tomorrow 15:00 - 15:30', 'CAL_DATETIME_START_1': 'Tomorrow 15:00', 'CAL_DESC_1': 'TestDub3pm', 'CAL_DATETIME_2': 'Tomorrow 16:00 - 16:30', 'CAL_DATETIME_START_2': 'Tomorrow 16:00', 'CAL_DESC_2': 'Testutc4pm', 'CAL_DATETIME_3': 'Tomorrow 19:00 - 20:00', 'CAL_DATETIME_START_3': 'Tomorrow 19:00', 'CAL_DESC_3': '7:00 Insight Matters', 'CAL_DATETIME_4': 'Sunday  ', 'CAL_DATETIME_START_4': 'Sunday  ', 'CAL_DESC_4': 'Sift Sand and Scrub Litter boxes', 

These look correct! Whoopee. 
1st event is using Irish time at 3PM. 
2nd event is UTC for 4PM
3rd was an imported one from way back
4th is an all day. 

This looks great. I was experimenting with a multi cal addon a while back, so now I just gotta get everything into one ICS now. 

Thanks!